SmingHub/Sming #2300: Version 10 build issue

frankdownunder, 2021-04-05:
With version 10 of the c++ compiler,
 ``
Building /opt/Sming/Sming/out/Host/debug/lib/clib-Storage.a
C+ /opt/Sming/Sming/Components/Storage/src/ProgMem.cpp
C+ /opt/Sming/Sming/Components/Storage/src/Storage.cpp
In file included from /opt/Sming/Sming/Wiring/WConstants.h:20,
                 from /opt/Sming/Sming/Wiring/WString.h:59,
                 from /opt/Sming/Sming/Components/Storage/src/include/Storage/Device.h:12,
                 from /opt/Sming/Sming/Components/Storage/src/include/Storage/CustomDevice.h:7,
                 from /opt/Sming/Sming/Components/Storage/src/include/Storage/ProgMem.h:12,
                 from /opt/Sming/Sming/Components/Storage/src/ProgMem.cpp:11:
/usr/include/c++/10/cstdint:38:10: fatal error: bits/c++config.h: No such file or directory
   38 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
``

To reproduce:

- install g++-10
- make it the default
- make -j6 SMING_ARCH=Host

mikee47, 2021-04-05:
`sudo apt install g++-10-multilib` shoudl fix this

mikee47, 2021-04-05:
We could do with a way to check for the 32 bit libraries and give a more useful message...

mikee47, 2021-04-05:
Point of interest - running `sudo apt install g++-10-multilib` will also install dependent packages gcc-10, g++10 and gcc-10-multilib.

Matzz, 2021-04-06:
Log long time ago when I was using Windows I also had series of issues. I end up using https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk (Ubuntu on Windows)

slaff, 2021-04-08:
@frankdownunder Feel free to re-open the issue if the problem still persists after installing `g++-10-multilib`

closed 2021-04-08 by slaff