SmingHub/Sming #2309: Errors on compiling after a fresh git clone

teicors, 2021-04-20:
Hi,
I cloned today the git repository and issuing a make, I got this error:

```
Building /opt/sming/Sming/out/Esp8266/debug/lib/clib-IFS-577940125a31051edc0fbae78576d71a.a
C+ /opt/sming/Sming/Components/IFS/src/Access.cpp
C+ /opt/sming/Sming/Components/IFS/src/Compression.cpp
C+ /opt/sming/Sming/Components/IFS/src/Directory.cpp
C+ /opt/sming/Sming/Components/IFS/src/Error.cpp
C+ /opt/sming/Sming/Components/IFS/src/FileAttributes.cpp
C+ /opt/sming/Sming/Components/IFS/src/File.cpp
C+ /opt/sming/Sming/Components/IFS/src/FileDevice.cpp
C+ /opt/sming/Sming/Components/IFS/src/FileSystem.cpp
C+ /opt/sming/Sming/Components/IFS/src/GdbFileSystem.cpp
C+ /opt/sming/Sming/Components/IFS/src/Helpers.cpp
C+ /opt/sming/Sming/Components/IFS/src/HostUtil.cpp
/opt/sming/Sming/Components/IFS/src/HostUtil.cpp:24:18: error: conflicting declaration of C function 'char* strerror_r(int, char*, size_t)'
   24 | extern "C" char* strerror_r(int, char*, size_t);
      |                  ^~~~~~~~~~
In file included from /opt/esp-quick-toolchain/xtensa-lx106-elf/xtensa-lx106-elf/include/assert.h:9,
                 from /opt/sming/Sming/Arch/Esp8266/Components/libc/include/assert.h:1,
                 from /opt/sming/Sming/Components/IFS/src/include/IFS/Host/../Types.h:25,
                 from /opt/sming/Sming/Components/IFS/src/include/IFS/Host/../Error.h:31,
                 from /opt/sming/Sming/Components/IFS/src/include/IFS/Host/Util.h:25,
                 from /opt/sming/Sming/Components/IFS/src/HostUtil.cpp:20:
/opt/esp-quick-toolchain/xtensa-lx106-elf/xtensa-lx106-elf/include/string.h:106:5: note: previous declaration 'int strerror_r(int, char*, size_t)'
  106 | int _EXFUN(strerror_r,(int, char *, size_t))
      |     ^~~~~~
/opt/sming/Sming/Components/IFS/src/HostUtil.cpp: In function 'String IFS::Host::getErrorString(int)':
/opt/sming/Sming/Components/IFS/src/HostUtil.cpp:62:68: error: call of overloaded 'strerror_r(int, char [256], unsigned int)' is ambiguous
   62 |   auto r = strerror_r(-Error::toSystem(err), buffer, sizeof(buffer));
      |                                                                    ^
In file included from /opt/esp-quick-toolchain/xtensa-lx106-elf/xtensa-lx106-elf/include/assert.h:9,
                 from /opt/sming/Sming/Arch/Esp8266/Components/libc/include/assert.h:1,
                 from /opt/sming/Sming/Components/IFS/src/include/IFS/Host/../Types.h:25,
                 from /opt/sming/Sming/Components/IFS/src/include/IFS/Host/../Error.h:31,
                 from /opt/sming/Sming/Components/IFS/src/include/IFS/Host/Util.h:25,
                 from /opt/sming/Sming/Components/IFS/src/HostUtil.cpp:20:
/opt/esp-quick-toolchain/xtensa-lx106-elf/xtensa-lx106-elf/include/string.h:106:5: note: candidate: 'int strerror_r(int, char*, size_t)'
  106 | int _EXFUN(strerror_r,(int, char *, size_t))
      |     ^~~~~~
/opt/sming/Sming/Components/IFS/src/HostUtil.cpp:24:18: note: candidate: 'char* strerror_r(int, char*, size_t)'
   24 | extern "C" char* strerror_r(int, char*, size_t);
      |                  ^~~~~~~~~~
make[2]: *** [/opt/sming/Sming/component-wrapper.mk:170: src/HostUtil.o] Errore 1
make[1]: *** [/opt/sming/Sming/project.mk:433: IFS-build] Errore 2
make[1]: uscita dalla directory «/home/angelo/Software/esp8266/project/rsyslog_client»
make: *** [/opt/sming/Sming/project.mk:22: all] Errore 2
```

Cheers

slaff, 2021-04-20:
Try first `make -C $SMING_HOME dist-clean` after that clean your application `cd <path/to/app>; make clean` and finally try to rebuild it.

mikee47, 2021-04-20:
@teicors Assuming you've done as as slaff suggests, if the problem persists could you amend `Sming/Components/IFS/src/HostUtil.cpp` as follows and see if that fixes your problem:

```
 
#include <string.h>
#undef strerror_r ///< Add this line
extern "C" char* strerror_r(int, char*, size_t);
 
```

kmihaylov, 2021-04-20:
@teicors would you please recheck if you set ```ESP_HOME``` and ```EQT_ROOT``` environment variables?
```
env | grep -E "EQT|ESP"
```
You should get result like
```
EQT_ROOT=/opt/esp-quick-toolchain
ESP_HOME=/opt/esp-quick-toolchain
```

@slaff @mikee47 I believe it would be helpful if the make script checks somehow for proper compiler, since just a few minutes ago I had the same problem as @teicors  and it happened that my env variables were missing. Curious how it successfully compiles most of the stuff before ```HostUtil.cpp```.

teicors, 2021-04-20:
@kmihaylov here's my result:
```
(base) angelo@angelo-thinkpad-x1-carbon-4th:~/Software/esp8266/project/rsyslog_client$ env | grep -E "EQT|ESP"
ESP_HOME=/opt/esp-quick-toolchain
(base) angelo@angelo-thinkpad-x1-carbon-4th:~/Software/esp8266/project/rsyslog_client$ 
```
I'm missing the
`EQT_ROOT=/opt/esp-quick-toolchain
`

but

```
Building /opt/sming/Sming/out/Esp8266/debug/lib/clib-IFS-577940125a31051edc0fbae78576d71a.a
C+ /opt/sming/Sming/Components/IFS/src/HostUtil.cpp
/opt/sming/Sming/Components/IFS/src/HostUtil.cpp:24:18: error: conflicting declaration of C function 'char* strerror_r(int, char*, size_t)'
   24 | extern "C" char* strerror_r(int, char*, size_t);
      |                  ^~~~~~~~~~
In file included from /opt/esp-quick-toolchain/xtensa-lx106-elf/xtensa-lx106-elf/include/assert.h:9,
                 from /opt/sming/Sming/Arch/Esp8266/Components/libc/include/assert.h:1,
                 from /opt/sming/Sming/Components/IFS/src/include/IFS/Host/../Types.h:25,
                 from /opt/sming/Sming/Components/IFS/src/include/IFS/Host/../Error.h:31,
                 from /opt/sming/Sming/Components/IFS/src/include/IFS/Host/Util.h:25,
                 from /opt/sming/Sming/Components/IFS/src/HostUtil.cpp:20:
/opt/esp-quick-toolchain/xtensa-lx106-elf/xtensa-lx106-elf/include/string.h:106:5: note: previous declaration 'int strerror_r(int, char*, size_t)'
  106 | int _EXFUN(strerror_r,(int, char *, size_t))
      |     ^~~~~~
/opt/sming/Sming/Components/IFS/src/HostUtil.cpp: In function 'String IFS::Host::getErrorString(int)':
/opt/sming/Sming/Components/IFS/src/HostUtil.cpp:62:68: error: call of overloaded 'strerror_r(int, char [256], unsigned int)' is ambiguous
   62 |   auto r = strerror_r(-Error::toSystem(err), buffer, sizeof(buffer));
      |                                                                    ^
In file included from /opt/esp-quick-toolchain/xtensa-lx106-elf/xtensa-lx106-elf/include/assert.h:9,
                 from /opt/sming/Sming/Arch/Esp8266/Components/libc/include/assert.h:1,
                 from /opt/sming/Sming/Components/IFS/src/include/IFS/Host/../Types.h:25,
                 from /opt/sming/Sming/Components/IFS/src/include/IFS/Host/../Error.h:31,
                 from /opt/sming/Sming/Components/IFS/src/include/IFS/Host/Util.h:25,
                 from /opt/sming/Sming/Components/IFS/src/HostUtil.cpp:20:
/opt/esp-quick-toolchain/xtensa-lx106-elf/xtensa-lx106-elf/include/string.h:106:5: note: candidate: 'int strerror_r(int, char*, size_t)'
  106 | int _EXFUN(strerror_r,(int, char *, size_t))
      |     ^~~~~~
/opt/sming/Sming/Components/IFS/src/HostUtil.cpp:24:18: note: candidate: 'char* strerror_r(int, char*, size_t)'
   24 | extern "C" char* strerror_r(int, char*, size_t);
      |                  ^~~~~~~~~~
make[2]: *** [/opt/sming/Sming/component-wrapper.mk:170: src/HostUtil.o] Errore 1
make[1]: *** [/opt/sming/Sming/project.mk:433: IFS-build] Errore 2
make[1]: uscita dalla directory «/home/angelo/Software/esp8266/project/rsyslog_client»
make: *** [/opt/sming/Sming/project.mk:22: all] Errore 2
(base) angelo@angelo-thinkpad-x1-carbon-4th:~/Software/esp8266/project/rsyslog_client$ env | grep -E "EQT|ESP"
ESP_HOME=/opt/esp-quick-toolchain
EQT_ROOT=/opt/esp-quick-toolchain
(base) angelo@angelo-thinkpad-x1-carbon-4th:~/Software/esp8266/project/rsyslog_client$ 
```
I added the line as @mikee47 suggested, but the only change is the line number changed from 24 to 25 :)

```
Building /opt/sming/Sming/out/Esp8266/debug/lib/clib-IFS-577940125a31051edc0fbae78576d71a.a
C+ /opt/sming/Sming/Components/IFS/src/HostUtil.cpp
/opt/sming/Sming/Components/IFS/src/HostUtil.cpp:25:18: error: conflicting declaration of C function 'char* strerror_r(int, char*, size_t)'
   25 | extern "C" char* strerror_r(int, char*, size_t);
      |             
[......]
```     ^~~~~~~~~~

teicors, 2021-04-20:
I forgot: I did the @slaff  suggestion as first step !

teicors, 2021-04-20:
@mikee47 the new version solves my issue.

But I'm facing another issue: I cannot compile an old project because
```
(base) angelo@angelo-thinkpad-x1-carbon-4th:~/Software/esp8266/project/SoilMeter_nonos$ make help

SoilMeter_nonos: Invoking 'help' for Esp8266 (debug) architecture
Using configuration 'standard'
/opt/sming/Sming/Components/spiffs/component.mk:10: *** DISABLE_SPIFFS is no longer supported; please remove this option from your component.mk file.  Arresto.
```

But there were nothing inside the file component.mk. The value was inside the file config.mk:

```
(base) angelo@angelo-thinkpad-x1-carbon-4th:~/Software/esp8266/project/SoilMeter_nonos$ grep -ir DISABLE_SPIFFS *
out/Esp8266/debug/config.mk:DISABLE_SPIFFS=0
out/Esp8266/debug/config.mk:CACHED_VAR_NAMES := APP_NAME COM_OPTS COM_PORT COM_PORT_ESPTOOL COM_PORT_GDB COM_SPEED COM_SPEED_ESPTOOL COM_SPEED_SERIAL DEBUG_PRINT_FILENAME_AND_LINE DEBUG_VERBOSE_LEVEL DISABLE_SPIFFS DISABLE_WIFI ENABLE_CMD_EXECUTOR ENABLE_CUSTOM_HEAP ENABLE_CUSTOM_LWIP ENABLE_CUSTOM_PWM ENABLE_ESPCONN ENABLE_GDB ENABLE_HTTP_SERVER_MULTIPART ENABLE_LEGACY_BUILD ENABLE_LWIPDEBUG ENABLE_SMART_CONFIG ENABLE_SSL ENABLE_TASK_COUNT ENABLE_WPS FULL_COMPONENT_BUILD GDB_CMDLINE GDB_UART_SWAP HTTP_SERVER_EXPOSE_DATE HTTP_SERVER_EXPOSE_NAME HTTP_SERVER_EXPOSE_VERSION KILL_TERM LOCALE MQTT_NO_COMPAT PIP_ARGS PROJECT_DIR RBOOT_BIG_FLASH RBOOT_GPIO_ENABLED RBOOT_GPIO_SKIP_ENABLED RBOOT_LD_0 RBOOT_LD_1 RBOOT_LD_TEMPLATE RBOOT_ROM0_ADDR RBOOT_ROM1_ADDR RBOOT_ROM2_ADDR RBOOT_ROM_0 RBOOT_ROM_1 RBOOT_RTC_ENABLED RBOOT_SILENT RBOOT_SPIFFS_0 RBOOT_SPIFFS_1 RBOOT_TWO_ROMS SDK_BASE SERVER_OTA_PORT SPIFF_BIN SPIFF_FILEDESC_COUNT SPIFF_FILES SPIFF_SIZE SPI_MODE SPI_SIZE SPI_SPEED SSL_DEBUG STRICT STRING_OBJECT_SIZE TASK_QUEUE_LENGTH TERMINAL TRACE USER_CFLAGS USE_US_TIMER WIFI_PWD WIFI_SSID
(base) angelo@angelo-thinkpad-x1-carbon-4th:~/Software/esp8266/project/SoilMeter_nonos$ ls out/
build/         build-type.mk  Esp8266/       firmware/      
(base) angelo@angelo-thinkpad-x1-carbon-4th:~/Software/esp8266/project/SoilMeter_nonos$ rm out/Esp8266/debug/config.mk
(base) angelo@angelo-thinkpad-x1-carbon-4th:~/Software/esp8266/project/SoilMeter_nonos$ make help

SoilMeter_nonos: Invoking 'help' for Esp8266 (debug) architecture
[... The "make help" appears ... ]
```

May be exists a make's parameter to clean the old build/configuration ? Don't say "config-clean" or "dist-clean" please :)
No luck with "make components-clean submodules-clean part-clean"

But as usual, THANKS !

mikee47, 2021-04-20:
LOL. I'm going to say it. `config-clean`! All it does is remove `out/Esp8266/build/config.mk` (or whatever arch/build is currently set). If you don't want to do that, just edit the config.mk file and remove the lines manually - you'll need to remove both references to DISABLE_SPIFFS.

If that doesn't fix the problem then check DISABLE_SPIFFS isn't set in the environment? And run `grep` on the Sming repo too...

teicors, 2021-04-20:
mmhh
make config-clean
make SMING_ARCH=Esp8266 SPI_MODE=dio SPI_SIZE=4M COM_SPEED_ESPTOOL=1000000 SPIFF_SIZE=511024 SDK_BASE='$(SMING_HOME)' COM_PORT=/dev/ttyUSB3
Paris vaut une messe !
No more headache !
LOL

closed 2021-04-21 by teicors