SmingHub/Sming #2310: spiffs-two-roms hardware profile fails to boot

kmihaylov, 2021-04-20:
With Sming 07ecd3bdf33f5b56ec1c32c9587df456ffaa4c07 (older too) Basic_Blink using ```HWCONFIG := spiffs-two-roms``` set in ```component.mk``` fails to run on the device. 
At boot this happens:
```
--- DTR inactive ---
--- RTS inactive ---
rd$ܾ��l�|
            �$�p�r���{���
                              c��no�$no���bp��$rdrlp�o�

                                                           �
                                                            l
                                                             

                                                              "
                                                                o�|l��
                                                                        c��no�$��d �no$`Nr���N

                                                                                                    r�l`p�N�s�����

```

and it continues forever.

With ```standard``` hwconfig the device boots sucessfully.

mikee47, 2021-04-20:
I've attempted to reproduce this issue as follows:

1. Edit component.mk

```
HWCONFIG = standard
```

2. Build and flash

```
make flash
```

3. Edit component.mk

```
HWCONFIG = spiffs-two-roms
SPIFF_FILES =
```

4. Clean, build and flash

```
make config-clean
make flash
```

No problem so far...

mikee47, 2021-04-20:
Note: to see that actual error above set COM_SPEED=74880

slaff, 2021-04-20:
I guess you can try:
```
make flashinit
```

and then flash the application on the device.

kmihaylov, 2021-04-20:
I did
```
make flashinit
```
Modified ```component.mk``` with these changes:
```
## Select source of content for default `spiffs` partition when built
SPIFF_FILES = 

## Refer to Basic_rBoot sample for options relating to rBoot
HWCONFIG = spiffs-two-roms
```
Then
```
make config-clean
make flash
```
And this is the result:
```
--- DTR inactive ---
--- RTS inactive ---

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 1240, room 16 
tail 8
chksum 0xc5
load 0x3ffe8000, len 640, room 0 
tail 0
chksum 0x07
csum 0x07

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   DIO
Flash Speed:  40 MHz

Resetting boot config.
Invalid rom selected, defaulting to 0.
Rom 0 at 0 is bad.
Rom -1 at 0 is bad.
Rom -1 at 0 is bad.
Rom -1 at 0 is bad.
```

According to this output it should have flashed the partition table...
```
WriteFlash 0x00000=out/Esp8266/debug/firmware/rboot.bin 0x003fa000=out/Esp8266/debug/firmware/partitions.bin 0x00002000=out/Esp8266/debug/firmware/rom0.bin 0x00108000=out/Esp8266/debug/firmware/rom1.bin 0x00200000=out/Esp8266/debug/firmware/spiff_rom.bin  0x003fc000=/opt/Sming/Sming/Arch/Esp8266/Components/esp8266/ESP8266_NONOS_SDK/bin/esp_init_data_default.bin 
python3 /opt/Sming/Sming/Components/esptool/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 --chip esp8266 --before default_reset --after hard_reset write_flash -z -fs 4MB -ff 40m -fm dio 0x00000 out/Esp8266/debug/firmware/rboot.bin 0x003fa000 out/Esp8266/debug/firmware/partitions.bin 0x00002000 out/Esp8266/debug/firmware/rom0.bin 0x00108000 out/Esp8266/debug/firmware/rom1.bin 0x00200000 out/Esp8266/debug/firmware/spiff_rom.bin  0x003fc000 /opt/Sming/Sming/Arch/Esp8266/Components/esp8266/ESP8266_NONOS_SDK/bin/esp_init_data_default.bin
```

Btw is any shortcut to read the partition table?
```make readpart PART=rom0``` is for the slot, but for the partition table?

kmihaylov, 2021-04-22:
The problem was due to my (mis)use of EQT gnu23 instead of the supported gnu20 version.

closed 2021-04-22 by kmihaylov