SmingHub/Sming #2328: ESP32 Support

gbaranski, 2021-05-12:
Hello, I'm creator of [Houseflow](https://github.com/gbaranski/houseflow), it's an IoT platform for embedded devices like ESP8266 and also willing to support ESP32. I've been using [ESP8266_RTOS_SDK](https://github.com/espressif/ESP8266_RTOS_SDK), but for me it is too close to metal, and toolchain is quite complicated, no support for C++, and as far as I know no support for ESP32. I've recently tried Arduino Framework, but it has weird API, it encourages way of programming with `void loop()`. And now I found your project, it looks really great, and I have few questions about it,

- How compatible ESP8266 code is with ESP32? Is it like with Arduino framework, so you can copy-paste most things and they will work?
- How stable ESP32 is with Sming framework? 
- Do libraries such as websockets, HTTP,  OTA, and etc work on ESP32?

Thanks in advance.

slaff, 2021-05-12:
@gbaranski We do have some initial support for the ESP32 architecture but it is not complete. For example there is no SPI or I2S support yet.  ​

>  Do libraries such as websockets, HTTP, OTA, and etc work on ESP32?

All network protocols, clients and server that are supported in ESP8266 are supported also here. This means that the HTTP client and server will work, the SSL support is there, the MQTT client will work, FTP and so on. 

OTA on ESP32 is not working yet. For Esp8266 we use rBoot as the bootloader but ESP32 has another bootloader. 
With a bit of work the OTA component could support also ESP32. 

> How stable ESP32 is with Sming framework?

Hard to say for an architecture that is marked as experimental and incomplete. The network is stable and working. For the other features you might find a lot of rough edges.

> How compatible ESP8266 code is with ESP32? 

I guess you wanted to ask how portable is Sming code between the different supported architectures: Esp8266, Esp32, Host. If that is the question then all features coming from Sming that are supported for Esp8266 and Esp32 should work without any porting/re-writing. Sming itself takes care to hide the differences and do the heavy lifting.

> Is it like with Arduino framework, so you can copy-paste most things and they will work?

Yes and no. Take a look at the multiple [samples](https://github.com/SmingHub/Sming/tree/develop/samples) that we have to get a better understanding. And don't forget to read the [documentation](https://sming.readthedocs.io/).

slaff, 2021-05-27:
@gbaranski  OTA for Esp32 has landed with the latest commits in our `develop` branch.

gbaranski, 2021-05-27:
> @gbaranski We do have some initial support for the ESP32 architecture but it is not complete. For example there is no SPI or I2S support yet. ​
> 
> > Do libraries such as websockets, HTTP, OTA, and etc work on ESP32?
> 
> All network protocols, clients and server that are supported in ESP8266 are supported also here. This means that the HTTP client and server will work, the SSL support is there, the MQTT client will work, FTP and so on.
> 
> OTA on ESP32 is not working yet. For Esp8266 we use rBoot as the bootloader but ESP32 has another bootloader.
> With a bit of work the OTA component could support also ESP32.
> 
> > How stable ESP32 is with Sming framework?
> 
> Hard to say for an architecture that is marked as experimental and incomplete. The network is stable and working. For the other features you might find a lot of rough edges.
> 
> > How compatible ESP8266 code is with ESP32?
> 
> I guess you wanted to ask how portable is Sming-based application between the different supported architectures: Esp8266, Esp32, Host. If that is the question then all features coming from Sming that are supported for Esp8266 and Esp32 should work without any porting/re-writing. Sming itself takes care to hide the differences and do the heavy lifting.
> 
> > Is it like with Arduino framework, so you can copy-paste most things and they will work?
> 
> Yes and no. Take a look at the multiple [samples](https://github.com/SmingHub/Sming/tree/develop/samples) that we have to get a better understanding. And don't forget to read the [documentation](https://sming.readthedocs.io/).

That sound's great, I think I'll keep waiting on the OTA updates, it's must-have feature for me.

slaff, 2021-05-27:
> I think I'll keep waiting on the OTA updates

You don't have to wait. They are already part of the `develop` branch. Or you meant something else?

gbaranski, 2021-05-27:
> > I think I'll keep waiting on the OTA updates
> 
> You don't have to wait. They are already part of the `develop` branch. Or you meant something else?

I meant that I'd like to wait for official release with OTA Support, however I think I can play a bit with ESP8266 and Sming for now.

slaff, 2021-09-28:
The new stable version is officially released: https://github.com/SmingHub/Sming/releases/tag/4.4.0. It contains a lot of ESP32 related improvements. Make sure to use IDF v.4.3.0. If you are not sure how to install then download the latest stable version and run `$SMING/../Tools/install.sh esp32`

closed 2021-09-28 by slaff