SmingHub/Sming #2330: Projects won't compile when using vector<> or list<> arrays

AntonGrushin, 2021-05-14:
Hello, team!
I bumped into a problem with latest Sming versions that I can not use `std::vector<>` and `std::list<>` arrays. no projects will compile if I use any of that.

Here is an example on how to repeat the issue. Clean Ubuntu 20.04.2 LTS installation. Running Sming install and Basic_Blink compilation from docs:
```bash
sudo chown $USER:$USER /opt
git clone https://github.com/SmingHub/Sming /opt/sming
source /opt/sming/Tools/install.sh all
cd $SMING_HOME/../samples/Basic_Blink
make
```
Installs and compiles successfully.
Now, if I add vector<int> variable and use it somewhere in that Basic_Blink example, it won't compile.
```cpp
#include <SmingCore.h>
#include <list> //I tried without these includes too, same effect
#include <vector> //I tried without these includes too, same effect
#define LED_PIN 2 // GPIO2

Timer procTimer;
bool state = true;

void blink()
{
	digitalWrite(LED_PIN, state);
	state = !state;
}
void init()
{
	pinMode(LED_PIN, OUTPUT);
	procTimer.initializeMs(1000, blink).start();
	
	//here is my addition of two lines:
	std::vector<int> SomeVec;
	SomeVec.push_back(54);

}
```

Here is result after I run make command:
```
Basic_Blink: Invoking 'all' for Esp8266 (debug) architecture
make components application
make[1]: Entering directory '/opt/sming/samples/Basic_Blink'

Building /opt/sming/samples/Basic_Blink/out/Esp8266/debug/lib/clib-App-2433578f105fdac9bc633a786b98e8c3.a
C+ /opt/sming/samples/Basic_Blink/app/application.cpp
AR /opt/sming/samples/Basic_Blink/out/Esp8266/debug/lib/clib-App-2433578f105fdac9bc633a786b98e8c3.a
Validating hardware config 'standard'
Verifying partition table...
Writing to 'out/Esp8266/debug/firmware/partitions.bin'
Basic_Blink: Linking out/Esp8266/debug/build/app_0.out
/opt/esp-quick-toolchain/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/10.2.0/../../../../xtensa-lx106-elf/bin/ld: /opt/sming/samples/Basic_Blink/out/Esp8266/debug/lib/clib-App-2433578f105fdac9bc633a786b98e8c3.a(application.o):(.text._ZNSt6vectorIiSaIiEE17_M_realloc_insertIJiEEEvN9__gnu_cxx17__normal_iteratorIPiS1_EEDpOT_[_ZNSt6vectorIiSaIiEE17_M_realloc_insertIJiEEEvN9__gnu_cxx17__normal_iteratorIPiS1_EEDpOT_]+0x4): undefined reference to `_ZSt20__throw_length_errorPKc'
/opt/esp-quick-toolchain/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/10.2.0/../../../../xtensa-lx106-elf/bin/ld: /opt/sming/samples/Basic_Blink/out/Esp8266/debug/lib/clib-App-2433578f105fdac9bc633a786b98e8c3.a(application.o): in function `_ZNKSt6vectorIiSaIiEE12_M_check_lenEjPKc':
/opt/esp-quick-toolchain/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/10.2.0/bits/stl_vector.h:1760: undefined reference to `_ZSt20__throw_length_errorPKc'
collect2: error: ld returned 1 exit status
make[1]: *** [/opt/sming/Sming/Arch/Esp8266/app.mk:47: out/Esp8266/debug/build/app_0.out] Error 1
make[1]: Leaving directory '/opt/sming/samples/Basic_Blink'
make: *** [/opt/sming/Sming/project.mk:22: all] Error 2

```

And this is result when I replace `std::vector<int>` with `std::list<int>`:
```
Basic_Blink: Invoking 'all' for Esp8266 (debug) architecture
make components application
make[1]: Entering directory '/opt/sming/samples/Basic_Blink'

Building /opt/sming/samples/Basic_Blink/out/Esp8266/debug/lib/clib-App-2433578f105fdac9bc633a786b98e8c3.a
C+ /opt/sming/samples/Basic_Blink/app/application.cpp
AR /opt/sming/samples/Basic_Blink/out/Esp8266/debug/lib/clib-App-2433578f105fdac9bc633a786b98e8c3.a
Validating hardware config 'standard'
Verifying partition table...
Writing to 'out/Esp8266/debug/firmware/partitions.bin'
Basic_Blink: Linking out/Esp8266/debug/build/app_0.out
/opt/esp-quick-toolchain/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/10.2.0/../../../../xtensa-lx106-elf/bin/ld: /opt/sming/samples/Basic_Blink/out/Esp8266/debug/lib/clib-App-2433578f105fdac9bc633a786b98e8c3.a(application.o): in function `_ZZN12OsTimer64ApiI5TimerEC4EvENUlPvE_4_FUNES2_':
/opt/sming/Sming/Core/Timer.h:54: undefined reference to `_ZNSt8__detail15_List_node_base7_M_hookEPS0_'
/opt/esp-quick-toolchain/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/10.2.0/../../../../xtensa-lx106-elf/bin/ld: /opt/sming/samples/Basic_Blink/out/Esp8266/debug/lib/clib-App-2433578f105fdac9bc633a786b98e8c3.a(application.o): in function `_ZN9__gnu_cxx13new_allocatorISt10_List_nodeIiEE9constructIiJiEEEvPT_DpOT0_':
/opt/esp-quick-toolchain/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/10.2.0/ext/new_allocator.h:150: undefined reference to `_ZNSt8__detail15_List_node_base7_M_hookEPS0_'
collect2: error: ld returned 1 exit status
make[1]: *** [/opt/sming/Sming/Arch/Esp8266/app.mk:47: out/Esp8266/debug/build/app_0.out] Error 1
make[1]: Leaving directory '/opt/sming/samples/Basic_Blink'
make: *** [/opt/sming/Sming/project.mk:22: all] Error 2
```

I tried this with both master and develop branches with same result. Am I doing something wrong or is it Sming issue?

mikee47, 2021-05-14:
Sming doesn't enable exception handling for the ESP8266 - it's is explicitly turned off via `-fno-exceptions` to reduce code bloat. To use them would require linking `libstdc++` and probably also some linker map adjustments to mitigate memory usage.

AntonGrushin, 2021-05-14:
Erm, sorry, I don't understand how exception handling is connected to usage of `vector` and `list` arrays. Or do you mean that its needed to find out the source of those compilation errors?

There is a little bit of clue in the errors as it looks
`...vector ... _M_realloc_insert` `... gnu_cxx17__normal_iterator...` `undefined reference to ... throw_length_error`  `in function M_check_len`
Not sure if it helps. I'm afraid my skill is not enough to fix those myself.

Also, I found out that in [lts-master/4.2.x](https://github.com/SmingHub/Sming/tree/lts-master/4.2.x) branch its still working.

mikee47, 2021-05-14:
This may throw a little light on the subject https://stackoverflow.com/questions/4826838/do-standard-library-stl-containers-support-a-form-of-nothrow-allocation

mikee47, 2021-05-14:
Try adding this to your code:

```
namespace std
{
void __attribute__((noreturn)) __throw_length_error(const char*)
{
	abort();
}
}
```

AntonGrushin, 2021-05-14:
It works with your addition, thank you!
Now I understand what you meant with exceptions, appreciate the link with the explanation.
Closing this issue.

closed 2021-05-14 by AntonGrushin

mikee47, 2021-05-15:
Here's another very interesting discussion about using C++ without exceptions https://softwareengineering.stackexchange.com/questions/113479/are-there-any-real-world-cases-for-c-without-exceptions