SmingHub/Sming #2333: Native ESP8266 SDK APIs linking

mostorer, 2021-05-26:
I'm now playing with ESP8266 esp-now api to have modules talk to each other without a connected WiFi.

I added the "espnow.h" to my source code and used the NONOS SDK api. The compilation phase is good but the link process not. For each espnow api used I obtain something like: "**undefined reference to `_Z12esp_now_sendPhS_i'**".

As the SDK is included within Sming platform, I thought all libraries would be linked to sources, but it seems is not so, and I'm not able to find the right way to add the "libespnow.a" to linked libraries.

Looking around I have not been able to find the right suggestion yet.
Is it the use of native SDK api supported in some way ?

slaff, 2021-05-26:
Put the following line in your application's component.mk file:
```
EXTRA_LIBS += espnow
```

mostorer, 2021-05-26:
> You can read more about the build system from [our documentation](https://sming.readthedocs.io/en/latest/_inc/Sming/building.html#envvar-EXTRA_LIBS)

I started from there, but not able to make it working.

> EXTRA_LIBS += espnow

Notihng changes.

The strange thing is that if I use a wrong name for the library, I obtain a different error.
E.g. 
Added: EXTRA_LIBS += xxxespnow
Error: /opt/esp-quick-toolchain/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/10.2.0/../../../../xtensa-lx106-elf/bin/ld: cannot find -lxxxespnow

mikee47, 2021-05-26:
@mostorer The clue is "undefined reference to `_Z12esp_now_sendPhS_i'", the name is mangled. However, the SDK libraries are all standard C. Try this:

```
extern "C" {
#include <espnow.h>
}
```

mostorer, 2021-05-27:
At the end I put:
`
extern "C" {
#include <espnow.h>
}
` within the source code and:

`
EXTRA_LIBS += espnow
` within component.mk

as suggested from @mikee47 and @slaff and it worked.

Thank you so much for your support.

closed 2021-05-27 by mostorer