SmingHub/Sming #2339: Compile errors after clean install - git submodules errors

charlesschaefer, 2021-06-16:
Hi,
I've made a new installation of Sming in my computer and tryed to run the Basic_Blink example, but it can't find the `FlashString` module.

When I run `git submodule update` at the FlashString submodule directory, it then changes the error to the `http_parser` library.

Looks like the Makefiles aren't able to init submodules.

This is the output when I run make (branch develop, after fresh cloning and running `install.sh` ):

```
$ make

Basic_Blink: Invoking 'all' for Esp8266 (debug) architecture
make components application
make[1]: Entering directory '/var/www/projetos/pessoais/wifitracker/sming/Sming/samples/Basic_Blink'

Building /tmp/Sming/Tools/../Sming/out/Esp8266/debug/lib/clib-Storage.a
C+ /tmp/Sming/Sming/Components/Storage/src/ProgMem.cpp
In file included from /tmp/Sming/Sming/Components/Storage/src/include/Storage/Device.h:12,
                 from /tmp/Sming/Sming/Components/Storage/src/include/Storage/CustomDevice.h:7,
                 from /tmp/Sming/Sming/Components/Storage/src/include/Storage/ProgMem.h:12,
                 from /tmp/Sming/Sming/Components/Storage/src/ProgMem.cpp:11:
/tmp/Sming/Sming/Wiring/WString.h:64:10: fatal error: FlashString/String.hpp: No such file or directory
   64 | #include <FlashString/String.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [/tmp/Sming/Tools/../Sming/component-wrapper.mk:170: /tmp/Sming/Sming/Components/Storage/src/ProgMem.o] Error 1
make[1]: *** [/tmp/Sming/Tools/../Sming/project.mk:433: Storage-build] Error 2
make[1]: Leaving directory '/var/www/projetos/pessoais/wifitracker/sming/Sming/samples/Basic_Blink'
make: *** [/tmp/Sming/Tools/../Sming/project.mk:22: all] Error 2
```

Any idea on how to fix that?

slaff, 2021-06-17:
Where is your SMING_HOME env. variable pointing to? I can see the app located at `/var/www/projetos/pessoais/wifitracker/sming/Sming/samples/Basic_Blink` but the compilation takes place in a completely different location `/tmp/Sming/Sming/...`.

charlesschaefer, 2021-06-17:
You're right, I'm sorry. That's because I was in another place when detected the error and got confused.

But to be sure the error is reproducible, I've just cloned and tryed again. Same error. Pasting the result after the first `make` attempt:

```
Basic_Blink: Invoking 'all' for Esp8266 (debug) architecture
Using configuration 'standard'
Patching /tmp/Sming/Sming/Components/.patches/http-parser/component.mk
Patching /tmp/Sming/Sming/Libraries/.patches/ArduinoFFT/component.mk
Patching /tmp/Sming/Sming/Libraries/.patches/Adafruit_ST7735/component.mk
Patching /tmp/Sming/Sming/Libraries/.patches/Adafruit_VL53L0X/component.mk
Patching /tmp/Sming/Sming/Libraries/.patches/Adafruit_SSD1306/component.mk
Patching /tmp/Sming/Sming/Libraries/.patches/IR/component.mk
Patching /tmp/Sming/Sming/Libraries/.patches/Adafruit_BME280_Library/component.mk

Fetching submodule '/tmp/Sming/Tools/../Sming/Components/rboot/rboot' ...
Submodule 'rboot' (https://github.com/mikee47/rboot) registered for path 'rboot'
Cloning into '/tmp/Sming/Sming/Components/rboot/rboot'...
Submodule path 'rboot': checked out '4ad3ba2a8f6d48d8ab90e0e7c67d42cac49f4da3'

Fetching submodule '/tmp/Sming/Tools/../Sming/Components/rboot/esptool2' ...
Submodule 'esptool2' (https://github.com/raburton/esptool2.git) registered for path 'esptool2'
Cloning into '/tmp/Sming/Sming/Components/rboot/esptool2'...
Submodule path 'esptool2': checked out 'a550572b9de8c25f26b7a7675b47e58861227f64'
Checking patch elf.h...
Checking patch esptool2.c...
Checking patch esptool2.h...
Checking patch Makefile...
Applied patch elf.h cleanly.
Applied patch esptool2.c cleanly.
Applied patch esptool2.h cleanly.
Applied patch Makefile cleanly.

Fetching submodule '/tmp/Sming/Tools/../Sming/Arch/Esp8266/Components/esp8266/ESP8266_NONOS_SDK' ...
Submodule 'Esp8266.Sdk' (https://github.com/espressif/ESP8266_NONOS_SDK.git) registered for path 'ESP8266_NONOS_SDK'
Cloning into '/tmp/Sming/Sming/Arch/Esp8266/Components/esp8266/ESP8266_NONOS_SDK'...
Submodule path 'ESP8266_NONOS_SDK': checked out '3fe474e040450e0ab72ea75f8f00c3d9c6279025'
Checking patch lib/libc.a => lib/libc.orig.a...
Checking patch lib/libgcc.a => lib/libgcc.orig.a...
Checking patch include/version.h...
Applied patch lib/libc.a => lib/libc.orig.a cleanly.
Applied patch lib/libgcc.a => lib/libgcc.orig.a cleanly.
Applied patch include/version.h cleanly.

Fetching submodule '/tmp/Sming/Tools/../Sming/Arch/Esp8266/Components/driver/new-pwm' ...
Submodule 'Esp8266.new-pwm' (https://github.com/StefanBruens/ESP8266_new_pwm.git) registered for path 'new-pwm'
Cloning into '/tmp/Sming/Sming/Arch/Esp8266/Components/driver/new-pwm'...
Submodule path 'new-pwm': checked out '19cb69e9ce5071686d0e2a28962fd06e11d03a61'
Checking patch pwm.c...
Applied patch pwm.c cleanly.

Fetching submodule '/tmp/Sming/Tools/../Sming/Components/esptool/esptool' ...
Submodule 'esptool' (https://github.com/espressif/esptool) registered for path 'esptool'
Cloning into '/tmp/Sming/Sming/Components/esptool/esptool'...
Submodule path 'esptool': checked out '4fa0bd7b0d1f69f5ff22b043adc07c5e562a8931'

Fetching submodule '/tmp/Sming/Tools/../Sming/Libraries/Spiffs/spiffs' ...
Submodule 'spiffs' (https://github.com/pellepl/spiffs.git) registered for path 'spiffs'
Cloning into '/tmp/Sming/Sming/Libraries/Spiffs/spiffs'...
Submodule path 'spiffs': checked out 'ec68ba8208d7550860e4e78299d58a529b88bf85'
Checking patch src/spiffs.h...
Checking patch src/spiffs_hydrogen.c...
Checking patch src/spiffs_nucleus.c...
Applied patch src/spiffs.h cleanly.
Applied patch src/spiffs_hydrogen.c cleanly.
Applied patch src/spiffs_nucleus.c cleanly.

Fetching submodule '/tmp/Sming/Tools/../Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip' ...
Submodule 'Esp8266.esp-open-lwip' (https://github.com/pfalcon/esp-open-lwip.git) registered for path 'esp-open-lwip'
Cloning into '/tmp/Sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip'...
[... 32 lines cut ...]
Applied patch lwip/core/dhcp.c cleanly.
Applied patch lwip/core/ipv4/ip_addr.c cleanly.
Applied patch lwip/core/tcp.c cleanly.
Applied patch lwip/app/espconn.c cleanly.
Applied patch lwip/app/espconn_tcp.c cleanly.
Applied patch lwip/core/sntp.c cleanly.
Applied patch include/lwip/sntp.h cleanly.
make components application
make[1]: Entering directory '/tmp/Sming/samples/Basic_Blink'

Building /tmp/Sming/Tools/../Sming/out/Esp8266/debug/lib/clib-Storage-97b8f4f0ec21f2cd96f1fb04faba9736.a
C+ /tmp/Sming/Sming/Components/Storage/src/ProgMem.cpp
In file included from /tmp/Sming/Sming/Components/Storage/src/include/Storage/Device.h:12,
                 from /tmp/Sming/Sming/Components/Storage/src/include/Storage/CustomDevice.h:7,
                 from /tmp/Sming/Sming/Components/Storage/src/include/Storage/ProgMem.h:12,
                 from /tmp/Sming/Sming/Components/Storage/src/ProgMem.cpp:11:
/tmp/Sming/Sming/Wiring/WString.h:64:10: fatal error: FlashString/String.hpp: No such file or directory
   64 | #include <FlashString/String.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [/tmp/Sming/Tools/../Sming/component-wrapper.mk:170: /tmp/Sming/Sming/Components/Storage/src/ProgMem.o] Error 1
make[1]: *** [/tmp/Sming/Tools/../Sming/project.mk:435: Storage-build] Error 2
make[1]: Leaving directory '/tmp/Sming/samples/Basic_Blink'
make: *** [/tmp/Sming/Tools/../Sming/project.mk:22: all] Error 2
```

mikee47, 2021-06-18:
Hello. It looks like the build system has already found the `FlashString` and `http-parser` (and maybe other) libraries elsewhere. To find out, run `make list-components`. Does that help?

charlesschaefer, 2021-06-18:
Hi @mikee47! Thanks for the tip. I don't think this is the problem. I've tryed the `make list-components` command:
```
make list-components 

Basic_Blink: Invoking 'list-components' for Esp8266 (debug) architecture
ALL_SEARCH_DIRS:
- /tmp/Sming/Tools/../Sming/Arch/Esp8266/Components
- /tmp/Sming/Tools/../Sming/Components
- /tmp/Sming/Tools/../Sming/Libraries
Components:
> App: /tmp/Sming/samples/Basic_Blink
> FlashString: /tmp/Sming/Tools/../Sming/Components/FlashString
> IFS: /tmp/Sming/Tools/../Sming/Components/IFS
> Network: /tmp/Sming/Tools/../Sming/Components/Network
> Sming: /tmp/Sming/Tools/../Sming
> Spiffs: /tmp/Sming/Tools/../Sming/Libraries/Spiffs
> Storage: /tmp/Sming/Tools/../Sming/Components/Storage
> arch_driver: /tmp/Sming/Tools/../Sming/Components/arch_driver
> crypto: /tmp/Sming/Tools/../Sming/Components/crypto
> driver: /tmp/Sming/Tools/../Sming/Arch/Esp8266/Components/driver
> esp-open-lwip: /tmp/Sming/Tools/../Sming/Arch/Esp8266/Components/esp-open-lwip
> esp8266: /tmp/Sming/Tools/../Sming/Arch/Esp8266/Components/esp8266
> esp_wifi: /tmp/Sming/Tools/../Sming/Arch/Esp8266/Components/esp_wifi
> esptool: /tmp/Sming/Tools/../Sming/Components/esptool
> fatfs: /tmp/Sming/Tools/../Sming/Arch/Esp8266/Components/fatfs
> gdbstub: /tmp/Sming/Tools/../Sming/Arch/Esp8266/Components/gdbstub
> heap: /tmp/Sming/Tools/../Sming/Arch/Esp8266/Components/heap
> http-parser: /tmp/Sming/Tools/../Sming/Components/http-parser
> libb64: /tmp/Sming/Tools/../Sming/Components/libb64
> libc: /tmp/Sming/Tools/../Sming/Arch/Esp8266/Components/libc
> libyuarel: /tmp/Sming/Tools/../Sming/Components/libyuarel
> mqtt-codec: /tmp/Sming/Tools/../Sming/Components/mqtt-codec
> rboot: /tmp/Sming/Tools/../Sming/Components/rboot
> sming-arch: /tmp/Sming/Tools/../Sming/Arch/Esp8266/Components/sming-arch
> spi_flash: /tmp/Sming/Tools/../Sming/Arch/Esp8266/Components/spi_flash
> ssl: /tmp/Sming/Tools/../Sming/Components/ssl
> terminal: /tmp/Sming/Tools/../Sming/Components/terminal
> ws_parser: /tmp/Sming/Tools/../Sming/Components/ws_parser
make: Nothing to be done for 'list-components'.
```

Note that the FlashString is pointing to the correct directory (I ran the command after the first `make` attempt), but the directory is empty:
```
ls -al /tmp/Sming/Tools/../Sming/Components/FlashString
total 8
drwxrwxr-x  2 charles charles 4096 jun 18 09:36 .
drwxrwxr-x 25 charles charles 4096 jun 18 09:36 ..
```

Today I've worked a bit more on this and after running `make submodules` inside `Sming/Sming` directory everything was installed correctly, and then I was able to build the samples again.

But there's something weird that isn't allowing the submodules to be initialized during the first `make` call. 

I'm able to keep my work evolving here. But my concerns are about newcomers that won't be able to build after a fresh install :-(

I'll try in a clean environment to check if this problem is just mine or something that changed and messed the build system.

mikee47, 2021-06-18:
I'm not sure why you're using `/tmp/Sming/Tools/,,/Sming`, but if you normalise it to `/tmp/Sming/Sming` things will work...

mikee47, 2021-06-18:
There is a simple fix in the build system for this, but there could be other implications for using a 'dirty' path:

```
diff --git a/Sming/build.mk b/Sming/build.mk
index 9ad74cba..0d944378 100644
--- a/Sming/build.mk
+++ b/Sming/build.mk
@@ -71,7 +71,7 @@ else
 FixPath                        = $1
 endif

-SMING_HOME             := $(patsubst %/,%,$(call FixPath,$(SMING_HOME)))
+SMING_HOME             := $(patsubst %/,%,$(call FixPath,$(abspath $(SMING_HOME))))

 ifeq (,$(wildcard $(SMING_HOME)/Arch/$(SMING_ARCH)/build.mk))
 $(error Arch '$(SMING_ARCH)' not found)
```

charlesschaefer, 2021-06-22:
> I'm not sure why you're using `/tmp/Sming/Tools/../Sming`, but if you normalise it to `/tmp/Sming/Sming` things will work...

@mikee47 I'm using this as a workaround, because my `export.sh` was always exporting `SMING_HOME=/Sming`. Only now that you pointed the problem with the dirty path that I noticed that `export.sh` is using `$BASH_SOURCE`, which isn't supportted by `zsh` (my default shell).

Because of this error, to not hardcode the Sming path, I've inserted this inside `export.sh`:
```
export SMING_HOME=`pwd`/../Sming
```

Now I tried to `make` in a fresh install with the correct path and it worked as expected! Thanks for your help!

To help others that use `zsh` I made a PR that corrects the `export.sh` file to use the zsh alternative to `$BASH_SOURCE`.

closed 2021-06-22 by charlesschaefer