SmingHub/Sming #2359: ESP32: No Ethernet implementation in Sming

Asanga-Viraj, 2021-08-18:
Hi,

I am working on an ESP32 project which needs ethernet. I managed to work with esp-idf_V4.1. 
I like to work with Sming because it is well structured and easy to use. I did it with ESP8266. When reading examples, I didn't see an example related to ethernet. Also, I tried to add ethernet config to a Sming sample project and got compiling error. It seems to be ethernet is not included in the Sming project. 
When will it go to implement in Sming?

Thanks,
Asanga

mikee47, 2021-08-31:
@Asanga-Viraj I've started work on Ethernet support, see #2361. There are a lot of possible ethernet configurations so if there's a specific combination you're interested we'll try to get that implemented.

Asanga-Viraj, 2021-09-01:
@mikee47 Thank you for the reply. ESP32 Ethernet current implementations in idf_V4.1 example such as internal mac and some SPI chips  `LAN8720`, `IP101`, `DP83848` and `RTL8201`is OK for now. My current project is interested in internal mac. If I got another chip, I will mention it here. 
Thanks.

slaff, 2021-09-08:
@Asanga-Viraj Were you able to use Ethernet with the latest additions in Sming?

Asanga-Viraj, 2021-09-09:
Actually I modified the Sming to work with Ethernet. 

- I created a file and I added the file "Kconfig.projbuild" file in resource folder to %(SMING_HOME)/Sming/Arch/Esp32/Components/esp32/project/main. 

- Then I changes file "startup.cpp" in resource folder to %(SMING_HOME)/Sming/Arch/Esp32/Components/esp32/src. 

- Added this line "esp_eth \ " into the "SDK_COMPONENTS" after "esp_common \ " in components.mk file located in %(SMING_HOME)/Sming/Arch/Esp32/Components/esp32
- `make sdk clean`
  `make sdk fullclean`
   `make sdk-menuconfig    #then exit`
  `make sdk-menuconfig`

- delete folder %(SMING_HOME)/Sming/out/Esp32/debug/lib

-  `make sdk all`

 If you can help me to attach the relevant files in this thread, I will do it.

mikee47, 2021-09-09:
@Asanga-Viraj With the current develop branch of Sming you can find the [Basic_Ethernet](https://github.com/SmingHub/Sming/tree/develop/samples/Basic_Ethernet) sample which shows how to connect the embedded MAC + LAN8720 PHY. If you haven't yet updated to IDF 4.3, see https://sming.readthedocs.io/en/latest/_inc/Sming/Arch/Esp32/README.html.

To avoid any IDF-specific dependencies in applications I've created a C++ wrapper for the ethernet. See https://github.com/SmingHub/Sming/tree/develop/Sming/Components/Network/Arch/Esp32/Platform/include/Platform. Comments in the class header should be helpful. This is a new interface so please if you have any suggestions for changes or improvements let me know!

Asanga-Viraj, 2021-09-10:
@mikee47 Thank you for the upgrade. I will let you know.

slaff, 2021-09-22:
Feel free to re-open if you have issues with the Ethernet for ESP32 implementation.

closed 2021-09-22 by slaff