SmingHub/Sming #2366: Garbage on the console

jensahylda, 2021-09-15:
Hi,

I think this is a newbie question, but let me try anyway.

I am using a custom board with Esp-wroom-02 (esp8266) and flashsize of 4MB and Smiing stable 4.3.0

I can compile and run Basic_Blink and also output to the console. Also it can run older Sming frameworks perfectly, also with flash access. So hardware is fine.

But as soon as i start using the flash with , for example Flash_Storage or HttpServer_ConfigNetwork something fails.

Like this:
/usr/bin/python3 -m serial.tools.miniterm --raw --encoding ascii /dev/ttyUSB0 115200
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
rl��r��#�n���
             �
              ��p�<����8��ǒ��lp

                               �nn��;�n����

                                           b�$	rr�������n��8�
--- exit ---

The old sming framework is working fine, also with flash. Any hints?

Regards Jens

slaff, 2021-09-16:
@jensahylda Try to flashinit the device and try again. See more details here: https://sming.readthedocs.io/en/latest/troubleshooting/random-restart.html. Init-ing might fix the issue that you are experiencing.  Which toolchain are you using for ESP8266 ?

jensahylda, 2021-09-16:
> @jensahylda Try to flashinit the device and try again. See more details here: https://sming.readthedocs.io/en/latest/troubleshooting/random-restart.html. Init-ing might fix the issue that you are experiencing. Which toolchain are you using for ESP8266 ?

Thank you very much.

I am using Ubuntu 20.04 in a shell. make flashinit  followed by make flash does not help. No difference.

However i changed the baudrate, and this is the real text that is output:
--- Miniterm on /dev/ttyUSB0  74880,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

ets_main.c 

Thanks again.

mikee47, 2021-09-16:
@jensahylda It's odd that the bootloader isn't even getting to run.

Are you sure your device has 4M flash, not 2M? Try the `Basic_Blink` sample again with a 4M flash setting:

```
make HWCONFIG=standard-4m
make flash
```

For reference, you should see something like this:

```
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 1220, room 16

tail 4
chksum 0xda
load 0x3ffe8000, len 640, room 4
tail 12
chksum 0xa8
csum 0xa8

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   DIO
Flash Speed:  80 MHz

Found 'rom0' @ 0x00002000, size 0x000f8000, subtype 0x00
Booting rom 0 at 2000, load addr 40930.
ota1 not set
ota2 not set
V2
Mo
,�rf cal sector: 1019
freq trace enable 0
rf[112] ���T�UZ�ZJPK�+V
```

mikee47, 2021-09-16:
NB. You may also need to do this before building:

```
make clean components-clean
```

jensahylda, 2021-09-16:
> NB. You may also need to do this before building:
> 
> ```
> make clean components-clean
> ```

Bingo, that solved the problem. Thank you very much. :-)

jensahylda, 2021-09-16:
Sorry, the program runs, but the flash is not working properly.

I tried your suggestion with Basic_Blink and got this:
rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   32 Mbit

Can i set another size like 2M, your suggestion sets 4M.

The original Basic_Storage example gives this output:

Registered storage devices:
  name = 'spiFlash', type = flash, size = 0x400000, ID = 0x1540c8

** Enumerate registered SPIFFS partitions
115298 Partition 'user0' not found
118282 [Partition] Invalid
120663 Error reading from partition ''

Writing some stuff to partition...
127437 [Partition] Invalid
129886 [Partition] Invalid
132264 [Partition] Invalid
134697 Error reading from partition ''

Thank you again.

jensahylda, 2021-09-16:
Another comment:
I have to run 
make HWCONFIG=standard-4m

That probably overrides basic_storage.hw - and thats why the flash is not partitioned correct for the code

So how do it make it with basic_storage.hw but as standard-4m - sorry, but i cannot see the link yet.

Regards Jens

mikee47, 2021-09-16:
If you run `make config-clean` that will reset the working configuration back to the defaults for that sample. Then you can run `make flash` again.

See https://sming.readthedocs.io/en/latest/_inc/Sming/Components/Storage/index.html?highlight=HWCONFIG#partition-maps for details about partition maps.

mikee47, 2021-09-16:
Also note the current develop branch contains many improvements so always worth a try if you run into problems.

mikee47, 2021-09-16:
> Can i set another size like 2M, your suggestion sets 4M.

You'll need to create a [custom hardware configuration](https://sming.readthedocs.io/en/latest/_inc/Sming/Components/Storage/index.html?highlight=HWCONFIG#custom-configurations).

mikee47, 2021-09-16:
There's also a graphical editor, `make hwconfig-edit`.

jensahylda, 2021-09-16:
Ok, tried the development branch, same result, let me explain what i did, i must miss something.
- git clone https://github.com/SmingHub/Sming /opt/sming
- source /opt/sming/Tools/install.sh all
- source /opt/sming/Tools/export.sh
- cd /opt/sming/samples/Basic_Storage
- make
- make flashinit
- make flash

I get garbage, on the output, and start terminal
python -m serial.tools.miniterm /dev/ttyUSB0 74880

--- Miniterm on /dev/ttyUSB0  74880,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
 ets Jan  8 2013,rst cause:1, boot mode:(3,7)
ets_main.c 

If i do a 
make HWCONFIG=standard-4m
and reflash, then the program starts, but obviously the partitions are bad.

Basic_Blink works fine, no problem, but that is not using the flash. I need to use the flash :-)

What step do i miss here?

Regards Jens

mikee47, 2021-09-16:
From the `Basic_Storage` sample, do a full clean, rebuild, full flash wipe then flash application:

```
make config-clean clean components-clean
make
make flashinit
make flash
```

jensahylda, 2021-09-16:
> From the `Basic_Storage` sample, do a full clean, rebuild, full flash wipe then flash application:
> 
> ```
> make config-clean clean components-clean
> make
> make flashinit
> make flash
> ```

Same thing, doing these 4 steps, and this is what i get:
--- Miniterm on /dev/ttyUSB0  74880,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

ets_main.c 

Actually what i need is just a filesystem with 4 files in it, one being writable, maybe this example is too complex should i go for something simpler...

Regards Jens

mikee47, 2021-09-17:
When `ets_main.c` appears as above it indicates that the boot sector hasn't been programmed, and is still erased. That indicates the `make flash` hasn't done its job. To diagnose this if you paste your entire build and flash log it should identify where things are going wrong.

jensahylda, 2021-09-17:
> When `ets_main.c` appears as above it indicates that the boot sector hasn't been programmed, and is still erased. That indicates the `make flash` hasn't done its job. To diagnose this if you paste your entire build and flash log it should identify where things are going wrong.

Thank you for fast and qualified help.

I have attached the complete log for all points you mentioned (from cleaning to flashing)

[basic_storage-makeandbuild.txt](https://github.com/SmingHub/Sming/files/7183663/basic_storage-makeandbuild.txt)

I hope you can see something... 

Regards Jens

mikee47, 2021-09-17:
The log looks perfect... however the boot sector clearly isn't getting programmed. I think the flash speed is too high: If you take a look in `basic_storage.hw` and comment out line 14:

```
diff --git a/samples/Basic_Storage/basic_storage.hw b/samples/Basic_Storage/basic_storage.hw
index 4d28b7fa2..14e341926 100644
--- a/samples/Basic_Storage/basic_storage.hw
+++ b/samples/Basic_Storage/basic_storage.hw
@@ -14 +14 @@
-			"speed": 80
+			// "speed": 80
```

This will revert to the default flash speed of 40M.

jensahylda, 2021-09-17:
Reverted default speed to 40M and verified in the flash command its changed, and did the four points again.

/usr/bin/python3 /opt/sming/Sming/Components/esptool/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 --chip esp8266 --before default_reset --after hard_reset write_flash -z -fs 4MB -ff **40m** .....

The result is unchanged. Really weird...

--- Miniterm on /dev/ttyUSB0  74880,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

ets_main.c

slaff, 2021-09-17:
@jensahylda can you paste your current hardware config? 

```
cd {sample-dir}/
make hwconfig
```

After that dump your flash memory to a file, zip it and attach it here. You can dump the whole flash with the following command:

```
/usr/bin/python3 /opt/sming/Sming/Components/esptool/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 --chip esp8266 read_flash 0x00000 0x400000 /tmp/sming-esp8266-4m.dump
```

mikee47, 2021-09-17:
@jensahylda Can you try a simple `make flashboot` and see what happens?

jensahylda, 2021-09-17:
> @jensahylda Can you try a simple `make flashboot` and see what happens?

If i do that, it programs, then it shows this with the terminal afterwards:
--- Miniterm on /dev/ttyUSB0  74880,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 1220, room 16 
tail 4
chksum 0xda
load 0x3ffe8000, len 640, room 4 
tail 12
chksum 0xa8
csum 0xa8

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   DIO
Flash Speed:  40 MHz

Resetting boot config.
Found 'rom0' @ 0x00002000, size 0x000f8000, subtype 0x00
Rom 0 at 2000 is bad.
No good rom available.
user code done

mikee47, 2021-09-17:
OK. If you run `make flashapp` then it should run the application. I have a suspicion that your device has only 2M flash. If that is the case, then accessing locations above that will wrap around. Running `make flash` will write each partition in turn, starting with the boot sector, so the start of the `spiffs0` partition at 0x200000 will overwrite the boot sector at 0x000000.

If you run `python $SMING_HOME/Components/esptool/esptool/esptool.py -p /dev/ttyUSB0 flash_id` that should confirm exactly what flash chip you have, and its size.

jensahylda, 2021-09-17:
> @jensahylda can you paste your current hardware config?
>
> ```
> cd {sample-dir}/
> make hwconfig
> ```
>
> After that dump your flash memory to a file, zip it and attach it here. You can dump the whole flash [...]

Thank you, please find both in attached zip.

[hwconfig.zip](https://github.com/SmingHub/Sming/files/7187476/hwconfig.zip)

jensahylda, 2021-09-17:
> OK. If you run `make flashapp` then it should run the application. I have a suspicion that your device has only 2M flash. If that is the case, then accessing locations above that will wrap around. [...]

You are a genious :-) Its correct, below is a dump, and the flash is 2M. A little embarrasing i think, but i actually was sure it was 4M.

What is the best way to modify this example (Basic_Storage) to 2M? Set the flashsize to 2M in basic_storage.hw under device, and then modify the partitions so they fit in? Anything else?

esptool.py v3.2-dev
Serial port /dev/ttyUSB0
Connecting.....................
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 2c:3a:e8:35:09:41
Uploading stub...
Running stub...
Stub running...
Manufacturer: c8
Device: 4015
Detected flash size: 2MB
Hard resetting via RTS pin...

mikee47, 2021-09-17:
It would be lovely if manufacturers could print the flash size on the device! I'd always assumed that esptool did a check for valid memory ranges: now we know it doesn't!

To get the filing system working I think it `Basic_WebSkeletonApp` would be a simpler starting point. Create a file called `spiffs2m.hw` in the project root directory and paste this in:

```
{
    "name": "SPIFFS with 2M flash",
    "base_config": "spiffs",
    "devices": {
        "spiFlash": {
            "size": "2M"
        }
    },
    "partitions": {
        "spiffs0": {
            // Change the default location so it fits inside 2M flash
            "address": "0x120000",
            "size": "512K"
        }
    }
}
```

You can then `make flash HWCONFIG=spiffs2m` and it should work.

To get the `Basic_Storage` sample working would require setting the flash size of course, but shifting around the partitions as well which is more work.

mikee47, 2021-09-17:
Try this with `Basic_Storage`. Save it as `basic_storage_2m.hw`:

```
{
    "name": "Basic Storage sample (2M)",
    "base_config": "basic_storage",
    "devices": {
        "spiFlash": {
            "size": "2M"
        }
    },
    "partitions": {
        "user0": {
            "address": "0x000fa000"
        },
        "user1": {
            "address": "0x000fe000"
        },
        "spiffs0": {
            "address": "0x00102000"
        },
        "spiffs1": {
            "address": "0x00182000",
            "size": "240K"
        },
        "spiffs2": {
            "address": "0x001be000",
            "size": "216K"
        }
    }
}
```

`make flash HWCONFIG=basic_storage_2m`

jensahylda, 2021-09-17:
Thanks a lot, it works now. And i can open, read and write files.

I hope you learned something, i learned a lot. Thank you for fast and qualified help.

Regards Jens

mikee47, 2021-09-18:
@jensahylda You are most welcome, and framework is improved by this so thank you. Happy Sming-ing!

closed 2021-09-22 by slaff