SmingHub/Sming #2380: Installing on linux aborts and log me out prematurely

icodk, 2021-10-04:
Followed the installation instructions, when entering:  `source /opt/sming/Tools/install.sh esp8266`
The script start and the output looks like:
```
Hit:1 http://dk.archive.ubuntu.com/ubuntu hirsute InRelease
Hit:2 http://dk.archive.ubuntu.com/ubuntu hirsute-updates InRelease
Hit:3 http://dk.archive.ubuntu.com/ubuntu hirsute-backports InRelease
Hit:4 http://dk.archive.ubuntu.com/ubuntu hirsute-security InRelease
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
```
Then the putty session is terminate probably prematurely because I cant compile the Basic_Blink example afterword
```

:/opt/sming/samples/Basic_Blink$ make
Makefile:6: *** SMING_HOME is not set: please configure it as an environment variable.  Stop.
```

etmmahi, 2021-10-04:
Hi,

I have experience the same issue. 
I was able to solve it by changing the clang-format-8 in the tools/install.sh script to clang-format-12

slaff, 2021-10-04:
At the end of successful installation you should get the following message:
```
Installation complete

You may need to set environment variables:
  source $SMING_HOME/../Tools/export.sh
```

Did you set the environment variables? Did you make sure to persist them?

mikee47, 2021-10-04:
@icodk The installer can be run without `source` (i.e. just  `/opt/sming/Tools/install.sh esp8266`) which will avoid terminating the shell. You can then run `source /opt/sming/Tools/export.sh` to configure environment variables.

icodk, 2021-10-04:
@etmmahi 
Your solution WORKS !

@slaff  & @mikee47  May be to update the install script to use clang-format-12 instead of clang-format-8 ?
This solved my problem

closed 2021-10-04 by icodk

mikee47, 2021-10-04:
@icodk Be aware that clang-format 12 produces different results to version 8! We moved to version 8 in March (see #2268) as it is more widely supported. What's your OS version? (`cat /etc/os-release`)

icodk, 2021-10-04:
What do you mean by different results (8 vs 12) or what is good and what is bad ?
I run the installation again without source `/opt/sming/Tools/install.sh esp8266` as @mikee47 suggested and got the following error:
```
ico@ubuntu21:~$ /opt/sming/Tools/install.sh esp8266
Hit:1 http://dk.archive.ubuntu.com/ubuntu hirsute InRelease
Hit:2 http://dk.archive.ubuntu.com/ubuntu hirsute-updates InRelease
Hit:3 http://dk.archive.ubuntu.com/ubuntu hirsute-backports InRelease
Hit:4 http://dk.archive.ubuntu.com/ubuntu hirsute-security InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package clang-format-8
update-alternatives: error: alternative path /usr/bin/clang-format-8 doesn't exist
ico@ubuntu21:~$

```
My OS version:
`cat /etc/os-release`
```
NAME="Ubuntu"
VERSION="21.04 (Hirsute Hippo)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 21.04"
VERSION_ID="21.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=hirsute
UBUNTU_CODENAME=hirsute

```

mikee47, 2021-10-04:
> What do you mean by different results (8 vs 12) or what is good and what is bad ?

clang-format is used to enforce a consistent [coding style](https://sming.readthedocs.io/en/latest/information/develop/coding-style.html) across the framework. This ensures that code changes are not obscured by differences in layout style.

Unfortuately, different versions of clang-format (usually) give different results for the same connfiguration. There's a discussion of exactly this issue here https://stackoverflow.com/questions/59395507/how-to-maintain-a-clang-format-file-for-different-clang-format-versions.

Of course with each new distribution upgrade older versions of clang-format get removed from the standard repositories. So here, going from Ubuntu 20 -> 21 they've dropped cf version 8.

This probably doesn't matter to most Sming users, so perhaps the simplest solution is to skip installation of clang-format if version 8 isn't available. with a warning. @slaff any thoughts?

icodk, 2021-10-04:
May be to keep it only for the actual framework development and let the framework users (like me) do whatever they want

slaff, 2021-10-04:
> This probably doesn't matter to most Sming users, so perhaps the simplest solution is to skip installation of clang-format if version 8 isn't available. with a warning. @slaff any thoughts?

Yes, this can be a solution.