SmingHub/Sming #2437: Esp32 SPI 8 bit (single byte) transfer

kmihaylov, 2021-12-03:
Hello. Esp32's ```SPI.transfer()``` produces 16 clock cycles. ```SPI.transfer16()``` produces 24 clock cycles.

ESP8266's ```SPI.transfer()``` makes only 8 clock cycles.

Here is example:
```
#include <SmingCore.h>
#include <SPI.h>

#define SPI_MISO 12
#define SPI_MOSI 14
#define SPI_CLK 27

//struct SpiPins spispi = {SPI_CLK, SPI_MISO, SPI_MOSI, 255};

Timer spiTimer;

void spiLoop() {
	SPI.beginTransaction(SPISettings(8000000, MSBFIRST, SPI_MODE0));
	//digitalWrite(csPin, LOW);

	//uint8_t msb = SPI.transfer32(2, 8);
	uint8_t msb = SPI.transfer(0x01);

	//digitalWrite(csPin, HIGH);
	SPI.endTransaction();
}

void init()
{
	//SPI.setup(SpiBus::HSPI, spispi);
	SPI.begin();
	spiTimer.initializeMs(250, spiLoop).start();
}
```

With this behavior my slave device would read 2 bytes.
Same with ```SPI.transfer32(1, 8)```.

Did I misconfigure the SPI or ```SPI.transfer()``` should send exacly 8 bits?

kmihaylov, 2021-12-04:
OK, i've added
```
	spi_ll_set_dummy(&dev, 0);
	spi_ll_set_command_bitlen(&dev, 0);
	spi_ll_set_addr_bitlen(&dev, 0);
```
in ```Sming/Arch/Esp32/Core/SPI.cpp``` and there's some progress (I'll test extensively later). ```spi_ll_set_dummy()``` didn't seem to make difference. Then I've added cmd and addr bitlen and the clock cycles now equal the bits set in ```SPI.transfer32(1,7)```.

mikee47, 2021-12-04:
That makes sense. Well spotted! The assumption would be that `spi_ll_master_init` would initialise these to some value but on checking the code it doesn't. These lines should probably be added after the initial `spi_ll_master_init` call.

mikee47, 2021-12-04:
@kmihaylov Would you raise a PR with a fix for this?

kmihaylov, 2021-12-04:
Sure @mikee47 . Let me do it tonight.
Would you comment whether these should be configurable during SPI.setup()? Are you aware if they should be preserved for any Arduino compatibility or etc?

mikee47, 2021-12-04:
No, the SPI class is very basic and doesn't accommodate such hardware features. Everything is done using data transfers.

NB. The `HardwareSPI` library makes better use of the hardware if you're interested.

kmihaylov, 2021-12-05:
With the above mentioned functions SPI.transfer32() works fine, but SPI.read8() doesn't.

```
uint8_t msb = SPI.transfer32(0xff, 8);
//uint8_t msb = SPI.read8();
```
Actually SPI.read8() makes 1 clock cycle and the MOSI line is low during the entire transaction.
Ideas?
Once this is solved I'll also try the other functions before the commit.

mikee47, 2021-12-05:
The bit length hasn't been set up. I've pushed a fix, should work!

slaff, 2021-12-06:
@kmihaylov Check the latest develop and close this issue if the problem has been fixed.

closed 2021-12-06 by kmihaylov