SmingHub/Sming #2445: ESP32 HadwareSerial Callbacks has variable time delay

Asanga-Viraj, 2021-12-10:
Hi,

My application is to implement a Serial2 handler to run serial protocol like ModBus on ESP32. My another microcontroller is polling for reply from ESP32 with baud rate 38400. The time between two packets is 25ms. I attached data received and transmit events like below.

```
HardwareSerial Serial2(2);

Serial2.setRxBufferSize(100);
Serial2.setTxBufferSize(100);
Serial2.setTxWait(true);
Serial2.onDataReceived(onDataCallback);
Serial2.onTransmitComplete(onTxCompleteDelegate);
Serial2.begin(38400);
```

I experienced that the receive complete event is not calling at the end of the receiving. I hooked up a oscilloscope and check the rx data line and receive event calling by toggling digital pin inside the event function. There is a variable delay from the end of the rx until calling the onDataCallback event. Also this issue is on transmit complete event as well.

I tested this on ESP8266. It doesn't have a variable delay between end of rx and receive complete event. In ESP8266, the event calling is working perfectly.

One more thing. If I connected ESP32 RX2 with my other controller polling and initialise the Serial2 without 'Serial2.onDataReceived(onDataCallback);', ESP32 resets due to watch dog reset.

Thanks.

Asanga-Viraj, 2021-12-10:
The I went deep inside of the Sming framework. I think I found the issue. In the file Sming/Core/HardwareSrial.cpp line 184

```
	// If required, queue a callback
	if((status & serial->statusMask) != 0 && !serial->callbackQueued) {
		System.queueCallback(staticOnStatusChange, serial);
		serial->callbackQueued = true;
	}
```
The system quecallback takes variable time to call it. This calling time delay is too much for some protocol implementation like ModBus.

teicors, 2021-12-10:
If I'm not out of topic, I'm having a similar behavior with ModBus and a basic html page: when I "wget" the jquery.js.gz from spiffs, I'm having this result

 jquery.js.gz                          100%[=============================>]  29,45K  5,04KB/s    in 5,8s

and this from the debug window
```
34209686 TCP 3fff2b28 onReadyToSendData: 2
34210643 TCP 3fff2b28 Written: 1024, Available: 1390, isFinished: 0, PushCount: 1
34212718 TCP 3fff2b28 Written: 366, Available: 366, isFinished: 0, PushCount: 2
34220475 TCP 3fff2b28 sent: 1390
34222476 TCP 3fff2b28 onReadyToSendData: 2
34236003 TCP 3fff2b28 Written: 1024, Available: 1390, isFinished: 0, PushCount: 1
34236938 TCP 3fff2b28 Written: 366, Available: 366, isFinished: 0, PushCount: 2
34241910 TCP 3fff2b28 sent: 1390
34243493 TCP 3fff2b28 onReadyToSendData: 2
34247954 TCP 3fff2b28 Written: 1024, Available: 1390, isFinished: 0, PushCount: 1
34254714 TCP 3fff2b28 Written: 366, Available: 366, isFinished: 0, PushCount: 2
34262567 TCP 3fff2b28 sent: 1390
```
I'm using an ESP8266 with ModBus connected to the first serial and the debug (as in the example) at the second serial. The code is a simple join of the modbus example and a spiffs area where are placed an index.html, jquery.js.gz and bootstrap.css.gz
I thought an error in my code, but looking at this message, I think is not my fault.

With another my project non using ModBus,I catch this result with the wget
jquery.js                           100%[===========================================>]  84,29K   274KB/s    in 0,3s

mikee47, 2021-12-10:
I found a bug! Try this patch and see if it fixes your problem. Run `make Sming-build` to pick up the change.

```
diff --git a/Sming/Arch/Esp32/Components/driver/uart.cpp b/Sming/Arch/Esp32/Components/driver/uart.cpp
index b6cd59f0c..78a14e48c 100644
--- a/Sming/Arch/Esp32/Components/driver/uart.cpp
+++ b/Sming/Arch/Esp32/Components/driver/uart.cpp
@@ -236,7 +236,7 @@ void IRAM_ATTR uart_isr(smg_uart_instance_t* inst)
 			if(usis.rxfifo_ovf) {
 				uart_ll_disable_intr_mask(dev, UART_INTR_RXFIFO_OVF);
 			} else if(read == 0) {
-				uart_ll_ena_intr_mask(dev, UART_INTR_RXFIFO_FULL | UART_INTR_RXFIFO_TOUT);
+				uart_ll_disable_intr_mask(dev, UART_INTR_RXFIFO_FULL | UART_INTR_RXFIFO_TOUT);
 			}
 		}
 
```

mikee47, 2021-12-10:
If you need tight timing on the callback (e.g. switching modbus transceiver from TX to RX) then use interrupt callbacks by setting `HardwareSerial::setUartCallback`. The callback might look something like this:

```
void IRAM_ATTR mySerialCallbackHandler(smg_uart_t* uart, uint32_t status)
{
  // Revert back to receive when transmit completes
  if(status & UART_STATUS_TXFIFO_EMPTY) {
    DigitalWrite(MODBUS_DIR, IN);
  }

  // Handle receive events
  if(status & (UART_STATUS_RXFIFO_FULL | UART_STATUS_RXFIFO_TOUT | UART_STATUS_RXFIFO_OVF)) {
    System.queueCallback(handleReceiveEvent);
  }
}
```

mikee47, 2021-12-10:
Also worth doing a search for 'modbus' in previous issues, there's been some discussion about this issue.

slaff, 2021-12-10:
> I found a bug! 

@mikee47 Can you send a PR with the fix?

Asanga-Viraj, 2021-12-13:
Thanks everyone for the support.

I replaced default Sming uart call back using  `HardwareSerial::setUartCallback(mySerialCallbackHandler)`. Still got the same variable delay (sometimes more than 30ms) when calling of `System.queueCallback(handleReceiveEvent);` handleReceiveEvent. 
The same delay issues gets because the function `handleReceiveEvent` is calling over System que task. Then I replaced system que with timer calling `eventTimer.initializeUs(400, handleReceiveEvent).startOnce();`. The serial ModBus replies as I expected. 

According to my application, the ESP32 serve a web page over WiFi STA mode. When ModBus replies over timer event, I did browsing web page over LAN. The ESP resets and shows like below. If I turned of ModBus polling, ESP doesn't reset.

```
1751186 '/ajax/hrbeat' registered
1751287 '/favicon.ico' registered
1753675 '*' registered
1756009 === STA WEB SERVER STARTED ===
1759182 10.150.0.189
1761095 ==========================

I (2188) esp_netif_handlers: sta ip: 10.150.0.189, mask: 255.255.254.0, gw: 10.150.0.254
Re-enable cpu cache.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40148dc4: 0007fffd 16004136 22881012
Core  1 register dump:
PC      : 0x40148dca  PS      : 0x00060031  A0      : 0x80086eb4  A1      : 0x3ffb0b80
A2      : 0x3ffcc804  A3      : 0x3ffb4f90  A4      : 0x3ffcca2b  A5      : 0x000000ff
A6      : 0x00000000  A7      : 0x3ffb427c  A8      : 0x80085d42  A9      : 0x3ffb4e8a
A10     : 0x00000001  A11     : 0x00000001  A12     : 0xfffffff7  A13     : 0x3ffc36c0
A14     : 0x00000020  A15     : 0x00000004  SAR     : 0x00000018  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

Backtrace:0x40148dc7:0x3ffb0b80 0x40086eb1:0x3ffb0ba0 0x40081d39:0x3ffb0c50 0x4008c1d9:0x3ffc36c0 0x4008bff0:0x3ffc36e0 0x4009382d:0x3ffc3700 0x40082f89:0x3ffc3780 0x4008313d:0x3ffc37c0 0x40148884:0x3ffc37e0 0x40139ecd:0x3ffc3840 0x4013a0fe:0x3ffc3860 0x4013b035:0x3ffc3890 0x4014101d:0x3ffc38b0 0x4013c115:0x3ffc38e0 0x4013e030:0x3ffc3910 0x4013e908:0x3ffc3950 0x4013b0ef:0x3ffc3980 0x40152f77:0x3ffc39d0 0x401384f2:0x3ffc39f0 0x4014441f:0x3ffc3a10 0x40144977:0x3ffc3e30 0x40144c9e:0x3ffc3e50 0x40145f65:0x3ffc3e80 0x401440e6:0x3ffc3eb0 0x4014410a:0x3ffc3ed0 0x40144cff:0x3ffc3ef0 0x40144771:0x3ffc3f10 0x4014478e:0x3ffc3f30 0x40122259:0x3ffc3f50 0x40126a0e:0x3ffc3fc0 0x4012a7da:0x3ffc3ff0 0x4011cb75:0x3ffc4010

ELF file SHA256: 00eaac90ebe64985

Entering gdb stub now.
$T04#b8
```

I think (not sure), the problem is not in the UART ISR, it is in the System queues. System queue takes more than 30ms to call a CallBack function and it is varying even we have fast CPU on the board.

slaff, 2021-12-13:
@Asanga-Viraj please decode your stack trace. See here for more details https://github.com/SmingHub/Sming/blob/develop/samples/LiveDebug/README.rst#exception-handling

mikee47, 2021-12-13:
@Asanga-Viraj Or run in the debugger https://smingdev.readthedocs.io/en/latest/arch/esp32/debugging/index.html. Re-build first though:

```
make clean components-clean
make -j ENABLE_GDB=1
make flashapp gdb
```

mikee47, 2021-12-13:
I've done some basic checks on system task vs. timer and cannot as yet see any difference. @Asanga-Viraj Can you push your app. or a simplified version to github? I can then investigate further...

Asanga-Viraj, 2021-12-13:
Sorry for my late reply.

I called UART interrupts Tx complete delegate and RX complete delegate directly without going through the system queues. This was happened when ESP trying to send binary files to the web page on requests. I re-run the issue and got this back trace. 
```
0x40115eef: HardwareUart::getStatus() at ??:?
0x40086e9d: (anonymous namespace)::uart_isr((anonymous namespace)::smg_uart_instance_t*) at c:/tools/Sming/Sming/Arch/Esp32/Components/driver/uart.cpp:278
0x40081d39: _xt_lowint1 at C:/tools/esp-idf/v4.3/components/freertos/port/xtensa/xtensa_vectors.S:1105
0x400825ce: spi_flash_op_block_func at C:/tools/esp-idf/v4.3/components/spi_flash/cache_utils.c:114
0x40086680: ipc_task at C:/tools/esp-idf/v4.3/components/esp_ipc/ipc.c:62
```
This make sense. Uart interrupted the flash operations because Uarts interrupt is not going over System queue. 
I fixed this issue by stopping the Uart interrupts (disabling Serial) for 2 second when HTTP server requests files.

Sorry, I don't have ESP debug tool now. I will buy one.

My initial question was System queues are not fast. It takes average time 30ms and it varies between 5ms-50ms. I think this is nothing to do with Sming framework. It is inside of the ESP FreeRTOS. For now, I can handle it.

My app consist of lots of things. I will upload simplified version later.