SmingHub/Sming #2490: Win10 MinGW: Make for Samples fails with Error 2

TheEclectus, 2022-02-15:
I installed Sming, the ESP tools, and Python per Sming's documentation. When I run `make -j` to build individual samples, the process is aborted with a message akin to 
`make[2]: *** No rule to make target 'c\:/tools/Sming/Sming/Components/Storage/src/CustomDevice.cpp', needed by 'src/CustomDevice.o'. Stop.`

It varies between the samples, seemingly without reason. For instance, Basic_Blink will fail because of a missing target for `src/CustomDevice.cpp`, then for `Arch/Esp8266/Components/esp8266/crash_handler.c`. 

Is it known if this is caused by a missed step perhaps? Or perhaps some other form of mis-configuration on my development machine? I'm running Windows 10 64-bit using the MinGW binaries linked in Sming's documentation, Python 3.10 and the latest release branch of Sming.

mikee47, 2022-02-15:
`make -j` uses **unlimited** build jobs and doesn't work well in Windows and virtual environments. https://sming.readthedocs.io/en/latest/_inc/Sming/building.html?highlight=%22-j%22#building.

You can get away with it in linux, but should specify a limit, e.g. `make -j5` according to your available CPU resources.

Before rebuilding, you may need to run `make dist-clean` and/or `make config-clean` to tidy things up.

Does that fix your issue?

mikee47, 2022-02-15:
If things are still not working for you, run `make list-config` and post the results,, that'll help.

mikee47, 2022-02-15:
See also https://sming.readthedocs.io/en/latest/getting-started/windows/wsl.html

TheEclectus, 2022-02-17:
Oh, I was unaware of the `-j` flag being so picky on Windows, and that I'd need to run `dist-clean` before rebuilding. That seems to have solved it!