SmingHub/Sming #2494: Running Websocket_Client sample causes `HTTP parser error: HPE_CB_headers_complete`

gbaranski, 2022-02-22:
I'm using sample from https://github.com/SmingHub/Sming/tree/develop/samples/Websocket_Client.
Running on ESP8266 Wemos D1 Mini Board.

slaff, 2022-02-22:
Ok, I see the problem. The websocket server that we use in the sample `echo.websocket.org` is no longer publicly available. Take a look at application.cpp:

https://github.com/SmingHub/Sming/blob/develop/samples/Websocket_Client/app/application.cpp#L39

and replace the websocket server URL with your own websocket server.

gbaranski, 2022-02-22:
alright that works now. Don't you think that those errors should be more readable to the user?

mikee47, 2022-02-27:
> alright that works now. Don't you think that those errors should be more readable to the user?

Agreed. I've had a dig into exactly what's going on here using the _**Simple Websocket Client**_ plugin for firefox. These are the response headers:

```
HTTP/1.1 200 OK
x-amz-id-2: aDP+2nX5soleb7+lPTk6c8QMuXDvTCWpToz6IMrI9rgsw6VyWJUmQ0+kmcS9NdkbqH4OB8zZeoY=
x-amz-request-id: 2C8VTRTYZA541D4K
Date: Sun, 27 Feb 2022 13:35:37 GMT
Last-Modified: Thu, 26 Aug 2021 03:35:42 GMT
ETag: "6d5a28f4c5e297e95a29988884c9e896"
Content-Type: text/html
Server: AmazonS3
Content-Length: 4413
Connection: close
```

So no error, just that the `Sec-WebSocket-Accept` header isn't present. Running the sample application in host mode with debugger locates the point of failure and I've opened a PR so we can have  a look at improving this.

In this particular instance the exact cause is only obvious by visiting `http://echo.websocket.org`...