SmingHub/Sming #2498: Hardware config 'spiffs' has invalid partition table

Asanga-Viraj, 2022-04-05:
Hi There,

For Sming 4.5.0 release, I tried command 'make flashinit' for ESP8266. After erasing the flash, then I tested flashing of an app with 'HWCONFIG := spiffs' hardware configs. It is not working. Esp8266 keeps resetting. May be some error with the partition table.

Thanks

slaff, 2022-04-05:
Take a look at this help article: https://sming.readthedocs.io/en/latest/troubleshooting/random-restart.html and tell us what is printed in your terminal. Also are you testing the code with some of the already provided samples or you are using your own application to test?

Asanga-Viraj, 2022-04-05:
Thank you for the tip. I tested on provided example.

I tested ESP8266 on win PC Sming V4.5.0 example 'C:\tools\Sming\samples\HttpServer_AJAX'.

`make flashinit`
`make rebuild`
`make flash`

Then, ESP8266 keeps printing junks. I didn't decode the stack because it didn't show decodable stack.

I spend hours and found a solution. Flashing of partitions.bin into address locations such 0x000fa000 and 0x003fa000 fixed the issues.