SmingHub/Sming #2500: Properly System.restart() esp32

kmihaylov, 2022-04-13:
Hello. When ESP utilizes WiFi it couldn't properly disconnect or restart the device.
```
E (18626) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (18626) task_wdt:  - Sming (CPU 1)
E (18626) task_wdt: Tasks currently running:
E (18626) task_wdt: CPU 0: IDLE
E (18626) task_wdt: CPU 1: IDLE
E (18626) task_wdt: Aborting.

abort() was called at PC 0x400de9fc on core 0

Backtrace:0x40087b6a:0x3ffb0580 0x400882b5:0x3ffb05a0 0x4008e466:0x3ffb05c0 0x400de9fc:0x3ffb0630 0x40081d49:0x3ffb0650 0x4012fae7:0x3ffbaee0 0x400de80a:0x3ffbaf00 0x40089408:0x3ffbaf20

ELF file SHA256: 5e395847006b60a9

Rebooting...
```

I use the Basic_WiFi example and I've added ```System.restart(10000);``` at the end of the ```init()``` function.

Also the same applies if I try to use ```Wifi.disconnect();``` or ```WifiStation.enable(false);```.

Asanga-Viraj, 2022-05-24:
I saw the same issue in ESP32. `System.restart()` stuck in infinite loop and do Watchdog reset. Watchdog reset doesn't start the ESP as fresh start. It makes a core synching issue and keeps restarting. The problem origin is [idf-v4.3](https://github.com/espressif/esp-idf/tree/v4.3) from expressif. I hope new [idf-4.4.1](https://github.com/espressif/esp-idf/compare/v4.3...v4.4.1) version may not have issue like this. As temporarily, I connected a wire from digital output to EN to make reset (ugliest way to do it).

Methods of the WiFi are not working. It is a huge problem. I tried things in many ways. But they kept ESP32 making Watchdog reset.

kmihaylov, 2022-08-19:
Hm, some answers here are missing.

Anyway, I'm watching esp-idf's function ```esp_wifi_stop()```. How do I properly call it from within Sming?

kmihaylov, 2022-08-20:
I'm answering myself. the ```esp_wifi_stop()``` and ```esp_wifi_deinit()``` functions can easily be accessed once the user adds the ```#include <esp_wifi.h>``` header to the application.cpp file.

I found that ```System.restart()``` in the Basic_Wifi example will work if the Wifi Access Point is disabled. It seems that some task continues to work if the app isn't prepared for restart leading to WDT reset dropping to GDB.

Therefore I'm closing my issue now. The restart process seems to need more attention by the developer.

mikee47, 2022-08-20:
> The restart process seems to need more attention by the developer

If you'd like to suggest improvements please feel free to open a PR!

mikee47, 2022-08-20:
@kmihaylov If it's OK with you we should leave this issue open as it definitely needs a fix.

kmihaylov, 2022-08-22:
Still trying to figure out why the system panics. I was quick to say the Access Point feature of the Basic_Wifi sample was the reason for the panic.

Later I found that the panic still occurs, despite the AP feature was turned off.

Actually if I use ```System.onReady()``` to turn on the Wifi (```WifiStation.enable(true)```) the panic occurs. But if the code is consecutively executed:
```
	System.onReady(ready);

	WifiStation.enable(true);
	WifiStation.config(_F(WIFI_SSID), _F(WIFI_PWD));

	WifiStation.setIP(IpAddress(192, 168, 1, 171));

	// Set callback that should be triggered when we have assigned IP
	WifiEvents.onStationGotIP(connectOk);

	// Set callback that should be triggered if we are disconnected or connection attempt failed
	WifiEvents.onStationDisconnect(connectFail);

	esp_wifi_stop();
	esp_wifi_deinit();
	System.restart(5000);
```

The panic is missing.

-----

One way to mitigate this is to change the  CONFIG_ESP_SYSTEM_PANIC  config variable.
It resides in ```Component config -> ESP System Settings```

https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32/api-guides/fatal-errors.html

```
make SMING_ARCH=Esp32 sdk-config
...
make SMING_ARCH=Esp32 Sming-build all
```

Obviously the default is ```CONFIG_ESP_SYSTEM_PANIC_GDBSTUB```.
I changed it to ```CONFIG_ESP_SYSTEM_PANIC_PRINT_REBOOT``` and now at least the system reboots after panicing.

This is important, otherwise I was unable to properly OTA update my devices.

Asanga-SPLat, 2022-08-23:
I tried to fix this while ago. Unable to do that. There is a problem in WiFi initialization on Sming. For your reference, try WiFi examples on on esp-idf-4.3. When you are calling esp_restart(); in idf code, it resets the chip perfectly. No watch dog reset.

You can change the panic result of the ESP32 by changing settings on sdk.
`make sdk menuconfig`
You can use arrows to move up and down, enter to enter into a item, Esc to exit, Q to quite and S to save. Then go to, Component config--> ESP System Settings--> Panic handler behaviour--> Print registers and reboot. Select it by pressing space/enter button. Press S and save it. Q to exit. Then run,
`make components-clean`
`make -j6`

kmihaylov, 2022-08-23:
@mikee47 

My Esp32 code used to do ```WifiStation.enable(true)``` before ```System.onReady()```. This way the Wifi is working.

I found that if I try to enable the wifi with ```System.onReady```, then it brakes.

See the code below:
```
#include <SmingCore.h>
#include "esp_wifi.h"

void wifiInit() {
	esp_wifi_deinit();
	esp_netif_create_default_wifi_sta();
	wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
	ESP_ERROR_CHECK(esp_wifi_init(&cfg));
	esp_wifi_set_mode(WIFI_MODE_STA);
	wifi_config_t wifiCfg;
	esp_wifi_set_config(WIFI_IF_STA, &wifiCfg);
	esp_wifi_start();
}

void init()
{
	wifiInit();
	//System.onReady(wifiInit);
}
```
This is the modified Basic_Blink (w/ networking enabled). If you comment out the ```wifiInit()``` function and use the ```onReady``` handler instead, the chip enters in endless restart (or drops to GDP shell, depending on IDF's build options).

I followed this example: https://github.com/espressif/esp-idf/blob/7c2b137ffe8a91ef0967cd5da1fe330ff5912399/examples/wifi/getting_started/station/main/station_example_main.c

And the above mentioned code is the extracted minimum of functions. I thik the system freezes exactly on ```esp_wifi_start()```.

```esp_netif_init()``` is executed in Sming/Components/Network/Arch/Esp32/Platform/init.cpp so I omit it. 

The only missing function from this initialization code is ```esp_event_loop_create_default()``` and Sming uses its own ```sming_create_event_loop()```. RTOS and events are outside my knowledge so I don't know if this could cause the problem.

mikee47, 2022-08-23:
I suspect one problem is that when `onReady()` gets to run the WiFi stack is being 'automatically' initialised. To diagnose what's happening with the endless rebooting we need to be able to see the bootloader diagnostics by setting `BOOTLOADER_LOG_LEVEL` appropriately in the SDK config. The Sming WiFi initialisation code can be found in `Components/Network/Esp32/Platform/init.cpp`, and the station/access code in `StationImpl.cpp` and `AccessPointImpl.cpp` accordingly.

kmihaylov, 2022-08-23:
I set the ```BOOTLOADER_LOG_LEVEL``` to Debug.

However now some overlap occurs:
```
$ make flash

Basic_Blink: Invoking 'flash' for Esp32/esp32 (debug) architecture
ninja: Entering directory `/home/opt/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk'
[1/3] Performing build step for 'bootloader'
ninja: no work to do.
Copying generated SDK libraries
make components application
make[1]: Entering directory '/home/opt/Sming/samples/Basic_Blink'

Building /home/opt/Sming/samples/Basic_Blink/out/Esp32/esp32/debug/lib/clib-App-98050d16e24ff8720e998f71db21cffd.a
make[2]: Nothing to be done for 'build'.
Validating hardware config 'standard'
Verifying partition table...
Writing to 'out/Esp32/esp32/debug/firmware/partitions.bin'
SSL support disabled
Basic_Blink: Linking out/Esp32/esp32/debug/build/app.out
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFBDB60|    3FFC174C|   15340
    rodata|           ReadOnly Data (SPI)|    3F400020|    3F412130|   74000
       bss|      Uninitialized Data (RAM)|    3FFC1750|    3FFC5590|   15936
 iram_text|            Cached Code (IRAM)|    40080400|    40093C60|   79968
      text|           Uncached Code (SPI)|    400D0020|    4012D1D3|  381363
Total Used RAM : 31276
Free RAM : 304596
Free IRam : 51104
esptool.py v3.2-dev
Merged 2 ELF sections
make[1]: Leaving directory '/home/opt/Sming/samples/Basic_Blink'
Killing Terminal to free /dev/ttyUSB0
make: [/home/opt/Sming/Sming/Components/terminal/component.mk:32: kill_term] Killed (ignored)
WriteFlash 0x1000=/home/opt/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/bootloader/bootloader.bin 0x00008000=out/Esp32/esp32/debug/firmware/partitions.bin   0x00010000=out/Esp32/esp32/debug/firmware/app.bin
/usr/bin/python3 /home/opt/Sming/Sming/Components/esptool/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 --chip esp32 --before default_reset --after hard_reset write_flash -z -fs 4MB -ff 40m -fm dio 0x1000 /home/opt/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/bootloader/bootloader.bin 0x00008000 out/Esp32/esp32/debug/firmware/partitions.bin   0x00010000 out/Esp32/esp32/debug/firmware/app.bin 
usage: esptool write_flash [-h] [--erase-all] [--flash_freq {keep,40m,26m,20m,80m}] [--flash_mode {keep,qio,qout,dio,dout}] [--flash_size FLASH_SIZE] [--spi-connection SPI_CONNECTION]
                           [--no-progress] [--verify] [--encrypt] [--encrypt-files <address> <filename> [<address> <filename> ...]] [--ignore-flash-encryption-efuse-setting]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0x8000 for file: out/Esp32/esp32/debug/firmware/partitions.bin
make: *** [/home/opt/Sming/Sming/component.mk:137: flash] Error 2
```

```
 $make map

Basic_Blink: Invoking 'map' for Esp32/esp32 (debug) architecture
Partition map: standard
options: 
Device            Start       End         Size        Type      SubType   Name              Filename
----------------  ----------  ----------  ----------  --------  --------  ----------------  ------------
spiFlash          0x00000000  0x00007fff         32K                      Boot Sector       
spiFlash          0x00008000  0x00008fff          4K                      Partition Table   
spiFlash          0x00009000  0x0000efff         24K  data      nvs       nvs               
spiFlash          0x0000f000  0x0000ffff          4K  data      phy       phy_init          
spiFlash          0x00010000  0x000fffff        960K  app       factory   factory           $(TARGET_BIN)
spiFlash          0x00100000  0x003fffff          3M                      (unused)          
```

kmihaylov, 2022-08-23:
Pretty good bug to bring me the knowledge to make arbitrary partition tables and esp-idf specific options!

I managed to make custom partition table and I learned how to change the partition table offset. Unfortunately I don't see any useful information in the debug messages.

Here I'll post all the options I changed and all the output.

Application code:
```
#include <SmingCore.h>
#include "esp_wifi.h"

Timer testTimer;

void wifiInit() {
	esp_wifi_deinit();
	esp_netif_create_default_wifi_sta();
	wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
	ESP_ERROR_CHECK(esp_wifi_init(&cfg));
	esp_wifi_set_mode(WIFI_MODE_STA);
	wifi_config_t wifiCfg;
	esp_wifi_set_config(WIFI_IF_STA, &wifiCfg);
	esp_wifi_start();
}

void shutdown() {
	esp_wifi_stop();
	esp_restart();
}

void init()
{
	wifiInit();
	testTimer.initializeMs(5000, shutdown).startOnce();
}
```

```
$ make map

Basic_Blink: Invoking 'map' for Esp32/esp32 (debug) architecture
Partition map: bigboot
options: 4m
Device            Start       End         Size        Type      SubType   Name              Filename
----------------  ----------  ----------  ----------  --------  --------  ----------------  ------------
spiFlash          0x00000000  0x0000ffff         64K                      Boot Sector       
spiFlash          0x00010000  0x00010fff          4K                      Partition Table   
spiFlash          0x00011000  0x00016fff         24K  data      nvs       nvs               
spiFlash          0x00017000  0x00017fff          4K  data      phy       phy_init          
spiFlash          0x00018000  0x0001ffff         32K                      (unused)          
spiFlash          0x00020000  0x0010ffff        960K  app       factory   factory           out/Esp32/esp32/debug/firmware/app.bin
spiFlash          0x00110000  0x003fffff       3008K                      (unused)          
```

I changed the boorloader verbosity level from ```sdk-menuconfig``` -> ```Bootloader config``` -> ```Bootloader log verbosity``` to ```Debug```. (I also changed the ```Bootloader optimization Level``` to ```Debug```).

Also I moved the partition table offset in ```sdk-menuconfig``` -> ```Partition Table``` -> ```Offset of partition table``` to ```0x10000```. (Depends on the custom HWCONFIG).

And here is the output:
```
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:9740
ho 0 tail 12 room 4
load:0x40078000,len:18064
load:0x40080400,len:4544
[... 114 lines cut ...]
load:0x40080400,len:4544
entry 0x4008071c
I (29) boot: ESP-IDF v4.3-1675-g4a43d72911 2nd stage bootloader
I (29) boot: compile time 20:16:49
```

The decoded stack:
```
0x40081682: panic_abort at /home/opt/esp-idf-4.3/components/esp_system/panic.c:404
0x400883f9: esp_system_abort at /home/opt/esp-idf-4.3/components/esp_system/system_api.c:112
0x4008e4fe: abort at /home/opt/esp-idf-4.3/components/newlib/abort.c:46
0x400da9e4: task_wdt_isr at /home/opt/esp-idf-4.3/components/esp_common/src/task_wdt.c:182 (discriminator 1)
0x40081dad: _xt_lowint1 at /home/opt/esp-idf-4.3/components/freertos/port/xtensa/xtensa_vectors.S:1105
0x401291fb: cpu_ll_waiti at /home/opt/esp-idf-4.3/components/hal/esp32/include/hal/cpu_ll.h:183
 (inlined by) esp_pm_impl_waiti at /home/opt/esp-idf-4.3/components/esp_pm/pm_impl.c:824
0x400da7f2: esp_vApplicationIdleHook at /home/opt/esp-idf-4.3/components/esp_common/src/freertos_hooks.c:63
0x40089544: prvIdleTask at /home/opt/esp-idf-4.3/components/freertos/tasks.c:3825
```

I'll write again the files containing the Wifi initialization code:
```
Sming/Arch/Esp32/Components/esp32/src/startup.cpp
Sming/Components/Network/Arch/Esp32/Platform/init.cpp
Sming/Components/Network/Arch/Esp32/Platform/StationImpl.cpp
```

mikee47, 2022-08-24:
@kmihaylov The debugging I was after is actually `LOG_DEFAULT_LEVEL=5` (verbose) - sorry for misdirecting you, but glad you learned something from it!

With your sample, the WDT reset looks like this:

```
V (4470) esp_adapter: thread sem get: sem=3fc96e54
E (11440) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (11440) task_wdt:  - Sming (CPU 0)
E (11440) task_wdt: Tasks currently running:
E (11440) task_wdt: CPU 0: IDLE
E (11440) task_wdt: Aborting.
```

Looks like a semaphore is involved, so probably a deadlock somewhere. Turns out it's the event handling loop, not the networking at all! Sming doesn't use the default event loop but creates its own so that all software timers and queued callbacks run in the Sming thread context. See https://github.com/SmingHub/Sming/pull/2371 for details.

It could be that the IDF expects the event loop to be running in a separate thread, still investigating the exact reason.

I've opened PR #2543 (work in progress) so you can try `make CREATE_EVENT_TASK=1`. This is NOT a solution!

Asanga-Viraj, 2023-02-09:
Hi All,

I fixed this issue... (I have no idea whether this issue was fixed or not)
It is the event loop issue. So have to apply changes into [event_loop.cpp](https://github.com/SmingHub/Sming/blob/4.5.0/Sming/Arch/Esp32/Components/esp32/src/event_loop.cpp).

    esp_event_loop_args_t loop_args = {
        .queue_size = CONFIG_ESP_SYSTEM_EVENT_QUEUE_SIZE,
        .task_name = "sys_evt",        
        .task_priority = ESP_TASKD_EVENT_PRIO,
        .task_stack_size = ESP_TASKD_EVENT_STACK,
        .task_core_id = 0
    };

mikee47, 2023-02-09:
Yes, this issue was fixed in #2543 - you can find an explanation of the cause there.

Asanga-Viraj, 2023-02-09:
@mikee47  Thanks.

mikee47, 2024-03-17:
Closing as fixed by #2543

closed 2024-03-17 by mikee47