SmingHub/Sming #2501: question: AWS IOT root certificate

calvernaz, 2022-04-18:
Hi, we are looking into the sming framework and specifically the MQTT implementation to connect the device to AWS IoT Core and in order for that communication to be successful, we need to set up TLS and the root authority certificate. We can’t find the correct API to assign a root certificate to the session. Do you have an example of how to do it?

The example you provide in the repository doesn't seem to use the authority certificate from Amazon, even though they are in the repo but no references to it.

slaff, 2022-04-19:
@calvernaz We have a sample that is demonstrating the use of Amazon AWS. The keys and certificates are loaded in application.cpp lines 14 until 16. 

   https://github.com/SmingHub/Sming/blob/develop/samples/Basic_AWS/app/application.cpp#L14-L16

And then assigned in line 23:

  https://github.com/SmingHub/Sming/blob/develop/samples/Basic_AWS/app/application.cpp#L23

The files are in the files/ directory of the sample and they MUST be in the binary DER format and not in text readable PEM format.

calvernaz, 2022-04-19:
Hi @slaff thanks for the quick response, but so I totally understand, in those code bits I only see the use of the private key and the certificate and no use of the root certificate ([AmazonRootCA1.pem.der](https://github.com/SmingHub/Sming/blob/develop/samples/Basic_AWS/files/AmazonRootCA1.pem.der)).