SmingHub/Sming #2502: IR lib FlashString handling

kmihaylov, 2022-04-18:
This is not Sming issue, but something that I can't figure out. I'm trying to port the latest IR lib to Sming and I stuck with IR's FlashStrings not build with Sming.

First let me show the compiler error:
```
Building /home/opt/Sming/Sming/out/Esp8266/debug/lib/clib-IR.a
C+ /home/opt/Sming/Sming/Libraries/IR/src/ir_Carrier.cpp
/home/opt/Sming/Sming/Libraries/IR/src/ir_Carrier.cpp: In member function 'String IRCarrierAc64::toString() const':
/home/opt/Sming/Sming/Libraries/IR/src/ir_Carrier.cpp:493:38: error: could not convert 'kPowerStr' from 'const __FlashStringHelper* const' to 'String'
  493 |   result += addBoolToString(_.Power, kPowerStr, false);
      |                                      ^~~~~~~~~
      |                                      |
      |                                      const __FlashStringHelper* const
```

The variable ```kPowerStr``` is defined in ```IRtext.cpp```. I'm posting the macros:
```
#define IRTEXT_CONST_BLOB_NAME(NAME)\
    NAME ## Blob

#define IRTEXT_CONST_BLOB_DECL(NAME)\
    const char IRTEXT_CONST_BLOB_NAME(NAME) [] PROGMEM

#define IRTEXT_CONST_BLOB_PTR(NAME)\
    IRTEXT_CONST_PTR(NAME) {\
        IRTEXT_CONST_PTR_CAST(IRTEXT_CONST_BLOB_NAME(NAME)) }

#define IRTEXT_CONST_STRING(NAME, VALUE)\
    static IRTEXT_CONST_BLOB_DECL(NAME) { VALUE };\
    IRTEXT_CONST_PTR(NAME) PROGMEM {\
        IRTEXT_CONST_PTR_CAST(&(IRTEXT_CONST_BLOB_NAME(NAME))[0]) }
```
and
```
IRTEXT_CONST_STRING(kPowerStr, D_STR_POWER);  ///< "Power"
```

Then Eclipse evaluated the macro to:
```
static const char kPowerStrBlob [] __attribute__((aligned(4))) { "Power" };
const char* const kPowerStr __attribute__((aligned(4))) {&(kPowerStrBlob)[0] }
```

where PROGMEM is(?) ``` __attribute__((aligned(4)))``` (Looking in the header files I thing it should be something longer...?

I don't know how ```const char* const``` went to ```const __FlashStringHelper* const```. I'm reading https://sming.readthedocs.io/en/latest/framework/core/pgmspace.html w/o luck.

mikee47, 2022-04-18:
The constructor is delcared `explicit String(flash_string_t pstr)` so you'll need to modify code like this:

```
result += addBoolToString(_.Power, String(kPowerStr), false);
```

mikee47, 2022-04-19:
On further inspection, the FlashString library can handle all this more easily. See https://github.com/mikee47/IRremoteESP8266/tree/sming. Builds but not tested.

kmihaylov, 2022-04-19:
@mikee47 thanks. 

Do I understand properly that ```String(kPowerStr)``` will copy the flash string to RAM?

The library heavily uses *Str like the above mentioned ```kPowerStr```. I have no problem to modify all the files to ```String(...)```, but is this the best way? Is this some Arduino thing that the flash strings can be directly used?

kmihaylov, 2022-04-19:
IR receive works just fine. I can solder second board to check IR send.

mikee47, 2024-04-12:
Assuming resolved

closed 2024-04-12 by mikee47