SmingHub/Sming #2511: Outdated Basic_AWS example

Asanga-Viraj, 2022-05-19:
I am working on connecting ESP into MQTT AWS IoT Core. I couldn't' connect it because it shows a debug message "please start sntp first !". I went deep and I saw  "sntp_init()" included in ["configTime"](https://github.com/igrr/axtls-8266/blob/9ebc3ee6ec659a09f08f6e3d8b4d6c19db5ce498/replacements/time.c) is not called any where. 
Then, I called below lines on gotIP function.

`sntp_stop();`
`sntp_setservername(0, (char*)"pool.ntp.org");`
`sntp_setservername(1, (char*)"0.au.pool.ntp.org");`
`sntp_set_timezone(10); // GTM+10`
`sntp_init();`

 Now, message "please start sntp first !" is gone and got "SSL Validator: list empty, allow connection". Can you help me with this?

Thanks.

slaff, 2022-05-19:
@Asanga-Viraj Isn't [this sample](https://github.com/SmingHub/Sming/tree/develop/samples/Basic_AWS) working for you? In it you can see how to attach SSL certificates:

https://github.com/SmingHub/Sming/blob/develop/samples/Basic_AWS/app/application.cpp#L21-L24

In addition if you want to syncrhonise the local clock you can use the [NTP sample](https://github.com/SmingHub/Sming/tree/develop/samples/SystemClock_NTP) as  a starting point.

Asanga-Viraj, 2022-05-19:
@slaff I couldn't manage to work [Basic_AWS](https://github.com/SmingHub/Sming/blob/develop/samples/Basic_AWS/app/application.cpp) example with the current MQTT AWS IoT Core. To connect to AWS, we need to assign a device certificate, private key, and Root CA1 certificate. But there is no entry to assign the Root CA1 certificate. Refer [this Arduino](https://how2electronics.com/connecting-esp8266-to-amazon-aws-iot-core-using-mqtt/) example. it is working with AWS.

I forgot to tell you that the message "please start sntp first !" only appears on ESP8266. Not in ESP32. I will test with NTP sample. Thank you for that.

slaff, 2022-05-20:
> To connect to AWS, we need to assign a device certificate, private key, and Root CA1 certificate. But there is no entry to assign the Root CA1 certificate.

Short Story
=========
The Amazon Root CA1 certificate is not used in the sample because we skip SERVER CERTIFICATE verification. We should probably update it to include full SERVER CERTIFICATE verification. 
But it does not mean that the connection is not working or that the client application is not able to send or receive MQTT data. 

Long Story
========
Ok, even with over-simplifications, this is probably going to be a long answer but once you read this you should know better 
- How TLS/SSL communication works
- How to debug network communication in Sming

How TLS/SSL communication works
-------------------------------------------------
Here we are focusing only on the communication from our device, the client, to the remote AWS cloud, the server. When a client connects to a remote server using TLS it does many things. Some of the important ones are:
- A) it establishes encrypted communication
- B) it trusts the remote side.

Let's start with A).  What happens here is the following:
- A1) client first tries to make a TCP connection to the provided hostname and port. 
- A2) IF this succeeds the client communicates with the remote server using ASYMMETRICAL encryption. Basically the server sends to the client  a **SERVER CERTIFICATE** which describes the server plus contains a public key. Further the client and the server negotiate a TOKEN to be used to continue the communication. This process is slow. The client encrypts its data using the public key from the server and the server decrypts the data using its private key. Once such a TOKEN is known to both sides the communication switches to SYMMETRICAL encryption. Now this is much faster. 

Let's focus on the B). In order to trust the remote side the client could/should verify the SERVER CERTIFICATE. The fact that the communication is encrypted does not prevent the client from talking to a malicious server pretending to be a valid AWS server. The SERVER CERTIFICATE verification does. A full verification is expensive, especially for embedded devices like ESP8266. A full verification requires the expiration date of the SERVER CERTIFICATE to be checked. For this your device will need a VALID TIME. Getting the current time will require another network call to an NTP server which additionally increases the time and resource needed. If your device is not time critical and not running on a battery then fine. If you want to go this route then make sure to replace the code below:
```c++
mqtt.setSslInitHandler([](Ssl::Session& session) {
		session.options.verifyLater = true;
		session.keyCert.assign(privateKeyData, certificateData);
	});
```

with

```c++
mqtt.setSslInitHandler([](Ssl::Session& session) {
		session.options.verifyLater = false; // Do expensive certificate validation!
		session.keyCert.assign(privateKeyData, certificateData);
	});
```

Now in order to fully verify a certificate [a lot of steps](https://www.ssl.com/article/browsers-and-certificate-validation/) need to be done. Not only the expiration date, hostname in the certificate matching the one one used BUT also its signature. My bold guess is that the SERVER CERTIFICATE is signed using the Amazon  Root CA1 certificate. This way the client can verify with high certainty if the SERVER CERTIFICATE is the one that is expected. And thus trust the remote server to be the one that it pretends to be.

A slightly less secure but much faster way is to verify the provided SERVER CERTIFICATE against its SHA1 or SHA256 finger-prints. Basically the content of the certificate is used to feed a SHA1/SHA256 hash function and compared against expected finger-print. Of course even small changes in the SERVER CERTIFICATE will cause our fingerprinting check to fail. Slightly less error-prone and a bit slower is to extract the public key from the SERVER CERTIFICATE and compare its SHA1/SHA256 hash against expected ones. Our Basic_SSL example explains how to set such fingerprint checks:
https://github.com/SmingHub/Sming/blob/da1ff089d29dfa00e6183722656221ae67ca4e9d/samples/Basic_Ssl/app/application.cpp#L78-L100

> But there is no entry to assign the Root CA1 certificate.

In our Basic_AWS example we DON'T do full SERVER CERTIFICATE verification this is why you don't see it used.

> got "SSL Validator: list empty, allow connection". 

Basic_AWS is not using any server certificate validators and this is the reason why you see this message. You can either set finger-print validators or even set a [callback validator](https://sming.readthedocs.io/en/latest/_inc/Sming/Components/ssl/certificates.html?highlight=validator#_CPPv4N3Ssl17ValidatorCallbackE) where the received SERVER CERTIFICATE is provided to the validator and in your application you can do all needed manual validations.

In the example the client provides its CLIENT CERTIFICATE and private key. 

```c++
mqtt.setSslInitHandler([](Ssl::Session& session) {
		session.options.verifyLater = true; 
		session.keyCert.assign(privateKeyData, certificateData);
	});
```

The CLIENT CERTIFICATE is sent to AWS and used by the remote server to check if it was issued by them. This way our client is ALLOWED to communicate further with the AWS servers.  The private key is used from our client to encrypt the communication. AWS generates the private key, then the CLIENT CERTIFICATE and allows you to download the p [... cut ...]
445312 SSL Validator: list empty, allow connection
445336 TCP 56632400 connected << ----- IMPORTANT
445342 TCP 56632400 onReadyToSendData: 0
445373 TCP 56632400 Written: 23, Available: 2780, isFinished: 0, PushCount: 1
445397 TcpClient stream finished
445407 [MQTT] Sending message type 8 << ----- IMPORTANT
```

There is communication with the remote server, data is sent and all looks good.

This is all you need to know. Now try to compile your application with DEBUG_VERBOSE_LEVEL=3 and paste the output here.

Asanga-Viraj, 2022-05-24:
@slaff 

Thank you for the full detail information. Extremely sorry for my lack of knowledge about TSL/SSL communication. Now I have the idea about how root certificate authority (RootCA1) file works. So, nothing wrong about setSslInitHandler(). I tried to work with Basic_AWS example on ESP32. Then I got this with DEBUG_VERBOSE_LEVEL=3

> 1117438 MQTT start connection
1118530 TCP 3ffb4354 +connection
1121764 TCP 3ffb4354 connect to "aXXXXXXXXa-ats.iot.us-west-1.amazonaws.com:8883"
1129938 subscription 'esp32/sub' registered
1132926 TCP 3ffb4354 connection closing
1136229 MQTT closed previous connection
1139758 MQTT start connection
1142475 MQTT replacing connect message
1146067 TCP 3ffb4354 +connection
1148903 TCP 3ffb4354 connect to "aXXXXXXXXa-ats.iot.us-west-1.amazonaws.com:8883"
1156530 subscription 'esp32/sub' registered
publish message
1162005 [MQTT] Sending message type 1
1165369 MemoryDataStream::realloc 0 -> 24
1168978 Sending stream. Bytes to send: 24
1172779 subscription '$aws/things/ESP32_Test/shadow/get' registered
1190234 TCP 3ffb4354 DNS record found: aXXXXXXXXa-ats.iot.us-west-1.amazonaws.com = 52.52.86.23
1197254 TCP connect result: 0
1198892 TCP 3ffb4354 DNS record found: aXXXXXXXXa-ats.iot.us-west-1.amazonaws.com = 52.52.86.23
1207651 TCP connect result: -10
1389417 TCP 3ffb4354 connected: useSSL: 1, Error: 0
1707433 TCP 3ffb4354 sent: 127
3845648 TCP 3ffb4354 sent: 1503
3853587 SSL Validator: list empty, allow connection
3853777 TCP 3ffb4354 connected
3853888 TCP 3ffb4354 onReadyToSendData: 0
3854748 TCP 3ffb4354 Written: 24, Available: 5744, isFinished: 0, PushCount: 1
3860644 TcpClient stream finished
3863724 [MQTT] Sending message type 8
3867182 MemoryDataStream::realloc 0 -> 16
3870820 Sending stream. Bytes to send: 16
3875077 TCP 3ffb4354 Written: 16, Available: 5659, isFinished: 0, PushCount: 1
3881456 TcpClient stream finished
3884555 [MQTT] Sending message type 8
3887884 MemoryDataStream::realloc 0 -> 16
3891630 Sending stream. Bytes to send: 16
3895838 TCP 3ffb4354 Written: 16, Available: 5574, isFinished: 0, PushCount: 1
3902281 TcpClient stream finished
3905350 [MQTT] Sending message type 3
3908732 MemoryDataStream::realloc 0 -> 31
3912464 Sending stream. Bytes to send: 31
3916671 TCP 3ffb4354 Written: 31, Available: 5489, isFinished: 0, PushCount: 1
3923114 TcpClient stream finished
3926192 [MQTT] Sending message type 8
3929550 MemoryDataStream::realloc 0 -> 41
3933297 Sending stream. Bytes to send: 41
3937549 TCP 3ffb4354 Written: 41, Available: 5404, isFinished: 0, PushCount: 1
3943948 TcpClient stream finished
4159341 TCP 3ffb4354 sent: 441
4159537 TCP 3ffb4354 onReadyToSendData: 2
4361913 TCP 3ffb4354 sent: 69
4362141 TCP 3ffb4354 onReadyToSendData: 2
4362241 TCP 3ffb4354 receive: pbuf is NULL
4362299 TCP 3ffb4354 received: (null)
4365310 TCP 3ffb4354 connection closing
4442872 -TCP connection

At the end, TCP connection is disconnected. The I commented mqtt subscribes and publishes.

- https://github.com/SmingHub/Sming/blob/develop/samples/Basic_AWS/app/application.cpp#L33
- https://github.com/SmingHub/Sming/blob/develop/samples/Basic_AWS/app/application.cpp#L78-L80

After, I tried this on ESP32 and got this.

> 1117588 MQTT start connection
1118532 TCP 3ffb4354 +connection
[... 91 lines cut ...]
4319176 TCP 3ffb4354 received: 4 bytes
4320636 TCP 3ffb4354 onReadyToSendData: 1
subscribing for messages
11628498 subscription 'esp32/sub' registered
12312929 [MQTT] Sending message type 8
12313302 MemoryDataStream::realloc 0 -> 16
12313387 Sending stream. Bytes to send: 16
12316889 TCP 3ffb4354 Written: 16, Available: 5744, isFinished: 0, PushCount: 1
12320190 TcpClient stream finished
12510655 TCP 3ffb4354 sent: 85
12510779 TCP 3ffb4354 onReadyToSendData: 2
12716334 TCP 3ffb4354 sent: 69
12716570 TCP 3ffb4354 onReadyToSendData: 2
12716671 TCP 3ffb4354 receive: pbuf is NULL
12716730 TCP 3ffb4354 received: (null)
12720080 TCP 3ffb4354 connection closing
12822842 -TCP connection

**Things I saw**
- while stablishing connection, it is better not to subscribe or publish. Needs some time.
- mqtt EvenHandlers are not calling during two way SSL mode. Only `mqtt.setCompleteDelegate()` calls when disconnected. It is on TcpClient class.
- `mqtt.subscribe()` disconnects TCP AWS connection.

These are the things I found. 
Thanks.