SmingHub/Sming #2514: Some idf issues

Asanga-Viraj, 2022-05-24:
@mikee47 
I am working on ESP32 using Sming framework. I saw few things (Win 10).

- Sming `choco install -y sming.esp32` doesn't install [idf-v4.3](https://github.com/mikee47/esp-idf/tree/sming/release/v4.3) from your repository. It installs idf-v4.1. 
- Sming re-installation doesn't clean and reinstall ESP32 components.
- Saw many issues with current version such as System.restart, WiFi.Disconnect, etc.

Also just a suggestion, it is better if you can make Sming support with latest release idf (like idf-v4.4.1).

Thanks

Asanga-Viraj, 2022-05-26:
@slaff 
I found a issues in ESP32. `WifiStation.connect()` is not working. I found `esp_wifi_connect()` is not calling on this method. My suggestions are below.

- https://github.com/SmingHub/Sming/blob/4.5.0/Sming/Components/Network/Arch/Esp32/Platform/StationImpl.cpp#L120  `return esp_wifi_start() == ESP_OK;`
- https://github.com/SmingHub/Sming/blob/4.5.0/Sming/Components/Network/Arch/Esp32/Platform/StationImpl.cpp#L125-L126
 `if(isConnected())
disconnect();`
`ESP_ERROR_CHECK(esp_wifi_connect());`
`return true;`

Thanks

slaff, 2022-06-01:
@Asanga-Viraj thank you for your remarks.

> Sming choco install -y sming.esp32 doesn't install [idf-v4.3](https://github.com/mikee47/esp-idf/tree/sming/release/v4.3) from your repository. It installs idf-v4.1.

This should be fixes today with [this commit](https://github.com/slaff/chocolatey-packages/commit/40ebba32be0ad4ea9fe393b4d4d79fa4b8577d74).

> Sming re-installation doesn't clean and reinstall ESP32 components.

sming.esp32 should be the only package responsible for this. 

>  it is better if you can make Sming support with latest release idf (like idf-v4.4.1).

We usually skip a minor version and jump to the next one. But if you have the time and desire you are free to submit a PR with the needed changes.

kmihaylov, 2022-08-19:
@Asanga-Viraj do you have some progress on System.restart() ?

On my ESP8266 board I have to disconnect from the wifi network before calling System.restart() in order not to trigger (some) WD timer. I have to try wifi-less app with Esp32 to verify this.