SmingHub/Sming #2524: Failed to install on Mac

IanAber, 2022-07-11:
Using the instructions I installed the tools and the sming source. I changed to the Basic_Blink folder to try to run the sample and executed 'make'

The script ran and downloaded a couple of submodules but then I got this error...

`Building /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/out/Esp8266/debug/lib/clib-Storage-52412ccea9c0ff555e8ecb0b5573ceec.a
C+ /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Components/Storage/src/CustomDevice.cpp
In file included from /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Components/Storage/src/include/Storage/Device.h:12,
                 from /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Components/Storage/src/include/Storage/CustomDevice.h:7,
                 from /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Components/Storage/src/CustomDevice.cpp:5:
/Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Wiring/WString.h:64:10: fatal error: FlashString/String.hpp: No such file or directory
   64 | #include <FlashString/String.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [src/CustomDevice.o] Error 1
make[1]: *** [Storage-build] Error 2
make: *** [all] Error 2
`

I tried to clean everything out and start again but got the same result.

IanAber, 2022-07-11:
After individually downloading the missing components I now get a redefinition error.
`/Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Components/http-parser/http_parser.h:152:3: note: in expansion of macro 'XX'
  152 |   XX(511, NETWORK_AUTHENTICATION_REQUIRED, Network Authentication Required) \
      |   ^~
/Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Components/Network/src/Network/Http/HttpCommon.h:62:1: note: in expansion of macro 'HTTP_STATUS_MAP'
   62 | HTTP_STATUS_MAP(XX)
      | ^~~~~~~~~~~~~~~
In file included from /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Components/Network/src/Network/Http/HttpCommon.h:32,
                 from /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Components/Network/src/Network/HttpClient.h:22,
                 from /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Components/rboot/include/Network/RbootHttpUpdater.h:18,
                 from /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Components/rboot/src/Network/RbootHttpUpdater.cpp:16:
/Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Components/http-parser/http_parser.h:156:31: note: previous declaration 'http_status HTTP_STATUS_NETWORK_AUTHENTICATION_REQUIRED'
  156 | #define XX(num, name, string) HTTP_STATUS_##name = num,
      |                               ^~~~~~~~~~~~
`

slaff, 2022-07-11:
@IanAber can you go to the Basic_Blink sample and type:
```
make list-config
```

The top part of the result is important.

IanAber, 2022-07-11:
This is after downloading the missing components. If it would help for me to wife the sming installation and provide an output with a fresh installation let me know. Many thanks for your help.

`Basic_Blink: Invoking 'list-config' for Esp8266 (debug) architecture

** Sming build configuration **

PREREQUISITES:
CUSTOM_TARGETS:
- /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Arch/Esp8266/Components/esp8266/sdk/bin/esp_init_data_default.bin
- /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Arch/Esp8266/Components/esp8266/sdk/bin/esp_init_data_vdd_default.bin
- out/Esp8266/debug/firmware/rboot.bin
- out/Esp8266/debug/firmware/rom0.bin
- partmap-build
LIBS:
- c
- crypto
- gcc
- hal
- lwip
- m
- main-cff65c904b5b4f06763e697bf9b55747
- phy
ARDUINO_LIBRARIES:
SUBMODULES:
- /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Arch/Esp8266/Components/esp8266/sdk
- /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Components/esptool/esptool
- /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Components/rboot/esptool2
- /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Components/rboot/rboot
- /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Libraries/Spiffs/spiffs
CONFIG_VARS:
- COM_SPEED_SERIAL = $(COM_SPEED)
- DEBUG_PRINT_FILENAME_AND_LINE = 0
- DEBUG_VERBOSE_LEVEL = 2
- ENABLE_CMD_EXECUTOR = 1
- ENABLE_GDB = 
- ENABLE_SPI_DEBUG = 0
- ENABLE_TASK_COUNT = 
- GDB_UART_SWAP = 
- HWCONFIG = standard
- HWCONFIG_OPTS = 
- LOCALE = 
- RBOOT_GPIO_ENABLED = 0
- RBOOT_GPIO_SKIP_ENABLED = 0
- RBOOT_RTC_ENABLED = 0
- RBOOT_SILENT = 0
- STRING_OBJECT_SIZE = 12
- TASK_QUEUE_LENGTH = 10
- USER_CFLAGS = 
- USE_US_TIMER = 1
CACHE_VARS:
- APP_NAME = app
- COM_OPTS = --raw --encoding ascii --rts 0 --dtr 0
- COM_PORT = /dev/tty.usbserial-0001
- COM_PORT_ESPTOOL = $(COM_PORT)
- COM_PORT_GDB = $(COM_PORT)
- COM_SPEED = 115200
- COM_SPEED_ESPTOOL = $(COM_SPEED)
- FSBUILD_OPTIONS = 
- FULL_COMPONENT_BUILD = 
- GDB_CMDLINE = trap '' INT; $(GDB) -x $(GDBSTUB_DIR)/gdbcmds -b $(COM_SPEED_GDB) -ex "target remote $(COM_PORT_GDB)"
- KILL_TERM = pkill -9 -f "$(COM_PORT) $(COM_SPEED_SERIAL)" || exit 0
- PIP_ARGS = 
- SERVER_OTA_PORT = 9999
- SPIFF_BIN = spiff_rom
- SPIFF_FILES = files
- TERMINAL = $(PYTHON) -m serial.tools.miniterm $(COM_OPTS) $(COM_PORT) $(COM_SPEED_SERIAL)
- TRACE = 
RELINK_VARS:
- DISABLE_NETWORK = 1
- DISABLE_WIFI = 1
[... 49 lines cut ...]
- RBOOT_DIR = /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Components/rboot
- RBOOT_ROM0_ADDR = 0x00002000
- RBOOT_ROM1_ADDR = 
- RBOOT_ROM2_ADDR = 
- RBOOT_ROM_0_BIN = out/Esp8266/debug/firmware/rom0.bin
- RBOOT_ROM_1_BIN = 
- RBOOT_TWO_ROMS = 0
- SMING_ARCH = Esp8266
- SMING_ARCH_FULL = Esp8266
- SMING_CXX_STD = c++17
- SMING_C_STD = c11
- SMING_HOME = /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming
- SMING_RELEASE = 
- SMING_SOC = esp8266
- SPIFF_BIN_OUT = out/Esp8266/debug/firmware/spiff_rom.bin
- SPI_MODE = $(STORAGE_DEVICE_spiFlash_MODE)
- SPI_SIZE = $(STORAGE_DEVICE_spiFlash_SIZE)
- SPI_SPEED = $(STORAGE_DEVICE_spiFlash_SPEED)
- STRICT = 0
- TARGET_OUT_0 = out/Esp8266/debug/build/app_0.out
- TOOLS_BASE = $(SMING_HOME)/$(OUT_BASE)/tools
- UNAME = Darwin
- USER_LIBDIR = $(SMING_HOME)/$(OUT_BASE)/lib

make: Nothing to be done for `list-config'.`

IanAber, 2022-07-11:
I renames the sming directory to sming_complete and then ran the git clone command to create a new one. From the new sming/samples/basic_blink directory I executed 'make' and go this...

`Basic_Blink: Invoking 'all' for Esp8266 (debug) architecture
Using configuration 'standard'
Patching /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Components/.patches/http-parser/component.mk
Patching /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Libraries/.patches/Adafruit_BME280_Library/component.mk
Patching /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Libraries/.patches/Adafruit_BusIO/component.mk
Patching /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Libraries/.patches/Adafruit_GFX/component.mk
Patching /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Libraries/.patches/Adafruit_ILI9341/component.mk
Patching /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Libraries/.patches/Adafruit_NeoPixel/component.mk
Patching /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Libraries/.patches/Adafruit_PCD8544/component.mk
Patching /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Libraries/.patches/Adafruit_SSD1306/component.mk
Patching /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Libraries/.patches/Adafruit_ST7735/component.mk
Patching /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Libraries/.patches/Adafruit_Sensor/component.mk
Patching /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Libraries/.patches/Adafruit_VL53L0X/component.mk
Patching /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Libraries/.patches/ArduinoFFT/component.mk
Patching /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Libraries/.patches/IR/component.mk
Patching /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Libraries/.patches/ToneGenerator/component.mk

Fetching submodule 'rboot' ...
Submodule 'rboot' (https://github.com/mikee47/rboot) registered for path 'rboot'
Cloning into '/Users/ianaber/Documents/Projects/ESP8266-VSCode/Sming/Sming/Components/rboot/rboot'...
Submodule path 'rboot': checked out 'cc00071f593ef7d55c0c0a7b97828173f9cda89b'

Fetching submodule 'esptool2' ...
Submodule 'esptool2' (https://github.com/raburton/esptool2.git) registered for path 'esptool2'
Cloning into '/Users/ianaber/Documents/Projects/ESP8266-VSCode/Sming/Sming/Components/rboot/esptool2'...
Submodule path 'esptool2': checked out 'a550572b9de8c25f26b7a7675b47e58861227f64'
Checking patch elf.h...
Checking patch esptool2.c...
Checking patch esptool2.h...
Checking patch Makefile...
Applied patch elf.h cleanly.
Applied patch esptool2.c cleanly.
Applied patch esptool2.h cleanly.
Applied patch Makefile cleanly.

Fetching submodule 'sdk' ...
Submodule 'Esp8266.Sdk' (https://github.com/espressif/ESP8266_NONOS_SDK.git) registered for path 'sdk'
Cloning into '/Users/ianaber/Documents/Projects/ESP8266-VSCode/Sming/Sming/Arch/Esp8266/Components/esp8266/sdk'...
Submodule path 'sdk': checked out '3fe474e040450e0ab72ea75f8f00c3d9c6279025'
Checking patch lib/libc.a => lib/libc.orig.a...
Checking patch lib/libgcc.a => lib/libgcc.orig.a...
Checking patch include/version.h...
Applied patch lib/libc.a => lib/libc.orig.a cleanly.
Applied patch lib/libgcc.a => lib/libgcc.orig.a cleanly.
Applied patch include/version.h cleanly.

Fetching submodule 'esptool' ...
Submodule 'esptool' (https://github.com/espressif/esptool) registered for path 'esptool'
Cloning into '/Users/ianaber/Documents/Projects/ESP8266-VSCode/Sming/Sming/Components/esptool/esptool'...
Submodule path 'esptool': checked out 'c04d0f42db9e85b73171a2030bc2e9842ec476ac'

Fetching submodule 'spiffs' ...
Submodule 'Libraries.spiffs' (https://github.com/pellepl/spiffs.git) registered for path 'spiffs'
Cloning into '/Users/ianaber/Documents/Projects/ESP8266-VSCode/Sming/Sming/Libraries/Spiffs/spiffs'...
Submodule path 'spiffs': checked out 'ec68ba8208d7550860e4e78299d58a529b88bf85'
Checking patch src/spiffs.h...
Checking patch src/spiffs_hydrogen.c...
Checking patch src/spiffs_nucleus.c...
Applied patch src/spiffs.h cleanly.
Applied patch src/spiffs_hydrogen.c cleanly.
Applied patch src/spiffs_nucleus.c cleanly.
/Library/Developer/CommandLineTools/usr/bin/make components application

Building /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/out/Esp8266/debug/lib/clib-Storage-52412ccea9c0ff555e8ecb0b5573ceec.a
C+ /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Components/Storage/src/CustomDevice.cpp
In file included from /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Components/Storage/src/include/Storage/Device.h:12,
                 from /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Components/Storage/src/include/Storage/CustomDevice.h:7,
                 from /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Components/Storage/src/CustomDevice.cpp:5:
/Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Wiring/WString.h:64:10: fatal error: FlashString/String.hpp: No such file or directory
   64 | #include <FlashString/String.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [src/CustomDevice.o] Error 1
make[1]: *** [Storage-build] Error 2
make: *** [all] Error 2
`

IanAber, 2022-07-11:
In the new sming directory I ran 'make list-config' and got this...

`
Basic_Blink: Invoking 'list-config' for Esp8266 (debug) architecture

** Sming build configuration **

PREREQUISITES:
CUSTOM_TARGETS:
- /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Arch/Esp8266/Components/esp8266/sdk/bin/esp_init_data_default.bin
- /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Arch/Esp8266/Components/esp8266/sdk/bin/esp_init_data_vdd_default.bin
- out/Esp8266/debug/firmware/rboot.bin
- out/Esp8266/debug/firmware/rom0.bin
- partmap-build
LIBS:
- c
- crypto
- gcc
- hal
- lwip
- m
- main-cff65c904b5b4f06763e697bf9b55747
- phy
ARDUINO_LIBRARIES:
SUBMODULES:
- /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Arch/Esp8266/Components/esp8266/sdk
- /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Components/esptool/esptool
- /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Components/rboot/esptool2
- /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Components/rboot/rboot
- /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Libraries/Spiffs/spiffs
CONFIG_VARS:
- COM_SPEED_SERIAL = $(COM_SPEED)
- DEBUG_PRINT_FILENAME_AND_LINE = 0
- DEBUG_VERBOSE_LEVEL = 2
- ENABLE_CMD_EXECUTOR = 1
- ENABLE_GDB = 
- ENABLE_SPI_DEBUG = 0
- ENABLE_TASK_COUNT = 
- GDB_UART_SWAP = 
- HWCONFIG = standard
- HWCONFIG_OPTS = 
- LOCALE = 
- RBOOT_GPIO_ENABLED = 0
- RBOOT_GPIO_SKIP_ENABLED = 0
- RBOOT_RTC_ENABLED = 0
- RBOOT_SILENT = 0
- STRING_OBJECT_SIZE = 12
- TASK_QUEUE_LENGTH = 10
- USER_CFLAGS = 
- USE_US_TIMER = 1
CACHE_VARS:
- APP_NAME = app
- COM_OPTS = --raw --encoding ascii --rts 0 --dtr 0
- COM_PORT = /dev/tty.usbserial-0001
- COM_PORT_ESPTOOL = $(COM_PORT)
- COM_PORT_GDB = $(COM_PORT)
- COM_SPEED = 115200
- COM_SPEED_ESPTOOL = $(COM_SPEED)
- FULL_COMPONENT_BUILD = 
- GDB_CMDLINE = trap '' INT; $(GDB) -x $(GDBSTUB_DIR)/gdbcmds -b $(COM_SPEED_GDB) -ex "target remote $(COM_PORT_GDB)"
- KILL_TERM = pkill -9 -f "$(COM_PORT) $(COM_SPEED_SERIAL)" || exit 0
- PIP_ARGS = 
- SERVER_OTA_PORT = 9999
- SPIFF_BIN = spiff_rom
- SPIFF_FILES = files
- TERMINAL = $(PYTHON) -m serial.tools.miniterm $(COM_OPTS) $(COM_PORT) $(COM_SPEED_SERIAL)
- TRACE = 
RELINK_VARS:
- DISABLE_NETWORK = 1
- DISABLE_WIFI = 1
- ENABLE_CUSTOM_HEAP = 0
- ENABLE_CUSTOM_PWM = 1
- PARTITION_TABLE_OFFSET = 0x000fa000
- RBOOT_LD_TEMPLATE = $(RBOOT_DIR)/rboot.rom0.ld
- RBOOT_ROM_0 = rom0
- RBOOT_ROM_1 = rom1
- SPIFFS_OBJ_META_LEN = 16
- SPIFF_FILEDESC_COUNT = 7
- SPISOFT_DELAY_FIXED = 0
- SPISOFT_DELAY_VARIABLE = 0
DEBUG_VARS:
- APP_CFLAGS =  -DCOM_SPEED_SERIAL=$(COM_SPEED_SERIAL) -DSERIAL_BAUD_RATE=$(COM_SPEED_SERIAL) -DRBOOT_INTEGRATION -DGDBSTUB_GDB_PATCHED=0
- APP_LIBDIR = out/Esp8266/debug/lib
- AR = /opt/esp-quick-toolchain/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar
- ARCH_BASE = /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Arch/Esp8266
- AS = /opt/esp-quick-toolchain/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc
- AWK = POSIXLY_CORRECT= awk
- BLANK_BIN = /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Components/Storage/blank.bin
- BUILD_BASE = $(OUT_BASE)/build
- CC = /opt/esp-quick-toolchain/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc
- CLANG_FORMAT = clang-format
- CMAKE = cmake
- COM_SPEED_GDB = $(COM_SPEED_SERIAL)
- CXX = /opt/esp-quick-toolchain/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++
- ESPTOOL = /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Components/esptool/esptool/esptool.py
- ESPTOOL2 = /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/out/Esp8266/debug/tools/esptool2
- ESP_HOME = /opt/esp-quick-toolchain
- FLASH_INIT_DATA = $(SDK_BASE)/bin/esp_init_data_default.bin
- FLASH_INIT_DATA_VCC = $(SDK_BASE)/bin/esp_init_data_vdd_default.bin
- FW_BASE = $(OUT_BASE)/firmware
- GCC_VERSION = 10.2.0
- GDB = /opt/esp-quick-toolchain/xtensa-lx106-elf/bin/xtensa-lx106-elf-gdb
- GDBSTUB_DIR = /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Arch/Esp8266/Components/gdbstub
- GIT = git
- GLOBAL_CFLAGS = -DSMING_ARCH=$(SMING_ARCH) -DSMING_SOC=$(SMING_SOC) -DPROJECT_DIR=\"$(PROJECT_DIR)\" -DSMING_HOME=\"$(SMING_HOME)\" $(USER_CFLAGS) -D$(SMING_SOC_VAR)=1 -DENABLE_CMD_EXECUTOR=$(ENABLE_CMD_EXECUTOR) -DDISABLE_NETWORK=1 -DDISABLE_WIFI=1 -DDEBUG_PRINT_FILENAME_AND_LINE=$(DEBUG_PRINT_FILENAME_AND_LINE) -DCUST_FILE_BASE=$$* -DDEBUG_VERBOSE_LEVEL=$(DEBUG_VERBOSE_LEVEL) -DSTRING_OBJECT_SIZE=$(STRING_OBJECT_SIZE)  -DUSE_US_TIMER
- LD = /opt/esp-quick-toolchain/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc
- LIBMAIN_COMMANDS = $(NOWIFI_LIBMAIN_COMMANDS) $(APP_LIBMAIN_COMMANDS)
- NINJA = ninja
- NM = /opt/esp-quick-toolchain/xtensa-lx106-elf/bin/xtensa-lx106-elf-nm
- OBJCOPY = /opt/esp-quick-toolchain/xtensa-lx106-elf/bin/xtensa-lx106-elf-objcopy
- OBJDUMP = /opt/esp-quick-toolchain/xtensa-lx106-elf/bin/xtensa-lx106-elf-objdump
- OS = 
- OUT_BASE = out/Esp8266/debug
- PROJECT_DIR = /Users/ianaber/Documents/Projects/ESP8266-VSCode/Sming/samples/Basic_Blink
- PROJECT_SOC = esp8266
- PYTHON = python3
- RBOOT_BIG_FLASH = 0
- RBOOT_DIR = /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming/Components/rboot
- RBOOT_ROM0_ADDR = 0x00002000
- RBOOT_ROM1_ADDR = 
- RBOOT_ROM2_ADDR = 
- RBOOT_ROM_0_BIN = out/Esp8266/debug/firmware/rom0.bin
- RBOOT_ROM_1_BIN = 
- RBOOT_TWO_ROMS = 0
- SMING_ARCH = Esp8266
- SMING_ARCH_FULL = Esp8266
- SMING_CXX_STD = c++17
- SMING_C_STD = c11
- SMING_HOME = /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming
- SMING_RELEASE = 
- SMING_SOC = esp8266
- SPIFF_BIN_OUT = out/Esp8266/debug/firmware/spiff_rom.bin
- SPI_MODE = $(STORAGE_DEVICE_spiFlash_MODE)
- SPI_SIZE = $(STORAGE_DEVICE_spiFlash_SIZE)
- SPI_SPEED = $(STORAGE_DEVICE_spiFlash_SPEED)
- STRICT = 0
- TARGET_OUT_0 = out/Esp8266/debug/build/app_0.out
- TOOLS_BASE = $(SMING_HOME)/$(OUT_BASE)/tools
- UNAME = Darwin
- USER_LIBDIR = $(SMING_HOME)/$(OUT_BASE)/lib

make: Nothing to be done for `list-config'.`

IanAber, 2022-07-13:
Any updates on this installation problem?
Ian Abercrombie

***@***.*** ***@***.***>
Phone: +1 (352) 281-6286
www.CedarTechnology.com <http://www.cedartechnology.com/>

mikee47, 2022-07-13:
I think the issue could be `SMING_HOME`, final 'sming' directory should be 'Sming', i.e.

` /Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming` 
NOT
` /Users/ianaber/documents/projects/ESP8266-VSCode/sming/Sming`

mikee47, 2022-07-13:
I think the issue could be `SMING_HOME`, final 'sming' directory should be 'Sming', i.e.

`/Users/ianaber/documents/projects/ESP8266-VSCode/sming/Sming`

NOT `/Users/ianaber/documents/projects/ESP8266-VSCode/sming/sming`

IanAber, 2022-07-13:
My goodness! You are a bona-fide genius! How the heck did you figure that one out? Since I am running on OS X which is a case insensitive file system it never occurred to me that this might be an issue. I guess it is not getting through some of the make or python code. I fixed the SMING_HOME environment variable and it compiles the samples and my code fine now.

As a suggestion, a note stating that all entries should be checked for case accuracy would be useful troubleshooting help in the article.

Also, the later versions of OS X use zsh not bash so where your instructions talk about editing the ~/.profile file to set the environment up they should now mention that these statements should be moved to ~/.zshenv instead.

Thanks for all the fantastic work you are doing and hopefully, sometime soon, I will be able to contribute myself too.

slaff, 2022-07-14:
@IanAber can you please add the recommended changes to the documentation as PR? The documentation can be found under `docs/` and the MacOS installation guide is in `docs/source/getting-started/macos/index.rst`

mikee47, 2024-06-22:
As a side note clang can catch stuff like this, so with #2804 should help.

closed 2024-06-22 by mikee47