SmingHub/Sming #2541: Sming esp-idf esp32 toolchain path

kmihaylov, 2022-08-19:
On my freshly installed Devuan machine, I found that the ```export.sh``` script doesn't add the xtensa's path ```/esp32/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/```.  The ```install.sh``` script exits normally.

My app builds normally once I add the path above to the PATH.

mikee47, 2022-08-20:
The compiler isn't added to the system PATH by the installer as it depends which variant you're building on (esp32, esp32c3, esp32s3, etc.) This is handled in `$SMING_HOME/Sming//Arch/Esp32/build.mk`. To help resolve this issue it would be helpful if you could post some further details:

- Dump of errors generated
- Output from `make list-config`

Thanks.

kmihaylov, 2022-08-20:
Good morning!

Forgive my lack of correctness in toolchain/compiler wording.

When I have manually added the compiler path (not to the login scripts, but by hand) and I have built some app, after logout/login Sming successfully compiles the app again. So maybe on the first build the magic about the compiler path happens.

Therefore I completely wiped esp32 and Sming from the disk and did it again. Here is the output:

```
$ make list-config 

fw: Invoking 'list-config' for Esp32/esp32 (debug) architecture
Executing action: reconfigure
Running cmake in directory /home/opt/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/opt/Sming/Sming/Arch/Esp32/Components/esp32/project/esp32/debug"...
-- Found Git: /usr/bin/git (found version "2.30.2") 
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: xtensa-esp32-elf-gcc
CMake Error at /opt/esp-idf/tools/cmake/project.cmake:283 (__project):
  The CMAKE_C_COMPILER:

    xtensa-esp32-elf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:9 (project)

CMake Error at /opt/esp-idf/tools/cmake/project.cmake:283 (__project):
  The CMAKE_CXX_COMPILER:

    xtensa-esp32-elf-g++

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:9 (project)

CMake Error at /opt/esp-idf/tools/cmake/project.cmake:283 (__project):
  The CMAKE_ASM_COMPILER:

    xtensa-esp32-elf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:9 (project)

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!
See also "/home/opt/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/CMakeFiles/CMakeOutput.log".
See also "/home/opt/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/CMakeFiles/CMakeError.log".
cmake failed with exit code 1

** Sming build configuration **

PREREQUISITES:
CUSTOM_TARGETS:
- check-ssl
[... 268 lines cut ...]
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:9 (project)

CMake Error at /opt/esp-idf/tools/cmake/project.cmake:283 (__project):
  The CMAKE_ASM_COMPILER:

    xtensa-esp32-elf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:9 (project)

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!
See also "/home/opt/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/CMakeFiles/CMakeOutput.log".
See also "/home/opt/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/CMakeFiles/CMakeError.log".
cmake failed with exit code 1
make: *** [/home/opt/Sming/Sming/Arch/Esp32/Components/esp32/component.mk:311: /home/opt/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/config/sdkconfig.h] Error 2
```

slaff, 2022-11-08:
@kmihaylov can you check if PR #2580 fixes the issue for you?

kmihaylov, 2022-11-09:
Hi @slaff, excuse my delay.

Yes, your PR solves the problem.

Greetings!

PR #2580 (2022-11-10): Fix issue with esp32 paths.
Fixes #2541
closed 2022-11-10 by slaff