SmingHub/Sming #2542: [ESP8266] Sending single or dissimilar UDP datagrams does not work

MayaPosch, 2022-08-24:
**Summary:**

* Trying to send UDP message using UdpConnection.
* When calling UdpConnection::sendStringTo() once, no message is sent (confirmed with Wireshark).
* When calling UdpConnection::sendStringTo() with a timer, all messages are sent.

**Environment:**

* Targeting ESP8266, tested on NodeMCU board.
* Using latest develop branch on Ubuntu 20.04.

**Minimal sample:**

```cpp
UdpConnection udp;
Timer procTimer;
uint32_t count = 0;

	
void sendMessage() {
	IpAddress remoteIP(192, 168, 0, 100);
	String text("Hi.");
	text.concat(count++);
	uint16_t remotePort = 1234;
	udp.sendStringTo(remoteIP, remotePort, text);
}

void gotIP(IpAddress ip, IpAddress gateway, IpAddress netmask) {
	Serial.println("\r\n=== UDP CLIENT STARTED ===");
	Serial.println(WifiStation.getIP());
	Serial.println("=============================\r\n");
	
	sendMessage();
	
	//procTimer.initializeMs(10 * 1000, sendMessage).start();
	
	while (1) {
		delay(5000);
		Serial.println(_F("Idle..."));
	}
}

void init() {
	Serial.begin(SERIAL_BAUD_RATE); // 115200 by default
	Serial.systemDebugOutput(true);

	WifiStation.config(WIFI_SSID, WIFI_PWD);
	WifiStation.enable(true);
	WifiEvents.onStationGotIP(gotIP);
}
```

With the procTimer running, all datagrams are sent as expected. With it commented out and the while loop used instead, no datagram is sent.

**Expectation:**

Calling UdpConnection::sendStringTo() once sends a datagram.

**Notes:**

* I have tried using the raw UDP API directly too, with the same result.
* Calling UdpConnection::sendStringTo() or udp_sendto() sequentially for different messages also seems to trigger this behaviour, where none of the packets are sent.
* Suspicion is that the running timer is needed to make things work, but haven't tested this theory yet.

mikee47, 2022-08-24:
Thank you for such a clear and concise report! The issue is that the `while` loop block all other activity; you must not do this in Sming! See https://sming.readthedocs.io/en/latest/information/multitasking.html

If you remove this section, your code should work:

```
	while (1) {
		delay(5000);
		Serial.println(_F("Idle..."));
	}
```

MayaPosch, 2022-08-24:
Obvious problem in hindsight, and I can send and receive UDP datagrams now. Using LwIP without an RTOS takes a bit more work, indeed :)

Only problem that I'm dealing with currently now is that I'm trying to receive small packets of binary data (<100 bytes), but they arrive rather mangled with sections missing and whitespace inserted.

I imagine that's an unrelated issue, though.

closed 2022-08-25 by MayaPosch