SmingHub/Sming #2544: ESP8266 Troubleshooting Make Error

IanAber, 2022-08-25:
I am getting an error at the end of flashing an application to an ESP8266. The code compiles and appears to flash but finally returns an error. How can I figure out what the error means and what its cause might be? The final error is Make Flash Error 134. What does that refer to? Where can I find more information on Error 134?

`make flash

temperatures: Invoking 'flash' for Esp8266 (debug) architecture
/Library/Developer/CommandLineTools/usr/bin/make components application

Building /Users/ianaber/Documents/Projects/ESP8266-VSCode/temperatures/out/Esp8266/debug/lib/clib-App-61ef69595f6ba6fff7ed06b5df1749a9.a
make[2]: Nothing to be done for `build'.
Validating hardware config 'spiffs-2m'
Verifying partition table...
Writing to 'out/Esp8266/debug/firmware/partitions.bin'
temperatures: Linking out/Esp8266/debug/build/app_0.out
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE87E8|    2024
    rodata|           ReadOnly Data (RAM)|    3FFE87F0|    3FFE91B8|    2504
       bss|      Uninitialized Data (RAM)|    3FFE91B8|    3FFEFBE8|   27184
      text|            Cached Code (IRAM)|    40100000|    40106FB6|   28598
irom0_text|           Uncached Code (SPI)|    40202010|    40267F19|  417545
Total Used RAM : 31712
Free RAM : 50208
Free IRam : 4188
ESPTOOL2 out/Esp8266/debug/firmware/rom0.bin
SSL support disabled
make[1]: Nothing to be done for `application'.
Killing Terminal to free /dev/tty.usbserial-0001
WriteFlash 0x00000=out/Esp8266/debug/firmware/rboot.bin 0x001fa000=out/Esp8266/debug/firmware/partitions.bin 0x00002000=out/Esp8266/debug/firmware/rom0.bin 0x00100000=out/Esp8266/debug/firmware/spiff_rom.bin  0x001fc000=/Users/ianaber/documents/projects/ESP8266-VSCode/sming/Sming/Arch/Esp8266/Components/esp8266/sdk/bin/esp_init_data_default.bin 
python3 /Users/ianaber/documents/projects/ESP8266-VSCode/sming/Sming/Components/esptool/esptool/esptool.py -p /dev/tty.usbserial-0001 -b 115200 --chip esp8266 --before default_reset --after hard_reset write_flash -z -fs 2MB -ff 40m -fm dio 0x00000 out/Esp8266/debug/firmware/rboot.bin 0x001fa000 out/Esp8266/debug/firmware/partitions.bin 0x00002000 out/Esp8266/debug/firmware/rom0.bin 0x00100000 out/Esp8266/debug/firmware/spiff_rom.bin  0x001fc000 /Users/ianaber/documents/projects/ESP8266-VSCode/sming/Sming/Arch/Esp8266/Components/esp8266/sdk/bin/esp_init_data_default.bin  
esptool.py v3.2-dev
Serial port /dev/tty.usbserial-0001
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 60:01:94:0f:4e:30
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00000fff...
Flash will be erased from 0x001fa000 to 0x001fafff...
Flash will be erased from 0x00002000 to 0x00070fff...
Flash will be erased from 0x00100000 to 0x0017ffff...
Flash will be erased from 0x001fc000 to 0x001fcfff...
Flash params set to 0x0230
Compressed 1888 bytes to 1414...
Wrote 1888 bytes (1414 compressed) at 0x00000000 in 0.2 seconds (effective 63.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 150...
Wrote 3072 bytes (150 compressed) at 0x001fa000 in 0.1 seconds (effective 188.7 kbit/s)...
Hash of data verified.
Compressed 450752 bytes to 292791...
Wrote 450752 bytes (292791 compressed) at 0x00002000 in 25.9 seconds (effective 139.0 kbit/s)...
Hash of data verified.
Compressed 524288 bytes to 54453...
Wrote 524288 bytes (54453 compressed) at 0x00100000 in 6.8 seconds (effective 620.0 kbit/s)...
Hash of data verified.
Compressed 128 bytes to 75...
Wrote 128 bytes (75 compressed) at 0x001fc000 in 0.1 seconds (effective 9.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
trap '' INT; /opt/esp-quick-toolchain/xtensa-lx106-elf/bin/xtensa-lx106-elf-gdb -x /Users/ianaber/documents/projects/ESP8266-VSCode/sming/Sming/Arch/Esp8266/Components/gdbstub/gdbcmds -b 115200 -ex "target remote /dev/tty.usbserial-0001"
GNU gdb (GDB) 8.2.50.20190202-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-apple-darwin14 --target=xtensa-lx106-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
/bin/sh: line 1: 31290 Abort trap: 6           /opt/esp-quick-toolchain/xtensa-lx106-elf/bin/xtensa-lx106-elf-gdb -x /Users/ianaber/documents/projects/ESP8266-VSCode/sming/Sming/Arch/Esp8266/Components/gdbstub/gdbcmds -b 115200 -ex "target remote /dev/tty.usbserial-0001"
make: *** [flash] Error 134`

IanAber, 2022-11-10:
I finally found the problem, or at least the solution.

I deleted the 'out' folder in the project and executed the Flash command. This time it recompiled everything and successfully ran.

I still have no idea what the error is telling me or why it should happen but even a full clean did not fix it.

mikee47, 2024-01-09:
@IanAber Sorry for the lack of response. The error is from GDB (debugger) which means code is built with `ENABLE_GDB=1`. Just run `make ENABLE_GDB=0` to revert to normal.

See also https://sming.readthedocs.io/en/latest/_inc/Sming/Arch/Esp8266/Components/gdbstub/index.html.

The error you see is from the debugger. Running with `make gdb V=1` might give more information about the problem.

mikee47, 2024-06-22:
Closing as issue resolved.

closed 2024-06-22 by mikee47