SmingHub/Sming #2547: Unable to mount SPIFFS partition. Porting from 3.8 to current.

MayaPosch, 2022-09-04:
**Description:** 

I have an existing project that I am updating to the current development branch. So far I have updated all of the APIs, but I'm finding that it's not clear how to change the mounting of SPIFFS. This project uses an rBoot setup, targeting 4 MB ESP8266 boards, with two 1 MB firmware slots, each with an accompanying 1 MB Flash storage as SPIFFS.

Previously I would mount this SPIFFS section using `spiffs_mount_manual`, as described in issue #1009, to work around issues with SPIFFS mount while using rBoot.This firmware worked without issues for years.

**Issues:**

I had hoped to find an easy description of how to convert from this setup, but between Basic_Storage, IFS and other new features I'm not sure how all of that relates, and what is relevant for a basic SPIFFS configuration to just read and write some configuration files and SSL certificates.

There is the rBoot and OTA updates page: https://sming.readthedocs.io/en/latest/information/rboot-ota.html

It lists a SPIFFS mounting example:

```cpp
int slot = rboot_get_current_rom();
// Find the n'th SPIFFS partition
auto part = PartitionTable().find(Partition::SubType::Data::spiffs, slot);
if(part) {
   //debugf("trying to mount SPIFFS at %x, length %d", part.address(), part.size());
   spiffs_mount(part);
} else {
   debug_e("SPIFFS partition missing for slot #%u", slot);
}
```

However, when I try to use this, I get a compile error. Adding the missing `Storage::` scope specifier I get another compile error, about the PartitionTable constructor requiring a parameter. This example seems defunct.

I have added `HWCONFIG += spiffs` to the project's component.mk. 

**Question:**

What is the right way to port the old style `spiffs_mount_manual` code to the current style, without rewriting everything FS-related in the code?

mikee47, 2022-09-04:
You can find an overview of the Storage management changes in https://sming.readthedocs.io/en/latest/upgrading/4.2-4.3.html. First and most important the hardware config needs to match your expectations. Setting `HWCONFIG := spiffs` provides only one SPIFFS partition, so you will need to create a custom config for your project. You can find an example for this in the `Basic_Storage` sample application.

You might like to edit the partition layout graphically using `make hwconfig-edit`. Remember to save it in your project directory when done. For example, `myconfig.hw`, then set `HWCONFIG := myconfig`.

The default SPIFFS partition is called `spiffs0`, so good practice would be to call the second one `spiffs`. This is the approach demonstrated in the `Basic_Ota` sample (although looking at it now the sample is missing a definition for `spiffs` in it's config!)
You can then use this label in code to uniquely identify which partition to mount:

```
Storage::Partition findSpiffsPartition(Storage::Partition partition)
{
	String name = F("spiffs");
	name += ota.getSlot(partition);
	auto part = Storage::findPartition(name);
	if(!part) {
		debug_w("Partition '%s' not found", name.c_str());
	}
	return part;
}
```

Thanks for advising about the outdated example, will update that information.

mikee47, 2022-09-04:
@MayaPosch Sorry, posted prematurely, have updated.

MayaPosch, 2022-09-04:
Thank you. I have added the ota.hw file from the Basic_Ota example, and am using it along with the Ota component as in the example.

Unfortunately, despite using the same code to find and mount the SPIFFS as in the Basic_Ota example, no partition is found. After adapting the code to fit the code that you gave (searching for 'spiffs0'), still no partition is found.

Partitions.bin is flashed at 0x0000 2000, which I presume is correct. At this point I'm not sure what else may be wrong.

Here's the current revision of the project: https://github.com/MayaPosch/BMaC/tree/master/esp8266

mikee47, 2022-09-05:
I've created a PR with a few changes, should get things going!

MayaPosch, 2022-09-05:
Thank you. I have merged and tested the changes. Unfortunately the behaviour persists: no partition called 'spiffs0' is ever found. Does this have to be defined explicitly in the partition layout as well?

mikee47, 2022-09-05:
`spiffs0` is defined in the base `spiffs` map. Run `make map` to check:

```
Partition map: ota
options: 4m, spiffs
Device            Start       End         Size        Type      SubType   Name              Filename
----------------  ----------  ----------  ----------  --------  --------  ----------------  ------------
spiFlash          0x00000000  0x00001fff          8K                      Boot Sector
spiFlash          0x00002000  0x000f9fff        992K  app       ota_0     rom0              $(RBOOT_ROM_0_BIN)
spiFlash          0x000fa000  0x00101fff         32K                      (unused)
spiFlash          0x00102000  0x001f9fff        992K  app       ota_1     rom1              $(RBOOT_ROM_0_BIN)
spiFlash          0x001fa000  0x001fffff         24K                      (unused)
spiFlash          0x00200000  0x0027ffff        512K  data      spiffs    spiffs0           $(SPIFF_BIN_OUT)
spiFlash          0x00280000  0x002fffff        512K  data      spiffs    spiffs1
spiFlash          0x00300000  0x003f9fff       1000K                      (unused)
spiFlash          0x003fa000  0x003fafff          4K                      Partition Table
spiFlash          0x003fb000  0x003fbfff          4K  data      rfcal     rf_cal
spiFlash          0x003fc000  0x003fcfff          4K  data      phy       phy_init          $(FLASH_INIT_DATA)
spiFlash          0x003fd000  0x003fffff         12K  data      sysparam  sys_param
```

Have you run `make flashmap` to write the new partition table to the device?

MayaPosch, 2022-09-05:
Probably was apparently the old build files that did not get cleaned with `make clean` or a regular build.

Before, `make map` shows the reason why 'spiffs0' was not found:

```
esp8266: Invoking 'map' for Esp8266 (debug) architecture
Partition map: standard
options: 
Device            Start       End         Size        Type      SubType   Name              Filename
----------------  ----------  ----------  ----------  --------  --------  ----------------  ------------
spiFlash          0x00000000  0x00001fff          8K                      Boot Sector       
spiFlash          0x00002000  0x000f9fff        992K  app       factory   rom0              $(RBOOT_ROM_0_BIN)
spiFlash          0x000fa000  0x000fafff          4K                      Partition Table   
spiFlash          0x000fb000  0x000fbfff          4K  data      rfcal     rf_cal            
spiFlash          0x000fc000  0x000fcfff          4K  data      phy       phy_init          $(FLASH_INIT_DATA)
spiFlash          0x000fd000  0x000fffff         12K  data      sysparam  sys_param         
```

After `rm -rf out` and running `make`, I now get the following on `make map`:

```
Partition map: ota
options: 4m, spiffs
Device            Start       End         Size        Type      SubType   Name              Filename
----------------  ----------  ----------  ----------  --------  --------  ----------------  ------------
spiFlash          0x00000000  0x00001fff          8K                      Boot Sector       
spiFlash          0x00002000  0x000f9fff        992K  app       ota_0     rom0              $(RBOOT_ROM_0_BIN)
spiFlash          0x000fa000  0x00101fff         32K                      (unused)          
spiFlash          0x00102000  0x001f9fff        992K  app       ota_1     rom1              $(RBOOT_ROM_0_BIN)
spiFlash          0x001fa000  0x001fffff         24K                      (unused)          
spiFlash          0x00200000  0x0027ffff        512K  data      spiffs    spiffs0           $(SPIFF_BIN_OUT)
spiFlash          0x00280000  0x002fffff        512K  data      spiffs    spiffs1           
spiFlash          0x00300000  0x003f9fff       1000K                      (unused)          
spiFlash          0x003fa000  0x003fafff          4K                      Partition Table   
spiFlash          0x003fb000  0x003fbfff          4K  data      rfcal     rf_cal            
spiFlash          0x003fc000  0x003fcfff          4K  data      phy       phy_init          $(FLASH_INIT_DATA)
spiFlash          0x003fd000  0x003fffff         12K  data      sysparam  sys_param 
```

I suspect it will work now. I'll report back with my findings.

MayaPosch, 2022-09-05:
I can confirm that the firmware can now mount the SPIFFS partition and successfully read configuration files from it.

Closing this issue now as it's been resolved. Thank you very much for the assistance, @mikee47 :)

closed 2022-09-05 by MayaPosch