SmingHub/Sming #2550: DHCP won't execute once WiFi credentials were saved

kmihaylov, 2022-09-08:
Stuck in tricky problem that surprised me just now.

With ESP32 I use small class to handle WiFi options (set callbacks, use SmartConfig and etc).

When booting the device I can successfully set the WiFi AP and password using SmartConfig. The device connects to the network and the app works fine.

On the second reboot however the device connects to the AP, but won't execute onConnect callback. Here is the debug output:
```
W (564) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (0) cpu_start: Starting scheduler on APP CPU.
Welcome to the Stream Command executor
364619 open('pt1000_offset.txt'): NOT_FOUND (-11002)
392598 open('pressure_measuredTbl.json'): NOT_FOUND (-11002)
419118 open('pressure_calibratedTbl.json'): NOT_FOUND (-11002)
446024 open('pressure_offset.txt'): NOT_FOUND (-11002)
01/01/1970 00:00:00
E (1131) wifi:sta is connecting, return error
I (2891) wifi:new:<10,2>, old:<1,0>, ap:<255,255>, sta:<10,2>, prof:1
I (2891) wifi:state: init -> auth (b0)
I (2891) wifi:state: auth -> assoc (0)
I (2901) wifi:state: assoc -> run (10)
I (3021) wifi:connected with mechka, aid = 20, channel 10, 40D, bssid = 7c:8b:ca:a6:cf:96
I (3021) wifi:security: WPA2-PSK, phy: bgn, rssi: -88
I (3021) wifi:pm start, type: 1

W (3041) wifi:<ba-add>idx:0 (ifx:0, 7c:8b:ca:a6:cf:96), tid:0, ssn:3, winSize:64
I (3111) wifi:AP's beacon interval = 102400 us, DTIM period = 1
E (3671) esp_netif_lwip: dhcpc cb: failed to post got ip event (107)
```

May this issue be related to #2500 ? The error ```esp_netif_lwip: dhcpc cb: failed to post got ip event``` didn't return any meaningful directions on the problem. I also forced station only mode, but the problem still exists. I have ESP32 boards running on the same network but don't remember what IDF and Sming versions they use. This problem appeared just now, however I didn't do any WiFi related changes to the app.

Any directions are welcome.

--------------

For the record, I updated Sming to the current version, deleted ```esp-idf``` and ```esp32``` directories, ```Sming-clean``` ```dist-clean``` and ```clean``` and the problem still exists.

mikee47, 2022-09-10:
> E (3671) esp_netif_lwip: dhcpc cb: failed to post got ip event (107)
0x107 is ESP_ERR_TIMEOUT, looks like it's returned from `esp_event_post_to` (in IDF `components/esp_event/esp_event.c`). There's either a problem getting hold of a semaphore, or the queue is full...

mikee47, 2022-09-10:
Perhaps try setting `SYSTEM_EVENT_QUEUE_SIZE` in SDK config to a larger value (say, 64) to see if that makes any difference?

mikee47, 2022-09-10:
Also, can you try patch in #2543 then rebuild with `make CREATE_EVENT_TASK=1` ?

kmihaylov, 2022-09-10:
Sure, I'll try. But first I need to test with another wifi network/router to see if the problem still exists. Many thanks!

kmihaylov, 2022-09-13:
@mikee47 I'm on #2543 and I rebuild the app with ```CREATE_EVENT_TASK=1``` and I don't see the 0x107 error. My MQTT client tries to connect, so I guess the app successfully continues execution.

I didn't change the ```SYSTEM_EVENT_QUEUE_SIZE```.

Also it restarts well.

kmihaylov, 2022-09-20:
I'm searching the dhcp related functions in order to put them in another task, to verify #2550. I tried to find when exactly is the DHCP client triggered, expecting it in the event handler with event ```WIFI_EVENT_STA_CONNECTED```.
The only place ```WIFI_EVENT_STA_CONNECTED``` is used is inside the ```StationImpl::startScan()``` method thas I don't use in my app.

So I grep'ed the code for ```esp_netif_dhcpc_start()``` and again, the only place I find it was in ```StationImpl::enableDHCP()```. So far so good.
Now ```enableDHCP()``` is used only in ```StationImpl::setIP()```. For a moment I thought that maybe somewhere in the code ```setIP()``` is called on purpose with invalid IP, so the if() clause inside ```setIP()``` would call ```enableDHCP(true)```.
Digging Sming for ```setIP()``` didn't result in any code that looks to be used on successfull ```WIFI_EVENT_STA_CONNECTED``` callback, not to say that according to me it looks provided for use to the app on top of Sming, and not especially used inside the framework.

Obviously the DHCP client is triggered, but I can't understand how.

In Espresiff's example also the DHCP is magically triggered: https://github.com/espressif/esp-idf/blob/master/examples/wifi/getting_started/station/main/station_example_main.c (note the absence of ```WIFI_EVENT_STA_CONNECTED```).

I couldn't find any DHCP related options in ```wifi_init_config_t``` and ```wifi_config_t``` as well...

PS: I just read in the esp-idf's doc that ```SYSTEM_EVENT_STA_CONNECTED``` triggers ```esp_netif_dhcpc_start()``` so I'll continue digging.

mikee47, 2022-09-20:
The line `E (1131) wifi:sta is connecting, return error` might be related to PR #2474 which added auto-connect support. Perhaps try disabling auto-connect? (Parameter to `WifiStation.config`)

kmihaylov, 2022-09-21:
Actually it is related to my code. I'm doing
```
WifiStation.enable(true);

	if(WifiStation.getSSID() != nullptr) {
		Serial.printf("Saved SSID is: %s\r\n", WifiStation.getSSID().c_str());
		//WifiStation.connect();
	} else ...
```

Commenting out ```WifiStation.connect()``` solves the error mentioned by you.

Here is the output:
```
I (0) cpu_start: Starting scheduler on APP CPU.
Welcome to the Stream Command executor
359317 open('pt1000_offset.txt'): NOT_FOUND (-11002)
387263 open('pressure_measuredTbl.json'): NOT_FOUND (-11002)
413695 open('pressure_calibratedTbl.json'): NOT_FOUND (-11002)
440066 open('pressure_offset.txt'): NOT_FOUND (-11002)
606379 open('shutdownFulltopic.conf'): NOT_FOUND (-11002)
01/01/1970 00:00:00
I (3204) wifi:new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (3204) wifi:state: init -> auth (b0)
I (3214) wifi:state: auth -> assoc (0)
I (3214) wifi:state: assoc -> run (10)
I (3324) wifi:connected with VIVACOM_FiberNet_392A, aid = 1, channel 11, BW20, bssid = e4:da:df:ff:31:f9
I (3324) wifi:security: WPA2-PSK, phy: bgn, rssi: -55
I (3324) wifi:pm start, type: 1

I (3414) wifi:AP's beacon interval = 102400 us, DTIM period = 1
W (3454) wifi:<ba-add>idx:0 (ifx:0, e4:da:df:ff:31:f9), tid:0, ssn:2, winSize:64
E (4174) esp_netif_lwip: dhcpc cb: failed to post got ip event (107)
```

Now, the device gets a IP address, I can ping it, but none of the services that should have been run by ```onStationGotIP``` is started.

Since nobody has reported dhcpc problems I get the feeling it is something in my wifi code that isn't proper. I might prepare minimal example to prove that. But my code works with ```CREATE_EVENT_TASK=1```...?

kmihaylov, 2022-09-21:
Got it working. My WifiHelper class was dynamically allocated (for some reason I moved it from static with Esp32)... Now moved it back to static, got out all Wifi functions out of the constructor and did ```start()``` in the last lines of ```init()```... The Wifi* methods I called in the constructor didn't succeed because of ```ESP_ERR_WIFI_NOT_INIT```, maybe I was lazy to get them out earlier.

I hope it will work flawlessly.

Maybe some IDF internals are too sensitive when related to the hardware/events/etc...

@mikee47 thanks again for your support!

closed 2022-09-21 by kmihaylov