SmingHub/Sming #2574: can't make a member function the MessgeHandler callback

pljakobs, 2022-10-27:
Hi all, I have started to port an older SMING project (based on 4.2) over to the current 4.6 (mainly because I want it to compile on the esp32c3 SOC). 
This project uses mqtt and more or less, follows the mqtt example from the SMING sampels, with one notable change: the mqtt code is encapsulated in an AppMqttClient class.

Therefore, the handler function is  a class member and I get a compiler error:
```
mqtt.cpp:104:61: error: invalid use of non-static member function 'int AppMqttClient::onMessageReceived(MqttClient, mqtt_message_t*)'
     mqtt->setMessageHandler(AppMqttClient::onMessageReceived);
```
the calling function is 
```
void AppMqttClient::start() {
    debug_i("Start MQTT");

    delete mqtt;
    mqtt = new MqttClient();
    //mqtt->setCallback(MqttStringSubscriptionCallback(&AppMqttClient::onMessageReceived, this));
    mqtt->setMessageHandler(AppMqttClient::onMessageReceived);
    connectDelayed(2000);
}
```

and the handler function is defined as 
```
int AppMqttClient::onMessageReceived(MqttClient client, mqtt_message_t* msg) {
    String topic=String(MqttBuffer(msg->publish.topic_name));
    String message=String(MqttBuffer(msg->publish.content));
```

I assume this is more due to the c++ object model/restrictions but - I think having a member function being the handler is a perfectly valid use case.

Is there a way to make this work in my code?

best

pj

mikee47, 2022-10-27:
Use this code:

```
mqtt->setCallback(MqttDelegate(&AppMqttClient::onMessageReceived, this));
```

Method callbacks require two parameters: (1) a pointer to the class method, and (2) a pointer to the class instance itself.

mikee47, 2022-10-27:
NB. See https://sming.readthedocs.io/en/latest/upgrading/4.4-4.5.html?highlight=MqttStringSubscriptionCallback#breaking-changes

mikee47, 2022-10-27:
Edit: Sorry, use `setEventHandler` not `setCallback`

pljakobs, 2022-10-27:
Hi mikee47, thanks for the quick response. 
That was one version I had tried before, seeing that a bit further up, a similiar construct was being used 
```  
mqtt->setCompleteDelegate(TcpClientCompleteDelegate(&AppMqttClient::onComplete, this));
```
so, copying and pasting your code, 
```
mqtt->setMessageHandler(MqttDelegate(&AppMqttClient::onMessageReceived, this));
```
I get a slightly different error:
```
mqtt.cpp:104:81: error: no matching function for call to 'Delegate<int(MqttClient&, mqtt_message_u*)>::Delegate(int (AppMqttClient::*)(MqttClient, mqtt_message_t*), AppMqttClient*)'
     mqtt->setMessageHandler(MqttDelegate(&AppMqttClient::onMessageReceived, this));
```
I'm sure now, that there's something important that I'm not seeing - but I'm not seeing it.

mikee47, 2022-10-27:
Your message handler signature needs fixing, requires `MqttClient&` (not `MqttClient`)

pljakobs, 2022-10-27:
sigh. 
that's it. 
Thanks mikee47. I was blind from staring at that bit of code.
The quick help is highly appreciated.

pj

closed 2022-10-27 by pljakobs