SmingHub/Sming #2594: CI: Windows Host Builds Taking Forever

slaff, 2022-12-08:
Sometimes the windows-host build is taking forever. The reason for it seems to be the tests execution. Might be due to a deadlock in the host emulator.

More about the issue here:

[image]

Example: https://github.com/SmingHub/Sming/actions/runs/3646634352/jobs/6157931489
Output:
```
Background timer: ticks 2478726, time 496ms, heap free 117509, used 13563, peak 24847

40754761 showFontStyle: notoSans15 DotMatrix (32 of 50)

40957857 showFontStyle: notoSans15 HLine (32 of 50)

signal_handler: SIGINT - Ctrl+C pressed
returning 0
make[3]: *** [run] Interrupt
make[2]: *** [execute] Interrupt
make[1]: *** [Libraries/Graphics/test/.host.complete] Interrupt
make: *** [tests] Interrupt
Terminate batch job (Y/N)? 

D:\a\Sming\Sming\Sming>echo Failed with error #66048. 
Failed with error #66048.

D:\a\Sming\Sming\Sming>exit /b 66048 
```

slaff, 2022-12-08:
@mikee47 when you have time can you look at the issue above?

mikee47, 2023-10-11:
Fixed in #2606.

closed 2023-10-11 by mikee47