SmingHub/Sming #2597: *** Preparation for Release 4.8.0 ***

slaff, 2022-12-19:
This issue is created to collect all the tasks we need to do before the 4.7.0 release.

### Source Code
- [ ] Update Sming/Core/SmingVersion.h.
- [ ] Update Readme.md.
- [ ] Merge develop into master.
- [ ] Tag the source code in the master branch.
- [ ] Add upgrade document

New features
------------------

Improvements
-------------------

Bug fixes
-------------

Changes
-----------

Deprecations
-----------------

Breaking Changes
------------------------

Security
-----------

All PRs scheduled for this release can be seen from [here](https://github.com/SmingHub/Sming/milestone/33)

slaff, 2023-03-16:
@mikee47 is your [USB](https://github.com/mikee47/Sming/commits/feature/usb) library ready for merging?

mikee47, 2023-03-16:
@slaff Nearly, needs documenting.

mikee47, 2023-04-11:
@slaff I've just noticed https://www.katacoda.com/ has been shuttered...

slaff, 2023-04-13:
> I've just noticed https://www.katacoda.com/ has been shuttered...

Yes, unfortunately. May be we can replace it with a gitpod or similar environment plus a text tutorial. I have started long ago some tests, see branch[ feature/devcontainers](https://github.com/SmingHub/SmingSkeletonApp/tree/feature/devcontainers) in SmingSkeletonApp, but haven't had the time to finish and polish.

closed 2023-06-15 by slaff