SmingHub/Sming #2616: Remove unnecessary null pointer checks

elfring, 2023-02-09:
[An extra null pointer check is not needed in functions](https://isocpp.org/wiki/faq/freestore-mgmt#delete-handles-null "Do I need to check for null before delete p?") like the following.
- [AccessPointImpl::config](https://github.com/SmingHub/Sming/blob/112da39f9d718a37e937de8825f6faf4a1c45cc1/Sming/Components/Network/Arch/Esp8266/Platform/AccessPointImpl.cpp#L85 "AccessPointImpl::config function")
- [doUpgrade](https://github.com/SmingHub/Sming/blob/2345c3e8aacd94af917441f93f55e9a60dd7cee3/samples/Basic_Ota/app/application.cpp#L51 "doUpgrade function")
- [YeelightBulb](https://github.com/SmingHub/Sming/blob/07ecd3bdf33f5b56ec1c32c9587df456ffaa4c07/Sming/Libraries/Yeelight/YeelightBulb.cpp#L14 "Destructor for YeelightBulb")

slaff, 2023-02-10:
@elfring can you prepare a PR with the suggested changes?

slaff, 2023-05-23:
> Some of those should be using std::unique_ptr as well :-)

@mikee47 do you want to create a PR with the changes?

elfring, 2023-05-23:
:thought_balloon: Would you become interested to [use a development tool like “clang-tidy” for corresponding source code adjustments](https://clang.llvm.org/extra/clang-tidy/checks/readability/delete-null-pointer.html "Deletion of unnecessary null pointer checks")?

slaff, 2023-05-26:
Fixed in #2643

mikee47, 2023-05-28:
I'm reopening this issue as it is not yet fully resolved. clang-tidy is undoubtedly useful, especially if it can be integrated into the build system in some way. Whether automated code changes are appropriate is another matter.... 

Are there any other offline code checking tools we might consider?

elfring, 2023-05-28:
[Another update candidate was left over](https://github.com/SmingHub/Sming/blob/02835d8f4e539931c46b16191a6b38f7c0a4b04e/Sming/Components/Network/src/Network/TelnetServer.cpp#L34 "Update candidate: TelnetServer::enableCommand()").

mikee47, 2024-07-04:
@elfring

closed 2024-07-04 by mikee47