SmingHub/Sming #2617: build issue for esp32 / esptool.CHIP_DEFS

pljakobs, 2023-02-12:
the line 
```
loader = esptool.CHIP_DEFS[os.environ['ESP_VARIANT']] 
```
from Sming/Arch/Esp32/Tools)/memanalyzer.py in the current devel branch (to which I have rebased my esp32_pwm branch) creates a build error in my environment (IDF 4.4, python 3.11.1)
```
Basic_HwPWM: Linking out/Esp32/esp32c3/debug/build/app.out
Traceback (most recent call last):
  File "/opt/sming/Sming/Arch/Esp32/Tools/memanalyzer.py", line 21, in <module>
    loader = esptool.CHIP_DEFS[os.environ['ESP_VARIANT']]
             ^^^^^^^^^^^^^^^^^
AttributeError: module 'esptool' has no attribute 'CHIP_DEFS'
```
reverting it back to the old 
```
loader = esptool._chip_to_rom_loader(os.environ['ESP_VARIANT'])
```
allows me to successfully build again. Not sure if that's an incompatibility between IDF 4.4 and 5.0? 

pj

mikee47, 2023-02-12:
The esptool submodule has been updated, so you'll need to reset using `make submodules-clean`.

mikee47, 2023-02-12:
... or the traditional `make dist-clean`

pljakobs, 2023-02-12:
oh, make submodules-clean did the trick, make dist-clean did not. 
Thank you.

closed 2023-02-12 by pljakobs