SmingHub/Sming #2618: faulty flash size detected by boot.esp32c3

pljakobs, 2023-02-13:
I hope this is not another "you're holding it wrong" issue but I also believe it's esptool, not sming, so I've opened an issue over there, this is just for information:
https://github.com/espressif/esptool/issues/837

mikee47, 2023-02-13:
#2613 offers a brief explanation of the behaviour. For Sming, the bootloader is built with the flash size setting configured in the hardware config file. If this changes, you should do a make `sdk-config-clean` then re-build/flash the bootloader.

pljakobs, 2023-02-13:
Interesting.
I had rebased my pwm stuff to the recent develop branch and then started to get this issue - with no other changes anywhere. 

My .hw file is configured for 4M - is that what you are referring to?
If I change this to 2M, things work fine, albeit the bootloader complains that the binary doesn't use all available flash.
I will try the sdk-clean.

pljakobs, 2023-02-13:
ok, I don't understand it, but sdk-config-clean seems to have done the trick. I guess there was something that was still lingering around.

closed 2023-02-13 by pljakobs