SmingHub/Sming #2641: ssl no verify

natxopedreira, 2023-05-20:
Hello

Sorry as this is more a question that an issue, i would like to know if this is possible to do with this framework.

Use ssl/mqtt in "insecure" mode?
Set ssl to not verify, basically i want to use mqtt through port 8883 with ssl without validation.

Hope you get what i mean.

I know that can be done in esp32-idf but i prefer a c++ approach if possible, and the idea to be able to switch between rp2040/esp32 platforms sounds cool.

Thanks

slaff, 2023-05-22:
@natxopedreira Take a look at the [MqttClient sample](https://github.com/SmingHub/Sming/tree/develop/samples/MqttClient_Hello).  In the [sample on line 100](https://github.com/SmingHub/Sming/blob/1a34cfa1b0b460e8cd26d557bdad40ffa8d15307/samples/MqttClient_Hello/app/application.cpp#L100) you will see how to disable the standard certificate validation:

https://github.com/SmingHub/Sming/blob/1a34cfa1b0b460e8cd26d557bdad40ffa8d15307/samples/MqttClient_Hello/app/application.cpp#L100

And if you don't want to have any other forms of validation you can delete line 101

https://github.com/SmingHub/Sming/blob/1a34cfa1b0b460e8cd26d557bdad40ffa8d15307/samples/MqttClient_Hello/app/application.cpp#L100

So in your case the SSL initialization should look like this:
```cpp
	mqtt.setSslInitHandler([](Ssl::Session& session) {
		session.options.verifyLater = true;
	});
```

natxopedreira, 2023-05-22:
Ook, got it, so:

this will set to not verify during handshake
`session.options.verifyLater = true; `

and this void to use any cert
`session.keyCert.assign(default_private_key, sizeof(default_private_key), default_certificate, `

Thank you @slaff i went another route as a fast solution, but will give this a go as this framework is very interesting

closed 2023-05-22 by natxopedreira