SmingHub/Sming #2646: Problem installing Sming on ARM(64). - toolchain download as Intel/AMD binaries for target ESP and RP2040.

laurentppol, 2023-05-27:
Hi there, trying to install Sming on OrangePi3, (ARM64): 
1). had to manually install pip and python-dev 
2). for ESP8266 script downloads AMD64 toolchain, will not work.
looks ESP32* toolchain build OK, 
3). for RP2040 same as for ESP8266 -> script downloads prebuilt x86_64 binaries. Will not work
script report success, trying to build "Basic-WiFi" "as-is" -> simply "make" in samples dir
Sming tries to build...
laurentppol
buid fails, as expected (for ESP8266), 

/opt/esp-quick-toolchain/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc: 1: ELF: not found
/opt/esp-quick-toolchain/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc: 1: Syntax error: ";" unexpected

4). have to install cmake manually
trying to buid for ESP32: /opt/Sming/samples/Basic_WiFi# make SMING_ARCH=Esp32

esp-idf look is building....
laurentppol
Sming builds up....

Basic-WiFi build success :)

mikee47, 2023-05-28:
I don't have an orangePi but I'll roll 64-bit debian onto a RaspberryPi and check this out. The installer needs some tweaking as it assumes x86_64.

mikee47, 2023-05-28:
... building... slowly... I'm spoilt with 5950x :-)

mikee47, 2023-08-10:
Supported added, closing.

closed 2023-08-10 by mikee47