SmingHub/Sming #2651: error: comparison of integer expressions of different signedness (in esp-idf 4.4)

pljakobs, 2023-06-27:
this seems to be a bug in esp-idf 4.4. I'm a bit confused why this would only show up now. 
I'm using SMING_SOC=esp32c3

```
In file included from /opt/sming/Sming/Libraries/SPI/src/Arch/Esp32/SPI.cpp:22:
/opt/esp-idf/components/hal/esp32c3/include/hal/spi_ll.h: In function 'void spi_ll_write_buffer(spi_dev_t*, const uint8_t*, size_t)':
/opt/esp-idf/components/hal/esp32c3/include/hal/spi_ll.h:344:23: error: comparison of integer expressions of different signedness: 'int' and 'size_t' {aka 'unsigned int'} [-Werror=sign-compare]
     for (int x = 0; x < bitlen; x += 32) {
                     ~~^~~~~~~~
/opt/esp-idf/components/hal/esp32c3/include/hal/spi_ll.h: In function 'void spi_ll_read_buffer(spi_dev_t*, uint8_t*, size_t)':
/opt/esp-idf/components/hal/esp32c3/include/hal/spi_ll.h:396:23: error: comparison of integer expressions of different signedness: 'int' and 'size_t' {aka 'unsigned int'} [-Werror=sign-compare]
     for (int x = 0; x < bitlen; x += 32) {
```

this seems to be a RISC-V related issue as the system compiles with no issues for the xtensa based ESP32 / ESP32S2 SOCs.

pljakobs, 2023-06-27:
that would fix it.

slaff, 2023-07-05:
@pljakobs Please, close this issue if the problem is solved.