SmingHub/Sming #2653: ESP32, IDF 5.0: error with nvs_flash_init

mr-sven, 2023-07-24:
- ESP-IDF v5.0-547-g20e8e86c80
- SOC: Esp32 4MB flash
- Sming 5.0.0

I have an issue with initialization of Wifi, I used a clean machine and try to deploy the Basic_Wifi sample. The ESP freezes at boot and the WDT is resetting the chip.

commands:

```bash
$ git clone https://github.com/SmingHub/Sming /opt/sming
$ INSTALL_IDF_VER=5.0 $SMING_HOME/../Tools/install.sh esp32
$ source /opt/sming/Sming/../Tools/export.sh
$ cd /opt/sming/samples/Basic_WiFi/
$ make SMING_ARCH=Esp32 flash
```

output of WDT and addresses

```bash
W (76) boot.esp32: PRO CPU has been reset by WDT.
W (82) boot.esp32: WDT reset info: PRO CPU PC=0x40082230
W (88) boot.esp32: WDT reset info: APP CPU PC=0x40080240

$ xtensa-esp32-elf-addr2line -fe out/Esp32/esp32/debug/build/app.out 0x40082230
_xt_highint5
/opt/esp-idf-5.0/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vector_defaults.S:192

$ xtensa-esp32-elf-addr2line -fe out/Esp32/esp32/debug/build/app.out 0x40080240
_Level5Vector
/opt/esp-idf-5.0/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1647
```

In the end I drilled it down to the function `nvs_flash_init`. 
I used the Basic_Blink sample and added the function and it freezes.

```c
#include <nvs_flash.h>

void init()
{
	pinMode(LED_PIN, OUTPUT);
	procTimer.initializeMs(1000, blink).start();
	esp_err_t ret = nvs_flash_init();
}
```

The crazy thing is, if I use the `nvs_rw_value` example of the installed IDF it is working.

slaff, 2023-07-24:
Just for my understanding: Is the Basic_Blink application working for you? 
Also why are you adding platform/arch/soc specific code in an application? It will make it hard to use in another architecture. 
And finally the nvs initialization for ESP32 is already done behind the scenes for you in Sming `Components/Network/Arch/Esp32/Platform/init.cpp`?

mr-sven, 2023-07-24:
Ahm, Basic_Blink is working, Basic_Wifi is not working.
As soon as I remove the `DISABLE_NETWORK` switch form the Basic_Blink, it is also not working. 
So I dig into.
In `Sming/Arch/Esp32/Components/esp32/src/startup.cpp` there is `esp_network_initialise` which is executed as soon as network is on.
In `Sming/Components/Network/Arch/Esp32/Platform/init.cpp` the first action is to initialize the NVS, and exactly there it freezes.

So for simple test you can execute the nvs_flash_init in the Basic_Blink and it freezes also.

slaff, 2023-07-24:
@mr-sven Thank you. 

> So for simple test you can execute the nvs_flash_init in the Basic_Blink and it freezes also.

@mikee47 any idea what can cause this issue with IDF 5.0 ?

mikee47, 2023-07-24:
I've traced it through to the `spi_flash_disable_interrupts_caches_and_other_cpu()` function (called during loading of the partition table) which hangs on the first `spi_flash_disable_cache()` call. I have a suspicion there's an interrupt service routine running out of flash, which shouldn't be.

mikee47, 2023-07-24:
I've attempted to load the partition table as early as possible by calling `esp_partition_find` directly from `app_main`, but just hangs there instead.

A temporary fix is to run `make sdk-menuconfig` and set `FREERTOS_UNICORE=y`.

mr-sven, 2023-07-24:
Yes, that works for the moment.
That was also my idea that anything stuck at task level or an ISR, because the WDT PC addresses are always at some interrupts.

mr-sven, 2023-07-25:
I've diffed the sdkconfig from IDF and Sming.
The only settings what points me are `CONFIG_ESP_SYSTEM_CHECK_INT_LEVEL_4` and `CONFIG_ESP_SYSTEM_CHECK_INT_LEVEL_5`, some WDT and FREERTOS Flags.
```diff
--- idf.sdkconfig       2023-07-25 08:51:34.486648025 +0200
+++ sming.sdkconfig     2023-07-25 08:53:53.128721174 +0200
@@ -16,4 +16,4 @@
-CONFIG_APPTRACE_DEST_NONE=y
-CONFIG_APPTRACE_DEST_UART_NONE=y
-CONFIG_APPTRACE_LOCK_ENABLE=y
-CONFIG_APPTRACE_UART_TASK_PRIO=1
+CONFIG_BLE_ADV_REPORT_DISCARD_THRSHOLD=20
+CONFIG_BLE_ADV_REPORT_FLOW_CONTROL_NUM=100
+CONFIG_BLE_ADV_REPORT_FLOW_CONTROL_SUPPORTED=y
+CONFIG_BLE_SCAN_DUPLICATE=y
@@ -32,0 +33,87 @@
+CONFIG_BT_CONTROLLER_ENABLED=y
+CONFIG_BTDM_BLE_ADV_REPORT_DISCARD_THRSHOLD=20
+CONFIG_BTDM_BLE_ADV_REPORT_FLOW_CTRL_NUM=100
+CONFIG_BTDM_BLE_ADV_REPORT_FLOW_CTRL_SUPP=y
+CONFIG_BTDM_BLE_DEFAULT_SCA_250PPM=y
+CONFIG_BTDM_BLE_SCAN_DUPL=y
+CONFIG_BTDM_BLE_SLEEP_CLOCK_ACCURACY_INDEX_EFF=1
+CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN=3
+CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN_EFF=3
+CONFIG_BTDM_CONTROLLER_BR_EDR_MAX_ACL_CONN_EFF=0
+CONFIG_BTDM_CONTROLLER_BR_EDR_MAX_SYNC_CONN_EFF=0
+CONFIG_BTDM_CONTROLLER_FULL_SCAN_SUPPORTED=y
+CONFIG_BTDM_CONTROLLER_HCI_MODE_VHCI=y
+CONFIG_BTDM_CONTROLLER_MODE_BLE_ONLY=y
+CONFIG_BTDM_CONTROLLER_MODEM_SLEEP=y
+CONFIG_BTDM_CONTROLLER_PINNED_TO_CORE=0
+CONFIG_BTDM_CTRL_BLE_MAX_CONN=3
+CONFIG_BTDM_CTRL_BLE_MAX_CONN_EFF=3
+CONFIG_BTDM_CTRL_BR_EDR_MAX_ACL_CONN_EFF=0
+CONFIG_BTDM_CTRL_BR_EDR_MAX_SYNC_CONN_EFF=0
+CONFIG_BTDM_CTRL_BR_EDR_SCO_DATA_PATH_EFF=0
+CONFIG_BTDM_CTRL_FULL_SCAN_SUPPORTED=y
+CONFIG_BTDM_CTRL_HCI_MODE_VHCI=y
+CONFIG_BTDM_CTRL_HLI=y
+CONFIG_BTDM_CTRL_LPCLK_SEL_MAIN_XTAL=y
+CONFIG_BTDM_CTRL_MODE_BLE_ONLY=y
+CONFIG_BTDM_CTRL_MODEM_SLEEP_MODE_ORIG=y
+CONFIG_BTDM_CTRL_MODEM_SLEEP=y
+CONFIG_BTDM_CTRL_PCM_POLAR_EFF=0
+CONFIG_BTDM_CTRL_PCM_ROLE_EFF=0
+CONFIG_BTDM_CTRL_PINNED_TO_CORE=0
+CONFIG_BTDM_CTRL_PINNED_TO_CORE_0=y
+CONFIG_BTDM_RESERVE_DRAM=0xdb5c
+CONFIG_BTDM_SCAN_DUPL_CACHE_REFRESH_PERIOD=0
+CONFIG_BTDM_SCAN_DUPL_CACHE_SIZE=100
+CONFIG_BTDM_SCAN_DUPL_TYPE=0
+CONFIG_BTDM_SCAN_DUPL_TYPE_DEVICE=y
+CONFIG_BT_ENABLED=y
+CONFIG_BT_NIMBLE_50_FEATURE_SUPPORT=y
+CONFIG_BT_NIMBLE_ACL_BUF_COUNT=24
+CONFIG_BT_NIMBLE_ACL_BUF_SIZE=255
+CONFIG_BT_NIMBLE_ATT_PREFERRED_MTU=256
+CONFIG_BT_NIMBLE_COEX_PHY_CODED_TX_RX_TLIM_EFF=0
+CONFIG_BT_NIMBLE_ENABLED=y
+CONFIG_BT_NIMBLE_GAP_DEVICE_NAME_MAX_LEN=31
+CONFIG_BT_NIMBLE_GATT_MAX_PROCS=4
+CONFIG_BT_NIMBLE_HCI_EVT_BUF_SIZE=70
+CONFIG_BT_NIMBLE_HCI_EVT_HI_BUF_COUNT=30
+CONFIG_BT_NIMBLE_HCI_EVT_LO_BUF_COUNT=8
+CONFIG_BT_NIMBLE_HOST_TASK_STACK_SIZE=4096
+CONFIG_BT_NIMBLE_HS_FLOW_CTRL_ITVL=1000
+CONFIG_BT_NIMBLE_HS_FLOW_CTRL_THRESH=2
+CONFIG_BT_NIMBLE_HS_FLOW_CTRL_TX_ON_DISCONNECT=y
+CONFIG_BT_NIMBLE_HS_FLOW_CTRL=y
+CONFIG_BT_NIMBLE_HS_STOP_TIMEOUT_MS=2000
[... 156 lines cut ...]
-CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU1=y
-CONFIG_TASK_WDT_TIMEOUT_S=5
+CONFIG_SYSTEM_EVENT_TASK_STACK_SIZE=16384
@@ -611 +694 @@
-CONFIG_TCPIP_TASK_STACK_SIZE=3072
+CONFIG_TCPIP_TASK_STACK_SIZE=8192
@@ -631,9 +713,0 @@
-CONFIG_UNITY_ENABLE_DOUBLE=y
-CONFIG_UNITY_ENABLE_FLOAT=y
-CONFIG_UNITY_ENABLE_IDF_TEST_RUNNER=y
-CONFIG_VFS_SEMIHOSTFS_MAX_MOUNT_POINTS=1
-CONFIG_VFS_SUPPORT_DIR=y
-CONFIG_VFS_SUPPORT_IO=y
-CONFIG_VFS_SUPPORT_SELECT=y
-CONFIG_VFS_SUPPORT_TERMIOS=y
-CONFIG_VFS_SUPPRESS_SELECT_DEBUG_OUTPUT=y
@@ -640,0 +715 @@
+CONFIG_WIFI_PROV_BLE_SEC_CONN=y
@@ -643,2 +717,0 @@
-CONFIG_WL_SECTOR_SIZE=4096
-CONFIG_WL_SECTOR_SIZE_4096=y
@@ -647,2 +719,0 @@
-CONFIG_WS_BUFFER_SIZE=1024
-CONFIG_WS_TRANSPORT=y
```

mr-sven, 2023-07-25:
Ok, that was also not the problem. Its only caused because the message is enclosed in `ESP_LOGI` and not `ESP_EARLY_LOGI`.

mr-sven, 2023-07-31:
I've tried hard on setting sdkconfig switches, but nothing helped. It looks like there are some compiler switches missing or any other thing. I'm totally lost at the moment.

mikee47, 2023-07-31:
Do you need both CPUs? If not, stick to unicore for now.

mikee47, 2023-07-31:
Getting rid of FreeRTOS entirely would be ideal, but the WiFi stack is closed-source and has some embedded dependencies on RTOS operation.

profjmer, 2023-08-12:
Other samples reset the ESP32 too. Those using WiFi.
I had a functional web site running on ESP32, but the new version brakes the code and ESP32 is always resetting.
How can we identify Sming version from the uart log ?

mr-sven, 2023-08-12:
Afaik. the problem is located in the SPI flash access. I currently had no chance to test against IDF 4.4, I think it is caused by IDF 5.

mikee47, 2024-01-06:
> Afaik. the problem is located in the SPI flash access. I currently had no chance to test against IDF 4.4, I think it is caused by IDF 5.

I've updated sming IDF 4.4 to latest Espressif release (date 3/1/24) and can confirm that the above issue is still present, however the same fix applies and Basic_Ssl sample is working.

To update existing IDF installation manually:

```
cd /opt/esp-idf-4.4
git pull
git submodule update --init --recursive
python3 tools/idf_tools.py --non-interactive install-python-env
```

mikee47, 2024-11-23:
Been taking another look at this issue, which persists. To summarise, system hangs on call to `spi_flash_disable_interrupts_caches_and_other_cpu` when `nvs_flash_init` is called.

There are reports as far back as 2017 https://esp32.com/viewtopic.php?t=1264#p5598 which indicate that earlier versions of the IDF called `nvs_flash_init` as part of `esp_phy_init` before the freeRTOS scheduler is started.

This patch to the IDF in `components/esp_system/startup.c` fixes the issue by initialising the partition table before the scheduler is started:

```
@@ -471,10 +471,12 @@ static void start_cpu0_default(void)
 
     // Execute init functions of other components; blocks
     // until all cores finish (when !CONFIG_ESP_SYSTEM_SINGLE_CORE_MODE).
     do_secondary_init();
 
+esp_partition_find(ESP_PARTITION_TYPE_ANY, ESP_PARTITION_SUBTYPE_ANY, NULL);
+
     // Now that the application is about to start, disable boot watchdog
 #ifndef CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE
     wdt_hal_context_t rtc_wdt_ctx = RWDT_HAL_CONTEXT_DEFAULT();
     wdt_hal_write_protect_disable(&rtc_wdt_ctx);
     wdt_hal_disable(&rtc_wdt_ctx);
```

The `Basic_WiFi` sample now starts successfully in dual-core more (setting `CONFIG_FREERTOS_UNICORE=n`).

I don't feel this explains things fully but it's a step towards a solution I think.

mikee47, 2024-11-25:
So we can insert the call without patching the IDF using `__attribute__((constructor))` on a function. These are called by `do_global_ctors()`.

mikee47, 2024-11-25:
So finally I've tracked down the cause of all this.

`Sming/Arch/Esp32/Components/esp32/sdk/esp_system.mk` refers to `ld_include_panic_highint_hdl`, which at some point was renamed to `ld_include_highint_hdl`.

This means the code in `/opt/esp-idf-5.2/components/esp_system/port/soc/esp32/highint_hdl.S` is not linked and is critical for supporting IPC calls on the esp32.

 I was incorrect above in stating that `spi_flash_disable_interrupts_caches_and_other_cpu` is the problem. I can happily call that and then `spi_flash_enable_interrupts_caches_and_other_cpu` directly afterwards without issue. So clearly the problem is somewhere in-between.

When `nvs_flash_init` is called it loads the partition table from flash:

```
esp_partition_find                 [esp_partition/partition.c]
-> ensure_partitions_loaded()
-> load_partitions()
-> spi_flash_mmap()                [spi_flash/flash_mmap.c]
-> esp_mmu_map()                   [esp_mm/esp_mmu_map.c]
-> s_do_mapping()
-> s_do_cache_invalidate()
-> cache_sync()                    [esp_mm/cache_esp32.c]
-> Cache_Flush()                   [esp_rom/include/esp32/rom/cache.h]
-> DPORT_STALL_OTHER_CPU_START()   [esp_hw_support/include/dport_access.h]
-> esp_ipc_isr_stall_other_cpu()
```
By a process of elimination (commenting-out bits of code) I determined that it stalls in this IPC call.

mikee47, 2024-11-25:
This was exactly the cause I expected to be, but I'd ignored `ld_include_panic_highint_hdl` because, well, what on earth has `panic` got to do with it?!

PR #2916 (2024-11-26): Fix esp32 multicore lockup
This PR fixes the broken IPC calling interrupts on esp32 chips. Closes #2653.

Cause is due to an incorrect linker flag which resulted in critical high-level interrupt handling code from the ESP IDF not getting linked.

Testing also revealed a race condition during access point initialisation. The `Basic_Wifi` sample first initialises the access point, then the station. Shortly after startup, some internal structure for the access point gets invalidated and causes an access violation. It appears to be related to calling `esp_wifi_start` twice.
commit 7b3ef4f3 (2024-11-26): Fix esp32 multicore lockup (#2916)

This PR fixes the broken IPC calling interrupts on esp32 chips. Closes #2653.

Cause is due to an incorrect linker flag which resulted in critical high-level interrupt handling code from the ESP IDF not getting linked.

Testing also revealed a race condition during access point initialisation. The `Basic_Wifi` sample first initialises the access point, then the station. Shortly after startup, some internal structure for the access point gets invalidated and causes an access violation. It appears to be related to calling `esp_wifi_start` twice.
closed 2024-11-26 by slaff