SmingHub/Sming #2659: dangling pointer warning in print.h?

pljakobs, 2023-08-12:
this might be a self-created issue, but I don't see how to fix it.
I have successfully built my firmware on sming 5.0 this morning, but that fails now and I cannot say why that is.
The compiler warns me of a dangling pointer here:
```
Building /home/pjakobs/devel/esp_rgbww_firmware/Sming/Sming/out/Host/debug/lib/clib-Storage-e2fa65b3ebab4603f1bf91105423d574.a
C+ /home/pjakobs/devel/esp_rgbww_firmware/Sming/Sming/Components/Storage/src/Debug.cpp
C+ /home/pjakobs/devel/esp_rgbww_firmware/Sming/Sming/Components/Storage/src/Device.cpp
C+ /home/pjakobs/devel/esp_rgbww_firmware/Sming/Sming/Components/Storage/src/Partition.cpp
In file included from /home/pjakobs/devel/esp_rgbww_firmware/Sming/Sming/Components/Storage/src/include/Storage/Debug.h:3,
                 from /home/pjakobs/devel/esp_rgbww_firmware/Sming/Sming/Components/Storage/src/Debug.cpp:1:
In member function 'size_t Print::print(const char*)',
    inlined from 'typename std::enable_if<(! std::is_array< <template-parameter-1-1> >::value), Print&>::type operator<<(Print&, const T&) [with T = char*]' at /home/pjakobs/devel/esp_rgbww_firmware/Sming/Sming/Wiring/Print.h:272:9,
    inlined from 'void Storage::Debug::listPartitions(Print&, const Storage::Device&)' at /home/pjakobs/devel/esp_rgbww_firmware/Sming/Sming/Components/Storage/src/Debug.cpp:21:29:
/home/pjakobs/devel/esp_rgbww_firmware/Sming/Sming/Wiring/Print.h:108:29: error: using a dangling pointer to 'buf' [-Werror=dangling-pointer=]
  108 |                 return write(str);
      |                        ~~~~~^~~~~
In file included from /home/pjakobs/devel/esp_rgbww_firmware/Sming/Sming/Components/FlashString/src/include/FlashString/config.hpp:24,
                 from /home/pjakobs/devel/esp_rgbww_firmware/Sming/Sming/Components/FlashString/src/include/FlashString/Utility.hpp:24,
                 from /home/pjakobs/devel/esp_rgbww_firmware/Sming/Sming/Components/FlashString/src/include/FlashString/Object.hpp:24,
                 from /home/pjakobs/devel/esp_rgbww_firmware/Sming/Sming/Components/FlashString/src/include/FlashString/String.hpp:24,
                 from /home/pjakobs/devel/esp_rgbww_firmware/Sming/Sming/Wiring/WString.h:64,
                 from /home/pjakobs/devel/esp_rgbww_firmware/Sming/Sming/Wiring/Print.h:24:
/home/pjakobs/devel/esp_rgbww_firmware/Sming/Sming/Components/Storage/src/Debug.cpp: In function 'void Storage::Debug::listPartitions(Print&, const Storage::Device&)':
/home/pjakobs/devel/esp_rgbww_firmware/Sming/Sming/Wiring/FakePgmSpace.h:86:27: note: 'buf' declared here
   86 |                 LOAD_PSTR(buf, __pstr__);                                                                                      \
      |                           ^~~
```

I've done a "make dist-clean" and a "make components-clean" but the error persists. Also, while the last change I made was about mounting the spiffs partition, I can't see how anything I did would bring this change about.

In fact, the last thing I ran before this came up was a " make ENABLE_LWIPDEBUG=0" because I was still getting lots of lwip debug output which is no longer needed. I wasn't expecting this :-o

pljakobs, 2023-08-12:
I believe this might be the root of this issue, but - why would the compiler warn about this? 
```
void Application::mountfs(int slot) {
    debug_i("Application::mountfs rom slot: %i", slot);
    // auto part = OtaUpgrader::getPartitionForSlot(slot);
    auto part = Storage::findPartition(F("spiffs") + slot);
    debug_i("Application::mountfs trying to mount spiffs at %x, length %d",
            part.address(), part.size());
    _fs_mounted = spiffs_mount(part);
}
```

pljakobs, 2023-08-12:
btw: I wish there was some sort of a cummunity forum for this kind of questions.

mikee47, 2023-08-13:
> btw: I wish there was some sort of a community forum for this kind of questions.

[gitter.im](https://app.gitter.im/#/room/#SmingHub_Sming:gitter.im)

mikee47, 2023-08-13:
What version of GCC is being used? (`make list-config | grep GCC`)
What architecture? (Host ?)

A temporary workaround is reporting this dangling pointer as a warning using `make STRICT=1`. This started happening with later GCC versions (not sure exactly when) but I'd consider it a false positive.

If you have a dig into the `_F(str)` macro it boils down to this:

```
#define _F(str)	{ \
	static const char flash_str[] PROGMEM_PSTR = str; \
	char name[ALIGNUP4(sizeof(flash_str))] __attribute__((aligned(4))); \
	memcpy_aligned(buf, flash_str, sizeof(flash_str)); \
	buf; \
	}
```

I would want a dangling pointer warning for this:

```
const char* ptr = _F("Some text);
print("%s", ptr);
```

but not for this:

```
print("%s", _F("Some text));
```

pljakobs, 2023-08-15:
thanks @mikee47 
```
- GCC_VERSION = 12
```
on Fedora 37 x64 

I wonder - is the gcc used for xtensa the same that I got through dnf install? in other words: is the cross compiling part of the toolchain installed with the esp-idf ?

```
gcc-12.3.1-1.fc37.x86_64
```

with the STRICT=1 setting, it does compile, but there are tons of int/uint compare warnings again. 

```
[pjakobs@devbox Sming]$ cd /opt/sming/Sming/
[pjakobs@devbox Sming]$ git describe --tags
5.0.0
```
on the validity of the dangling pointer warning: it's probably a bit aggressive but in general, seeing that the first example you provide is something you could easily have and the _F macro being something that application developers would use, in fact are encouraged to use, it's not bad to have it.
It would, if I'm not mistaken, be the way to store verbatim html documents in flash strings, right?

pj

mikee47, 2023-08-15:
Yes, the `STRICT` does enable additional checks, but also converts some errors into warnings so that builds can complete. This is used for CI builds so that the logs are more useful.

Each architecture has its own toolchain. In linux installs:

- Host uses the native GCC
- Esp8266 it's `/opt/esp-quick-toolchain`
- Raspberry Pi Pico it's `/opt/rp2040`
- Esp32 it's `/opt/esp32/tools`. There may be multiple toolchain versions here (and for Xtensa, RISC, etc) with the exact version dictated by the IDF SDK version and chip variant.

The dangling pointer issue isn't raised by GCC 10.2 (the default esp8266 version). Ironically, that's probably the only architecdture which _really_ needs all this messing about with flash pointers because unaligned accesses crash the system. All the others have workarounds.

I'm of the view that this is all down to a 'false positive' which may get resolved by GCC. Fingers crossed!

mikee47, 2023-08-15:
> It would, if I'm not mistaken, be the way to store verbatim html documents in flash strings, right?

Yes. The (FlashString)[https://sming.readthedocs.io/en/latest/_inc/Sming/Components/FlashString/index.html] library was written to handle this sort of thing more elegantly.

Check out the (Basic_WebSkeleton)[https://sming.readthedocs.io/en/latest/_inc/samples/Basic_WebSkeletonApp/index.html] sample application which demonstrates using flash directly without a filesystem.

mikee47, 2023-08-15:
Also check out https://sming.readthedocs.io/en/latest/framework/core/pgmspace.html

slaff, 2023-10-24:
Should be fixed in latest `develop`. Re-open this issue if the problem is still not fixed.

closed 2023-10-24 by slaff