SmingHub/Sming #2660: Link to interactive documentation (on katacoda) is defunct

iokhahon, 2023-08-27:
The link on the Sming github page to https://www.katacoda.com/slaff/scenarios/sming-host-emulator  takes you to the following message.

> Thank you to the Katacoda community for your years of support. Katacoda.com is now closed. Learn more about this decision and
> access interactive learning on O’Reilly [here](https://www.oreilly.com/online-learning/leveraging-katacoda-technology.html).

There are very few resources on the internet for new Sming users, a broken link like this suggests the project is a dead and nobody cares about attracting new users/contributors.

mikee47, 2023-08-28:
Looks like Killercoda supports Katacoda scenerios directly. Here's a test https://killercoda.com/smingdev/scenario/host
@slaff This requires configuration for your katacoda-scenarios repo. Do you want to update?

slaff, 2023-09-08:
> Do you want to update?

@mikee47 yep, I am doing it. The new Sming host emulator link should be: https://killercoda.com/slaff/scenario/sming-host-emulator. And the ESP-32 should be under https://killercoda.com/slaff/scenario/sming-on-esp32. Is this enough or I should do something more?

mikee47, 2023-09-08:
@slaff Thank you - looks good! I've had a quick run through the esp32 scenario and works as-is, though it does need updating as IDF versions have moved on.
I don't think katacoda required users to log in first, but killercoda definitely does.

PR #2663 (2023-09-08): Switch katacoda links to killercoda
Fixes #2660
closed 2023-09-08 by slaff