SmingHub/Sming #2662: Installation issue on ubuntu 22.04

dpechman, 2023-08-28:
in fresh installed ubuntu 22.04 after run: **/opt/sming/Tools/install.sh all** 
it crash in:

      Cython.Compiler.Errors.CompileError: src/gevent/libev/corecext.pyx
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.

mikee47, 2023-08-28:
I  can confirm this error, happens with the esp32 IDF installation. It looks like 'gevent' isn't compatible with recent python releases. https://github.com/gevent/gevent/issues/1982 and doesn't look like there's an approved fix as yet.

If you're not targetting esp32 devices then try `install.sh esp8266` (or `rp2040`, etc)

mikee47, 2023-08-28:
ESP-IDF version 5 doesn't seem to be affected by this problem. Try:

```
INSTALL_IDF_VER=5.0 $SMING_HOME/../Tools/install.sh all
```

See https://sming.readthedocs.io/en/latest/_inc/Sming/Arch/Esp32/README.html#idf-versions.

dpechman, 2023-08-30:
what is the recommended python version for use with idf 5.0? I am targetting esp32.

using 3.10.12 it was able to compile Basic_Blink using **make SMING_ARCH=Esp8266** no errros

But using **make SMING_ARCH=Esp32**

`Basic_Blink: Invoking 'all' for Esp32/esp32 (debug) architecture
ninja: Entering directory `/opt/sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk'
[1/8] Performing build step for 'bootloader'
[1/1] cd /opt/sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/bootloader/esp-idf/esptool_py && /usr/bin/python /opt/esp-idf-5.0/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /opt/sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/bootloader/bootloader.bin
Bootloader binary size 0x6710 bytes. 0x8f0 bytes (8%) free.
[4/8] Generating ld/sections.ld
FAILED: esp-idf/esp_system/ld/sections.ld /opt/sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/esp-idf/esp_system/ld/sections.ld 
cd /opt/sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/esp-idf/esp_system && /usr/bin/python /opt/esp-idf-5.0/tools/ldgen/ldgen.py --config /opt/sming/Sming/Arch/Esp32/Components/esp32/project/esp32/debug/sdkconfig --fragments-list "/opt/esp-idf-5.0/components/xtensa/linker.lf;/opt/esp-idf-5.0/components/esp_ringbuf/linker.lf;/opt/esp-idf-5.0/components/driver/linker.lf;/opt/esp-idf-5.0/components/esp_pm/linker.lf;/opt/esp-idf-5.0/components/spi_flash/linker.lf;/opt/esp-idf-5.0/components/esp_system/linker.lf;/opt/esp-idf-5.0/components/esp_system/app.lf;/opt/esp-idf-5.0/components/esp_common/common.lf;/opt/esp-idf-5.0/components/esp_common/soc.lf;/opt/esp-idf-5.0/components/esp_rom/linker.lf;/opt/esp-idf-5.0/components/hal/linker.lf;/opt/esp-idf-5.0/components/log/linker.lf;/opt/esp-idf-5.0/components/heap/linker.lf;/opt/esp-idf-5.0/components/soc/linker.lf;/opt/esp-idf-5.0/components/esp_hw_support/linker.lf;/opt/esp-idf-5.0/components/freertos/linker.lf;/opt/esp-idf-5.0/components/newlib/newlib.lf;/opt/esp-idf-5.0/components/newlib/system_libs.lf;/opt/esp-idf-5.0/components/esp_adc/linker.lf;/opt/esp-idf-5.0/components/esp_event/linker.lf;/opt/esp-idf-5.0/components/esp_gdbstub/linker.lf;/opt/esp-idf-5.0/components/lwip/linker.lf;/opt/esp-idf-5.0/components/esp_netif/linker.lf;/opt/esp-idf-5.0/components/esp_wifi/linker.lf;/opt/esp-idf-5.0/components/esp_phy/linker.lf;/opt/esp-idf-5.0/components/espcoredump/linker.lf" --input /opt/esp-idf-5.0/components/esp_system/ld/esp32/sections.ld.in --output /opt/sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/esp-idf/esp_system/ld/sections.ld --kconfig /opt/esp-idf-5.0/Kconfig --env-file /opt/sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/config.env --libraries-file /opt/sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/ldgen_libraries --objdump /opt/esp32/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-objdump
Traceback (most recent call last):
  File "/opt/esp-idf-5.0/tools/ldgen/ldgen.py", line 16, in <module>
    from ldgen.entity import EntityDB
  File "/opt/esp-idf-5.0/tools/ldgen/ldgen/entity.py", line 12, in <module>
    from pyparsing import (Group, Literal, OneOrMore, ParseException, SkipTo, Suppress, White, Word, ZeroOrMore, alphas,
ImportError: cannot import name 'rest_of_line' from 'pyparsing' (/usr/lib/python3/dist-packages/pyparsing.py)
ninja: build stopped: subcommand failed.
make: *** [/opt/sming/Sming/Arch/Esp32/Components/esp32/component.mk:372: checksdk] Erro 1
`

mr-sven, 2023-11-07:
At the moment it is impossible to build Sming for Esp32 & Wifi on current ubuntu releases. IDF 5.0 is broken due problem with SPI Flash access #2653 and IDF 4.4 due the gevent problem. 
The IDF fork is still at 4.4.3, at espressif they might fixed the gevent problem with 4.4.4+ https://github.com/espressif/esp-idf/issues/12249

mikee47, 2024-06-22:
CI building with ubuntu 22.04 and updated install scripts. Please reopen issue if problems persist.

closed 2024-06-22 by mikee47