SmingHub/Sming #2665: CommandProcessing with WebSocket

sreggy, 2023-09-18:
Hi all,
I've tryed CommandProcessing sample for manage command handler using websocket.

It works fine with serial and telnet but I'm not able to use it with websocket. It's like the command handler is not attached to websocket interface.

Can anyone help me?

Thanks in advance
Regards
Stefano

slaff, 2023-12-06:
@sreggy I am working on a PR #2687 that will make it possible to use CommandProcessor for any text line interface.

sreggy, 2023-12-12:
Thanks a lot @slaff.
As soon as something is usable, available to test it.

Just another question, is this text line interface useful also using MQTT?

Thanks again,
Regards
Stefano

slaff, 2023-12-12:
> Just another question, is this text line interface useful also using MQTT?

Yes. You can take the MqttClient_Hello sample as  a starting point and modify the onMessageReceived callback from app/application.cpp to process the messages via CommandHandler. Something like:

```C++
#include <CommandProcessing/Utils.h>

CommandProcessing::Handler commandHandler;

// Callback for messages, arrived from MQTT server
int onMessageReceived(MqttClient& client, mqtt_message_t* message)
{
	String output = commandHandler.processNow(reinterpret_cast<const char*>(message->publish.content.data), message->publish.content.length);
	if(output.length() > 0) {
		client.publish(MqttBuffer(message->publish.topic_name), output);
	}

	return 0;
}
```

Make sure also to add the dependency on CommandProcessor. This can be done by adding the following line to the application `component.mk` file.
```
COMPONENT_DEPENDS += CommandProcessing
```

slaff, 2023-12-12:
> As soon as something is usable

It is already merged in the `develop` branch and you can test it right away.

sreggy, 2023-12-12:
Hi @slaff I did some test using the provided sample for Telnet but I'm not able to have success.
I'm able to connect to the telnet session, but if I send a command, no output on Telenet session.
Out seems to be redirect on Serial with the following error message` 154552374 Cannot use this method when output stream is set`.

here Telnet session
```
❯ telnet 192.168.0.173
Trying 192.168.0.173...
Connected to testone.saruman.
Escape character is '^]'.
help
help
```

here Serial output
```
249433837 Cannot use this method when output stream is set
% !"'252789766 Cannot use this method when output stream is set
helpCommand not found, cmd = '%'
Sming>
258061171 Cannot use this method when output stream is set
helpCommands available are :
status | system | Displays System Information
echo | system | Displays command entered
help | system | Displays all available commands
debugon | system | Set Serial debug on
debugoff | system | Set Serial debug off
command | system | Use verbose/silent/prompt as command options
example | Application | Example Command
Sming>
```
Any suggestion?
Thanks in advance.
Regards 
Stefano

slaff, 2023-12-13:
> Any suggestion?

Yes, commend line 58 in app/application.cpp TelnetServer sample:
```
	commandHandler.setVerbose(true);
//	CommandProcessing::enable(commandHandler, Serial);

	WifiStation.enable(true);
```

And recompile the example.

slaff, 2023-12-13:
And something more. When you connect use the following command:

```
❯ telnet 192.168.0.173 23
```

Notice the port number after the IP. Without it your telnet client will switch to telnet protocol and will send an initial telnet packet which this simple sample does not support.

sreggy, 2023-12-13:
> > Any suggestion?
> 
> Yes, commend line 58 in app/application.cpp TelnetServer sample:
> 
> ```
> 	commandHandler.setVerbose(true);
> //	CommandProcessing::enable(commandHandler, Serial);
> 
> 	WifiStation.enable(true);
> ```
> 
> And recompile the example.

Thanks a lot @slaff .
Doing this, is CommandProcessing also working on Serial or I need register a different commandHandler?

thanks again
regards

Steafano

slaff, 2023-12-13:
> is CommandProcessing also working on Serial

Only on the TelnetServer. For Serial you need to create a new commandHandler.

sreggy, 2023-12-14:
Hi @slaff,
well using two commandHaldler I was able to use CommandProcessing from both Serial and from text line interface like Telnet and MQTT.

Some trouble on Websocket because receive a stacktrace on httpServer access for both  `http` and `ws` with a simple http://192.168.0.174 i receive (using the Sming/samples/HttpServer_WebSockets sample):

```
python3 -m serial.tools.miniterm --raw --encoding ascii --rts 0 --dtr 0 /dev/cu.usbserial-0001 916000
--- forcing DTR inactive
--- forcing RTS inactive
--- Miniterm on /dev/cu.usbserial-0001  916000,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
��rf[113] : 00
rf[114] : 01

SDK ver: 3.0.3(8427744) compiled @ Mar 19 2020 18:11:05
phy ver: 1156_0, pp ver: 10.2

134888 Station configuration is: ILUVATAR
mode : sta(c4:5b:be:49:55:3a)
add if0
179681 mode: 0 -> 3

scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt

connected with ILUVATAR, channel 1
dhcp client start...
298310 connect to ssid ILUVATAR, channel 1

ip:192.168.0.174,mask:255.255.255.0,gw:192.168.0.1
1161831 ip:192.168.0.174,mask:255.255.255.0,gw:192.168.0.1

1162188 TCP 3ffef270 +connection
1162450 '/' registered
1162689 '*' registered

=== WEB SERVER STARTED ===
192.168.0.174
==========================

5433120 onAccept, tcp: 3fff18d8, state: 0 K=0, Free heap size=42152
5433428 TCP 3fff1c68 +connection
5433755 TCP 3fff1c68 timeout updating: 70 -> 2
5433947 Opening connection. Total connections: 1
5434135 TcpServer onClient: 192.168.0.190

5435155 onAccept, tcp: 3fff1b20, state: 0 K=1, Free heap size=41448
5435404 TCP 3fff1f40 +connection
5435814 TCP 3fff1f40 timeout updating: 70 -> 2
5436292 Opening connection. Total connections: 2
5436794 TcpServer onClient: 192.168.0.190

5443808 The headers are complete
5450267 attached file: 'index.html' (1639 bytes) #0x3FFF1440
5451424 MemoryDataStream::realloc 0 -> 17
5451587 Sending stream. Bytes to send: 17
5451999 MemoryDataStream::realloc 17 -> 170
5452123 Sending stream. Bytes to send: 42
5452285 Sending stream. Bytes to send: 70
5452558 Sending stream. Bytes to send: 94
5452999 Sending stream. Bytes to send: 120
5453467 Sending stream. Bytes to send: 122
5453960 TCP 3fff1c68 onReadyToSendData: 1
5454629 TCP 3fff1c68 Written: 122, Available: 2780, isFinished: 0, PushCount: 1
5455285 TcpClient stream finished
[... 198 lines cut ...]
connected with ILUVATAR, channel 1
dhcp client start...
306592 connect to ssid ILUVATAR, channel 1

ip:192.168.0.174,mask:255.255.255.0,gw:192.168.0.1
1160559 ip:192.168.0.174,mask:255.255.255.0,gw:192.168.0.1

1160917 TCP 3ffef270 +connection
1161183 '/' registered
1161427 '*' registered

=== WEB SERVER STARTED ===
192.168.0.174
==========================

pm open,type:2 0
ip:192.168.0.174,mask:255.255.255.0,gw:192.168.0.1

```

any suggestion?

Thanks in advance
regards
Stefano

sreggy, 2023-12-15:
Hi @slaff , inline the info

> * First what is the (micro-controller) architecture  that you are using? Is it Esp32 or Esp8266 ?

Esp8266

> * Second: Run the following command from the root folder of the project:
> 
> ```
> make decode-stacktrace
> ```
> 
> And paste the text enclosed in '===='. Exactly as instructed above.
> 
I did it, but unfortunatly I'm not able to well analize output, no problem to paste it in this thread, let me to know ...

> * Third: In order to be able to test it properly the message that you send via your Websocket client must contain new line at the end. If your websocket client is not adding a new line at the end of a message then you have to add it (by pressing Enter and then clicking on submit or whatever button you have)

Yes, i've added it. On Mqtt it's working fine ...

> 
> Meanwhile I will run the stock `HttpServer_WebSockets` on Host and see if I can reproduce something.

Thanks again
Regards
Stefano

slaff, 2023-12-15:
> no problem to paste it in this thread, let me to know ...

Yes, please do so.

sreggy, 2023-12-15:
Hi @slaff 

I've tried to remove the WebSocker, leaving only the loading of the `index.html`

following the decode-stacktrace when with the browser i try to connect to `http://192.168.0.174/`

```
❯ make decode-stacktrace

ezri: Invoking 'decode-stacktrace' for Esp8266 (debug) architecture
Decode stack trace: Paste stack trace here
/opt/homebrew/Cellar/python@3.11/3.11.6_1/Frameworks/Python.framework/Versions/3.11/lib/python3.11/subprocess.py:1010: RuntimeWarning: line buffering (buffering=1) isn't supported in binary mode, the default buffer size will be used
  self.stdin = io.open(p2cwrite, 'wb', bufsize)

================================================================
3ffffd90:  4025adac 00000005 3ffffdd0 40240498  
3ffffda0:  3fff44d8 00000003 3fff40c8 402404de  
3ffffdb0:  4025adc0 3ffffdc0 3fff4460 40236958  
3ffffdc0:  3ffef7b0 00000003 3fff40c80x4025adac: operator delete[](void*) at ??:?
0x40240498: String::setlen(unsigned int) at /Users/a020900/sming/Sming/Sming/Wiring/WString.h:870
 (inlined by) String::copy(__FlashStringHelper const*, unsigned int) at /Users/a020900/sming/Sming/Sming/Wiring/WString.cpp:263
 (inlined by) String::copy(__FlashStringHelper const*, unsigned int) at /Users/a020900/sming/Sming/Sming/Wiring/WString.cpp:255
0x402404de: String::setString(__FlashStringHelper const*, unsigned int) at /Users/a020900/sming/Sming/Sming/Wiring/WString.cpp:128
0x4025adc0: operator delete(void*, unsigned int) at /Users/a020900/sming/Sming/Sming/Arch/Esp8266/Components/heap/alloc.cpp:47
 40256ebc  
3ffffdd0:  00000000 00000000 80000000 40260a48  
3ffffde0:  3ffef7b0 00000002 3fff40c8 40x40236958: String::String(__FlashStringHelper const*, unsigned int) at /Users/a020900/sming/Sming/Sming/Wiring/WString.h:180
0x40256ebc: String::operator==(String const&) const at /Users/a020900/sming/Sming/Sming/Wiring/WString.h:456
 (inlined by) HttpServerConnection::onReadyToSendData(TcpConnectionEvent) at /Users/a020900/sming/Sming/Sming/Components/Network/src/Network/Http/HttpServerConnection.cpp:235
0x40260a48: TcpClient::freeStreams() at /Users/a020900/sming/Sming/Sming/Components/Network/src/Network/TcpClient.cpp:19
0253eb4  
3ffffdf0:  54207525 6c437063 746e6569 720x40253eb4: TcpClient::pushAsyncPart() at /Users/a020900/sming/Sming/Sming/Components/Network/src/Network/TcpClient.cpp:201
747320  
3ffffe00:  206d6165 696e6966 64656873 00000a0d  
3ffffe10:  74c8fc6c 00000002 3fff40c8 402540e8  
3ffffe20:  3ffffe60 3ffffe50 00000004 000000x402540e8: TcpClient::onReadyToSendData(TcpConnectionEvent) at /Users/a020900/sming/Sming/Sming/Components/Network/src/Network/TcpClient.cpp:174
030  
3ffffe30:  3ffef7b0 00000002 3fff40c8 40256ef0  
3ffffe40:  3ffffe60 3ffffe50 00000004 00002200x40256ef0: HttpServerConnection::onReadyToSendData(TcpConnectionEvent) at /Users/a020900/sming/Sming/Sming/Components/Network/src/Network/Http/HttpServerConnection.cpp:252
0  
3ffffe50:  3ffef7b0 0000056e 3fff40c8 40252e79  
3ffffe60:  00000002 00000000 4000050c 00000000 0x40252e79: TcpConnection::trySend(TcpConnectionEvent) at /Users/a020900/sming/Sming/Sming/Components/Network/src/Network/TcpConnection.h:196
 
3ffffe70:  4021e47a 3fff4af0 3fff40c8 40252f26  0x4000050c: ?? ??:0

3ffffe80:  54207525 25205043 65732070 203a746e  
0x4021e47a: pbuf_free at /Users/a020900/sming/Sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/core/pbuf.c:758
0x40252f26: TcpConnection::onSent(unsigned short) at /Users/a020900/sming/Sming/Sming/Components/Network/src/Network/TcpConnection.cpp:153
3ffffe90:  0a0d7525 00000000 3ffef7a4 4021dad5  
30x4021dad5: tcp_seg_free at /Users/a020900/sming/Sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/core/tcp.c:1084
ffffea0:  40220000 3ffef7a0 3fff40c8 40254168  
3f0x40220000: dhcp_release at /Users/a020900/sming/Sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/core/dhcp.c:1263
0x40254168: TcpClient::onSent(unsigned short) at /Users/a020900/sming/Sming/Sming/Components/Network/src/Network/TcpClient.cpp:218
fffeb0:  3ffef76c 3ffef768 3fff40c8 40253988  
3ff0x40253988: TcpConnection::internalOnSent(unsigned short) at /Users/a020900/sming/Sming/Sming/Components/Network/src/Network/TcpConnection.cpp:501
ffec0:  00000000 3ffef7a0 3ffef79c 402539b1  
3fff0x402539b1: operator() at /Users/a020900/sming/Sming/Sming/Components/Network/src/Network/TcpConnection.cpp:313
 (inlined by) _FUN at /Users/a020900/sming/Sming/Sming/Components/Network/src/Network/TcpConnection.cpp:314
fed0:  3ffef7b0 3ffef7a0 3ffef79c 4021d769  
3ffff0x4021d769: tcp_input at /Users/a020900/sming/Sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/core/tcp_in.c:370 (discriminator 1)
ee0:  ae00a8c0 00000000 00001050 00000010  
3ffffef0:  00000000 3fff3530 00000184 4021e5f8  
3fffff00x4021e5f8: pbuf_alloc at /Users/a020900/sming/Sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/core/pbuf.c:389
0:  3ffe0000 00000024 00000008 3ffeaf46  
3fffff10:  3fff3540 3ffef768 3ffef770 4021b065  
3fffff20:0x4021b065: ip_input at /Users/a020900/sming/Sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/core/ipv4/ip.c:559
  00000014 00000000 a4603a55 3fff2748  
3fffff30:  00000000 3fff3540 0000001c 3ffeaf38  
3fffff40:  3fffdc80 3fff3540 3fff2748 4021abc5  
3fffff50:  00x4021abc5: ethernet_input at /Users/a020900/sming/Sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/netif/etharp.c:1379
[... 147 lines cut ...]
[... cut ...]
    response.sendNamedStream(tmpl); // this template object will be deleted automatically
}

void onFile(HttpRequest &request, HttpResponse &response)
{
    String file = request.uri.getRelativePath();

    if (file[0] == '.')
        response.code = HTTP_STATUS_FORBIDDEN;
    else
    {
        response.setCache(86400, true); // It's important to use cache for better performance.
        response.sendFile(file);
    }
}
```
Probably I'm doing some trivial error ...

Thanks a lot for your support
Regards
Stefano

slaff, 2023-12-18:
@sreggy Please put the source code that is segfaulting for you in a separate GitHub repository and paste the URL here. I will compile it for the Host architecture. And run valgrind and other tools if there is an issue.

sreggy, 2023-12-18:
Hi @slaff,
you can find my project in git at `https://github.com/sreggy/ezri.git` 

Available for any further information.

thanks in advance
regards
Stefano

slaff, 2023-12-18:
Since the issue is not related to CommandProcessing( see here: https://github.com/sreggy/ezri/issues/1) I will close this issue. You can create a new one if you are certain that there is a problem with CommandProcessing AND provide a small/simple/sample code that demonstrates it.

closed 2023-12-18 by slaff