SmingHub/Sming #2669: Latest Sming install broke my code, how to install an old Sming release ?

profjmer, 2023-10-02:
How can i detect the Sming framework from logs ?
 boot: ESP-IDF v4.4.3-360-g0a3cf0120e 2nd stage bootloader, is this the ESP-IDF installed version

cpu_start: Project name:     Sming
cpu_start: App version:      4.0.0-rc4-497-gc3684aba,  is this the Sming version installed
cpu_start: Compile time:     Dec 27 2022 20:17:48

How can I get back those specific versions ?
Thanks

slaff, 2023-10-04:
@mikee47 can you help here?

profjmer, 2023-10-04:
I will try that version. 
Thank you

profjmer, 2023-10-04:
I can't provide valuable info, esp32 is always resetting.

slaff, 2023-10-05:
Are getting back a stack trace? Have you tried this: https://sming.readthedocs.io/en/latest/_inc/samples/LiveDebug/index.html?highlight=decode-stacktrace#exception-handling ?

profjmer, 2023-10-05:
I'll try, thanks

slaff, 2023-12-18:
@profjmer closing due to inactivity. If you have the requested information paste it here and we might re-open the issue.

closed 2023-12-18 by slaff

profjmer, 2024-01-05:
It was not SMING but ESP-IDF, version ESP-IDF-4.3 works fine but versions
>4.3 have problems with wifi samples.

mikee47, 2024-01-05:
> I can't provide valuable info, esp32 is always resetting.

@profjmer If you're using a dual-core chip then it's probably the WDT as discussed here #2653.
Try running `make sdk-menuconfig` and setting `FREERTOS_UNICORE=y`, then `make clean components-clean` before rebuilding.