SmingHub/Sming #2682: ESP32 on MacOS M2 (ARM)

sreggy, 2023-11-14:
Dear all,
i've installed esp-idf 5.3 with Sming master branch. when I try to compile a sample like Basic_Blink i receive a lot of error:
any suggestion? I've also tried with SMING_RELEASE=1 without any success.

Thanks in advance for support.
regards
Stefano

---
❯ make SMING_ARCH=Esp32

Basic_Blink: Invoking 'all' for Esp32/esp32 (debug) architecture
Using configuration 'standard'
ninja: Entering directory `/Users/reggy/sming/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk'
[1/4] cd /Users/reggy/sming/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/esp-idf/esptool_py && /Users/reggy/.espressif/python_env/...ebug/build/esp32/sdk/partition_table/partition-table.bin /Users/reggy/sming/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/Sming.bin
Sming.bin binary size 0x24300 bytes. Smallest app partition is 0x100000 bytes. 0xdbd00 bytes (86%) free.
[2/4] Performing build step for 'bootloader'
[1/1] cd /Users/reggy/sming/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/bootloader/esp-idf/esptool_py && /Users/reggy/.espressif/python_env/idf5.3_py3.11_env/bin/python /Users/reggy/sming/esp32/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /Users/reggy/sming/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/bootloader/bootloader.bin
Bootloader binary size 0x6810 bytes. 0x7f0 bytes (7%) free.
[4/4] Completed 'bootloader'
Copying generated SDK libraries
/Library/Developer/CommandLineTools/usr/bin/make components application

Building /Users/reggy/sming/Sming/Sming/out/Esp32/esp32/debug/lib/clib-Storage-ddb3c2344901dd3c699241077ab0010f.a
C+ /Users/reggy/sming/Sming/Sming/Components/Storage/src/Debug.cpp
/Users/reggy/sming/Sming/Sming/Components/Storage/src/Debug.cpp: In function 'void Storage::Debug::listPartitions(Print&)':
/Users/reggy/sming/Sming/Sming/Components/Storage/src/Debug.cpp:12:20: **error: using a dangling pointer to 'buf' [-Werror=dangling-pointer=]**
   12 |         out.println(_F("Registered partitions:"));
      |         ~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/reggy/sming/Sming/Sming/Components/FlashString/src/include/FlashString/config.hpp:24,
                 from /Users/reggy/sming/Sming/Sming/Components/FlashString/src/include/FlashString/Utility.hpp:24,
                 from /Users/reggy/sming/Sming/Sming/Components/FlashString/src/include/FlashString/Object.hpp:24,
                 from /Users/reggy/sming/Sming/Sming/Components/FlashString/src/include/FlashString/String.hpp:24,
                 from /Users/reggy/sming/Sming/Sming/Wiring/WString.h:64,
                 from /Users/reggy/sming/Sming/Sming/Wiring/Print.h:24,
                 from /Users/reggy/sming/Sming/Sming/Components/Storage/src/include/Storage/Debug.h:3,
                 from /Users/reggy/sming/Sming/Sming/Components/Storage/src/Debug.cpp:1:
/Users/reggy/sming/Sming/Sming/Wiring/FakePgmSpace.h:86:27: note: 'buf' declared here
   86 |                 LOAD_PSTR(buf, __pstr__);                                                                                      \
      |                           ^~~
/Users/reggy/sming/Sming/Sming/Wiring/FakePgmSpace.h:149:14: note: in definition of macro 'LOAD_PSTR'
  149 |         char name[ALIGNUP4(sizeof(flash_str))] __attribute__((aligned(4)));                                                \
      |              ^~~~
/Users/reggy/sming/Sming/Sming/Components/Storage/src/Debug.cpp:12:21: note: in expansion of macro '_F'
   12 |         out.println(_F("Registered partitions:"));
      |                     ^~

slaff, 2023-11-14:
@sreggy can you get the latest Sming version from the `develop` branch? The issue seems similar to this one #2659.

sreggy, 2023-11-14:
I did it.
--
❯ git checkout origin/develop
M	Sming/Components/FlashString
M	Sming/Components/IFS
M	Sming/Components/rboot/component.mk
Previous HEAD position was 772a421e Merge branch 'develop'
HEAD is now at 9f55c7a8 Add MPU6050 library and sample code (#2655)
--- 
but the error is not solved.
Thanks again, regards
Stefano
--
Copying generated SDK libraries
/Library/Developer/CommandLineTools/usr/bin/make components application

Building /Users/reggy/sming/Sming/Sming/out/Esp32/esp32/release/lib/clib-Storage-ddb3c2344901dd3c699241077ab0010f.a
C+ /Users/reggy/sming/Sming/Sming/Components/Storage/src/Debug.cpp
/Users/reggy/sming/Sming/Sming/Components/Storage/src/Debug.cpp: In function 'void Storage::Debug::listPartitions(Print&)':
/Users/reggy/sming/Sming/Sming/Components/Storage/src/Debug.cpp:12:20: error: using a dangling pointer to 'buf' [-Werror=dangling-pointer=]
   12 |         out.println(_F("Registered partitions:"));
      |         ~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/reggy/sming/Sming/Sming/Components/FlashString/src/include/FlashString/config.hpp:24,
                 from /Users/reggy/sming/Sming/Sming/Components/FlashString/src/include/FlashString/Utility.hpp:24,
                 from /Users/reggy/sming/Sming/Sming/Components/FlashString/src/include/FlashString/Object.hpp:24,
                 from /Users/reggy/sming/Sming/Sming/Components/FlashString/src/include/FlashString/String.hpp:24,
                 from /Users/reggy/sming/Sming/Sming/Wiring/WString.h:64,
                 from /Users/reggy/sming/Sming/Sming/Wiring/Print.h:24,
                 from /Users/reggy/sming/Sming/Sming/Components/Storage/src/include/Storage/Debug.h:3,
                 from /Users/reggy/sming/Sming/Sming/Components/Storage/src/Debug.cpp:1:
/Users/reggy/sming/Sming/Sming/Wiring/FakePgmSpace.h:89:27: note: 'buf' declared here
   89 |                 LOAD_PSTR(buf, __pstr__);                                                                                      \
      |                           ^~~
/Users/reggy/sming/Sming/Sming/Wiring/FakePgmSpace.h:154:14: note: in definition of macro 'LOAD_PSTR'
  154 |         char name[ALIGNUP4(sizeof(flash_str))] __attribute__((aligned(4)));                                                \
      |              ^~~~
/Users/reggy/sming/Sming/Sming/Components/Storage/src/Debug.cpp:12:21: note: in expansion of macro '_F'
   12 |         out.println(_F("Registered partitions:"));
      |                     ^~
In member function 'size_t Print::print(const char*)',
    inlined from 'typename std::enable_if<(! std::is_array< <template-parameter-1-1> >::value), Print&>::type operator<<(Print&, const T&) [with T = char*]' at /Users/reggy/sming/Sming/Sming/Wiring/Print.h:272:9,
    inlined from 'void Storage::Debug::listPartitions(Print&, const Storage::Device&)' at /Users/reggy/sming/Sming/Sming/Components/Storage/src/Debug.cpp:21:29:
/Users/reggy/sming/Sming/Sming/Wiring/Print.h:108:29: error: using a dangling pointer to 'buf' [-Werror=dangling-pointer=]
  108 |                 return write(str);
      |                        ~~~~~^~~~~
/Users/reggy/sming/Sming/Sming/Components/Storage/src/Debug.cpp: In function 'void Storage::Debug::listPartitions(Print&, const Storage::Device&)':
/Users/reggy/sming/Sming/Sming/Wiring/FakePgmSpace.h:89:27: note: 'buf' declared here
   89 |                 LOAD_PSTR(buf, __pstr__);                                                                                      \

slaff, 2023-11-14:
@mikee47 can you help here?

mikee47, 2023-11-15:
You'll probably run into other problems building against IDF 5.3, though if you can get it working great! I've found the IDF needs a bit of patching to get it to build properly as many headers aren't stdC++ compliant. Sming only officially builds against IDF 5.0 at present. See https://sming.readthedocs.io/en/latest/_inc/Sming/Arch/Esp32/README.html#idf-versions.

(Current stable IDF version is 5.1.1 so any further work should probably target that? Or 5.2?)

Regarding the above flash string warnings, IDF 5.0 uses GCC 11.2 which isn't afflicted with this issue. AFAIK the esp32 MMU can handle byte accesses so the workaround in #2671 might work. Easy enough to try, though integrating that into Sming would require proper testing on all esp32 variants, just in case.

sreggy, 2023-11-16:
thanks a lot @mikee47.

I did a test but with no succes, I've cleaned my installation and cloned `mikee47/sming` (branch develop) and IDF `git clone -b release/v5.0 --recursive https://github.com/espressif/esp-idf.git`.

The error has changet to the following one ...

Any suggestion?

Thanks again, regards
Stefano

```
CC /Users/reggy/sming/Sming/Sming/Arch/Esp32/Components/esp32/src/clk.c
C+ /Users/reggy/sming/Sming/Sming/Arch/Esp32/Components/esp32/src/event_loop.cpp
In file included from /Users/reggy/sming/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:74,
                 from /Users/reggy/sming/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/portable.h:58,
                 from /Users/reggy/sming/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:70,
                 from /Users/reggy/sming/esp/esp-idf/components/esp_event/include/esp_event.h:12,
                 from /Users/reggy/sming/Sming/Sming/Arch/Esp32/Components/esp32/src/event_loop.cpp:14:
/Users/reggy/sming/esp/esp-idf/components/esp_hw_support/include/spinlock.h: In function 'bool spinlock_acquire(spinlock_t*, int32_t)':
/Users/reggy/sming/esp/esp-idf/components/esp_hw_support/include/spinlock.h:117:94: error: comparison of integer expressions of different signedness: 'esp_cpu_cycle_count_t' {aka 'long unsigned int'} and 'int32_t' {aka 'long int'} [-Werror=sign-compare]
  117 |     } while ((timeout == SPINLOCK_WAIT_FOREVER) || (esp_cpu_get_cycle_count() - start_count) <= timeout);
      |                                                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~
In file included from /Users/reggy/sming/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:76,
                 from /Users/reggy/sming/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/portable.h:58,
                 from /Users/reggy/sming/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:70,
                 from /Users/reggy/sming/esp/esp-idf/components/esp_event/include/esp_event.h:12,
                 from /Users/reggy/sming/Sming/Sming/Arch/Esp32/Components/esp32/src/event_loop.cpp:14:
/Users/reggy/sming/esp/esp-idf/components/esp_common/include/esp_macros.h: At global scope:
/Users/reggy/sming/esp/esp-idf/components/esp_common/include/esp_macros.h:62:25: error: static assertion failed: CHOOSE_MACRO_VA_ARG() result does not match for 0 arguments
   62 | ESP_STATIC_ASSERT(foo() == 2, "CHOOSE_MACRO_VA_ARG() result does not match for 0 arguments");
      |                         ^
cc1plus: all warnings being treated as errors
make[2]: *** [src/event_loop.o] Error 1
make[1]: *** [esp32-build] Error 2
make: *** [all] Error 2
 ~/sming/Sming/samples/Basic_Blink  @92b9737b ······························································································································ 12s
```

mikee47, 2023-11-16:
Use the normal Sming repository (SmingHub/Sming) but use IDF V5 from https://github.com/mikee47/esp-idf.git, branch `sming/release/v5.0'. If you follow the instructions here https://sming.readthedocs.io/en/latest/getting-started/linux/index.html the SDK will be installed for you at /opt/esp-idf.

sreggy, 2023-11-24:
thanks I did a test on my M2 using a Parallels VM and everithing is fine. Then I've tried the same test on the M2 machine 'physically' ... I'm not able to understand why it doesn't work ... followinf the error ...

Not able to undestand why on physical M2 the following check fail : `-- Detecting C compiler ABI info - failed` 
The same check is fine on Parallels VM.
On both environment, compiling an esp-idf sample with `idf.py build `works fine.

Any soggestions?

Thanks in advance

❯ make SMING_ARCH=Esp32

```
Basic_Blink: Invoking 'all' for Esp32/esp32 (debug) architecture
Using configuration 'standard'
Creating /Users/a020900/sming/Sming/Sming/Arch/Esp32/Components/esp32/project/esp32/debug/sdkconfig.defaults
/Users/a020900/sming/esp/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: reconfigure
Running cmake in directory /Users/a020900/sming/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /Users/a020900/sming/Sming/Sming/Arch/Esp32/Components/esp32/project/esp32/debug"...
-- Found Git: /opt/homebrew/bin/git (found version "2.43.0")
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/a020900/dev/esp/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /Users/a020900/dev/esp/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /Users/a020900/dev/esp/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - broken
CMake Error at /opt/homebrew/Cellar/cmake/3.27.8/share/cmake/Modules/CMakeTestCCompiler.cmake:67 (message):
  The C compiler

    "/Users/a020900/dev/esp/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: '/Users/a020900/sming/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/CMakeFiles/CMakeScratch/TryCompile-hK29IB'

    Run Build Command(s): /opt/homebrew/bin/ninja -v cmTC_7b6c3
    [1/2] /Users/a020900/dev/esp/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc   -mlongcalls -Wno-frame-address -o CMakeFiles/cmTC_7b6c3.dir/testCCompiler.c.obj -c /Users/a020900/sming/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/CMakeFiles/CMakeScratch/TryCompile-hK29IB/testCCompiler.c
    [2/2] : && /Users/a020900/dev/esp/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -mlongcalls -Wno-frame-address -Wl,--gc-sections -Wl,-Map=/Users/a020900/sming/Sming/Sming/Arch/Esp32/Components/esp32/project/esp32/debug/sdkconfig.map     -Wl,-wrap,_write_r -Wl,-wrap,_read_r -Wl,-wrap,putchar -Wl,-wrap,puts -Wl,-wrap,vprintf -Wl,-wrap,printf -Wl,-wrap,vsnprintf -Wl,-wrap,vsprintf -Wl,-wrap,sprintf     -T esp32.peripherals.ld -T esp32.rom.ld -T esp32.rom.api.ld -T esp32.rom.libgcc.ld -T esp32.rom.newlib-nano.ld -Wl,-wrap,esp_event_loop_create_default -Wl,-wrap,esp_event_handler_register -Wl,-wrap,esp_event_handler_unregister -Wl,-wrap,esp_event_handler_instance_register -Wl,-wrap,esp_event_handler_instance_unregister -Wl,-wrap,esp_event_post -Wl,-wrap,esp_event_isr_post -T esp32.rom.newlib-funcs.ld -T esp32.rom.newlib-data.ld -T esp32.rom.syscalls.ld -T esp32.rom.newlib-time.ld -T esp32.rom.eco3.ld -u esp_app_desc -u __cxx_fatal_exception -u __cxa_guard_dummy -u call_user_start_cpu0 -u start_app -u start_app_other_cores -u ld_include_panic_highint_hdl -u newlib_include_heap_impl -u newlib_include_syscalls_impl -u newlib_include_pthread_impl -u pthread_include_pthread_impl -u pthread_include_pthread_cond_impl -u pthread_include_pthread_local_storage_impl -Wl,-wrap,_Unwind_SetEnableExceptionFdeSorting -Wl,-wrap,__register_frame_info_bases -Wl,-wrap,__register_frame_info -Wl,-wrap,__register_frame -Wl,-wrap,__register_frame_info_table_bases -Wl,-wrap,__register_frame_info_table -Wl,-wrap,__register_frame_table -Wl,-wrap,__deregister_frame_info_bases -Wl,-wrap,__deregister_frame_info -Wl,-wrap,_Unwind_Find_FDE -Wl,-wrap,_Unwind_GetGR -Wl,-wrap,_Unwind_GetCFA -Wl,-wrap,_Unwind_GetIP -Wl,-wrap,_Unwind_GetIPInfo -Wl,-wrap,_Unwind_GetRegionStart -Wl,-wrap,_Unwind_GetDataRelBase -Wl,-wrap,_Unwind_GetTextRelBase -Wl,-wrap,_Unwind_SetIP -Wl,-wrap,_Unwind_SetGR -Wl,-wrap,_Unwind_GetLanguageSpecificData -Wl,-wrap,_Unwind_FindEnclosingFunction -Wl,-wrap,_Unwind_Resume -Wl,-wrap,_Unwind_RaiseException -Wl,-wrap,_Unwind_DeleteException -Wl,-wrap,_Unwind_ForcedUnwind -Wl,-wrap,_Unwind_Resume_or_Rethrow -Wl,-wrap,_Unwind_Backtrace -Wl,-wrap,__cxa_call_unexpected -Wl,-wrap,__gxx_personality_v0 -T memory.ld -T sections.ld        -nostdlib -Wl,-static CMakeFiles/cmTC_7b6c3.dir/testCCompiler.c.obj -o cmTC_7b6c3   && :
    FAILED: cmTC_7b6c3
    : && /Users/a020900/dev/esp/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -mlongcalls -Wno-frame-address -Wl,--gc-sections -Wl,-Map=/Users/a020900/sming/Sming/Sming/Arch/Esp32/Components/esp32/project/esp32/debug/sdkconfig.map     -Wl,-wrap,_write_r -Wl,-wrap,_read_r -Wl,-wrap,putchar - [... cut ...]
  /Users/a020900/sming/esp/esp-idf/tools/cmake/project.cmake:342 (__project)
  CMakeLists.txt:9 (project)

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1, output of the command is in the /Users/a020900/sming/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/log/idf_py_stderr_output_69354 and /Users/a020900/sming/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/log/idf_py_stdout_output_69354
make: *** [/Users/a020900/sming/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/config/sdkconfig.h] Error 2
```

sreggy, 2023-12-12:
Probably the final update,
Digging deep I discoverd the in my `.zshrc` there was an `LDFLAG` assignment `export LDFLAGS="-L/opt/homebrew/opt/readline/lib $LDFLAGS"`.

With the export the `make SMING_ARCH=Esp32` fails, if I comment the `export` everithing works fine.

Thanks a lot.
Regards
Stefano

closed 2024-06-16 by mikee47