SmingHub/Sming #2710: HardwareTimer repeating when it shouldn't on Esp8266

mikee47, 2024-01-13:
This code (modified `Basic_Blink` sample) doesn't work as expected:

```
#include <SmingCore.h>
#include <HardwareTimer.h>

#define LED_PIN 2

HardwareTimer procTimer;

bool state = true;

void IRAM_ATTR blink()
{
	digitalWrite(LED_PIN, state);
	state = !state;
}

void init()
{
	Serial.begin(COM_SPEED_SERIAL);
	Serial.systemDebugOutput(true);

	debug_i("Welcome");

	pinMode(LED_PIN, OUTPUT);
	procTimer.initializeMs<250>(blink).startOnce();

	auto timer = new SimpleTimer;
	timer->initializeMs<1000>([]() { debug_i("Hello"); });
	timer->start();
}
```

We expect the LED to toggle once then stop, but instead it keeps flashing at ~1.7s intervals. This is because the timer is still enabled and just keeps wrapping. Confirmed using `HardwareTimer1<TIMER_CLKDIV_256>` and we get ~27s intervals.

So timer needs to be explicitly disabled in the callback.

Adding this to the driver code is going to add latency, so will need to take a closer look at this. Suggestions welcome!

Discovered after dealing with another issue, see #2709

mikee47, 2024-01-13:
As far as I can tell the hardware timer has always behaved like this. Probably safest to keep as-is.

mikee47, 2024-01-15:
@slaff Any thoughts on this issue?

slaff, 2024-01-16:
> Probably safest to keep as-is.

Exactly, keep it as it is.

closed 2024-01-16 by mikee47