SmingHub/Sming #2714: maximum number of concurrent tcpConnections

pljakobs, 2024-01-17:
my Firmware uses httpServer to serve both normal http requests and also websocket connections.
In testing, I noticed that once more than two clients connect (clients are right now webapps, but can also be other systems like home automation) very soon, no new connections are accepted anymore.
From Sming, I can see that HttpServer uses TcpServer and that maxActiveConnections is set to 10 and tcp maxConnections defaults to -1 which means "no limit" - however from my logs I can see that the number of concurrent tcp connections maxes out at 5 and new requests are only accepted once one old connection has been freed.
I use persistent connections where possible, but I do have one http and one ws connection per client, I would think that at least three or four concurrent clients would be reasonable, so 10 concurrent tcp connections is what I'd be hoping to be ok with.
My question is: without digging into lwip (again) - is that configured from Sming somewhere? Is it something I can do from within the application or do I have to modify lwip?

mikee47, 2024-01-18:
If you're using an esp8266 then the main restriction will be memory. If this is the case you'll get `CONNECTION DROPPED (free heap...` messages in debug log. The minimum amount of free RAM is configured by `HttpServerSettings::minHeapSize` and defaults to 16K.

pljakobs, 2024-01-18:
that sounds like it is what's going on. Although I'm not getting the CONNECTION DROPPED messages - just no new connection. And also, they don't seem to be so much dropped but not accepted that is - the client waits for a reply until timeout.
but since my free heap is usually around 17000-18000 bytes, I assume that's my choking point then.

closed 2024-01-19 by pljakobs