SmingHub/Sming #2715: On ESP32 GPIO32 GPIO33 can't be used as output

profjmer, 2024-01-30:
I seems we can't use normal pinMode() and digitalWrite() on GPIO32 and GPIO33
If external 32Khz crystal is not used, those pins would be a + to have access.
Is there other way to access them in the framework ?

mikee47, 2024-01-31:
I've not tested those pins, but a quick search of the esp-idf and found this https://github.com/espressif/esp-idf/issues/285

Probably needs some attention in `Arch/Esp32/Core/Digital.cpp`.

mikee47, 2024-01-31:
Also checked logic in Digital.cpp and don't see any issues - the above link refers to disabling the pin as RTC which `pinMode` does already.

mikee47, 2024-01-31:
If you still can't get this working please provide details of your development setup, particularly `ESP32 variant`, `IDF version`. Make sure you're using latest develop branch of Sming please.

profjmer, 2024-01-31:
My wrong,
Basic_Blink works, so it's on my side.
Thank you for your time

closed 2024-01-31 by profjmer