SmingHub/Sming #2718: Kconfig.projbuild

profjmer, 2024-02-17:
Do the Sming build system use Kconfig.projbuild to configure sdk-menuconfig.
If yes where should I put this file

mikee47, 2024-02-17:
See SDK_CUSTOM_CONFIG (https://sming.readthedocs.io/en/latest/_inc/Sming/Arch/Esp32/README.html#envvar-SDK_CUSTOM_CONFIG)

Custom SDK settings for a project can be defined in a separate file and setting this value to the location, relative to the project source root directory.

These will be added to the default SDK settings.

To make the settings current, you must run `make sdk-config-clean`. This will discard any changes made via `make sdk-menuconfig`.

mikee47, 2024-02-17:
Also see https://sming.readthedocs.io/en/latest/_inc/Sming/Arch/Esp32/Components/esp32/index.html#background

profjmer, 2024-02-17:
Let's suppose that I have a Kconfig.projbuild in the root of my project.
Then I must set SDK_CUSTON_CONFIG with  (first attempt SDK_CUSTOMCONFIG =
Kconfig.projbuild, second attempt  SDK_CUSTOMCONFIG =./Kconfig.projbuild)
Then execute make sdk-config-clean
Then where the updated menu should appear ( menuconfig or sdk-menuconfig) ?
Should I change the name of the configuration file ?
What am I doing wrong ?
One more time thanks for your help

mikee47, 2024-02-18:
This should work:

```
make SDK_CUSTOM_CONFIG=Kconfig.projbuild
make sdk-config-clean
make
```

Note: You'd typically add `SDK_CUSTOM_CONFIG=Kconfig.projbuild` to your project's `component.mk` file. Remember to run `make config-clean` if it doesn't take.

Assuming a `debug` build for `esp32` variant the config used can be found in `$(SMING_HOME)/Arch/Esp32/Components/esp32/project/esp32/debug/sdkconfig.defaults`. This is a composite of all the sming-defined defaults with your `Kconfig.projbuild` settings appended.

The IDF tools creates the `sdkconfig` in the same directory. When you run `make sdk-menuconfig` it is this file which gets edited. Running `make sdk-config-clean` removes both of these files.

profjmer, 2024-02-18:
I tested the modification of a menu in ESP-IDF (peripherals/rmt/ir_protocols) with the Kconfig.projbuild provided. It worked under ESP-IDF idf.py menuconfig
I copied the Kconfig.projbuild from the example in BASIC_BLINK.
I proceeded according to your instructions and the compiler complains that all the lines are malformed.
Here are some lines
/tmp/confgen_tmpnkw6klgx:87: warning: ignoring malformed line 'menu "Example Configuration"'
/tmp/confgen_tmpnkw6klgx:88: warning: ignoring malformed line 'choice EXAMPLE_IR_PROTOCOL'
/tmp/confgen_tmpnkw6klgx:89: warning: ignoring malformed line 'prompt "Infrared Protocol"'

Does it work on your end ?

mikee47, 2024-02-18:
Sorry, I've got the wrong end of the stick. I found the example file you mention in IDF version 4.4. Copy that as `Kconfig` to the project root, then run `make menuconfig`.

profjmer, 2024-02-18:
Perfect
Thanks again

closed 2024-02-18 by profjmer

mikee47, 2024-02-19:
These variables are available in makefiles but not in code (no `.h` file is generated). If you require values in code then you might add something like this to your component.mk file:

```
GLOBAL_CFLAGS += \
	-DEXAMPLE_IR_PROTOCOL_NEC=$(EXAMPLE_IR_PROTOCOL_NEC) \
	-DEXAMPLE_IR_PROTOCOL_RC5=$(EXAMPLE_IR_PROTOCOL_RC5) \
	-DEXAMPLE_RMT_RX_CHANNEL=$(EXAMPLE_RMT_RX_CHANNEL) \
	-DEXAMPLE_RMT_RX_GPIO=1910$(EXAMPLE_RMT_RX_GPIO) \
	-DEXAMPLE_RMT_TX_CHANNEL=$(EXAMPLE_RMT_TX_CHANNEL) \
	-DEXAMPLE_RMT_TX_GPIO=18100$(EXAMPLE_RMT_TX_GPIO)
```

mikee47, 2024-02-19:
If you want to generate a .h file for your project, you can always add the required logic to your component.mk file, run commands manually or suggest a PR to the framework to automate this.

You can generate a header file directly like this:

```
CONFIG_= KCONFIG_CONFIG=out/Esp32/esp32/debug/kconfig.config /usr/bin/python3 -m genconfig
```

This will create the file `config.h` by default. Use `--header-path` to change this. See `genconfig --help` for other details.