SmingHub/Sming #2732: ESP32 IDF 4.3, 4.4 due for deprecation

mikee47, 2024-03-15:
I'm raising this issue so we can decide when to remove IDF 4.3, 4.4 support from the codebase.

According to https://github.com/espressif/esp-idf 4.3 is no longer supported and 4.4 support ends in August.
New designs should be using version 5.2.
The current default installed version (using the Sming install scripts) is IDF 4.4.

IDF v5.2 is now out of Beta and I've merged the latest release into the Sming fork at https://github.com/mikee47/esp-idf.
Existing 5.2-beta installations can be updated as follows:

```
cd /opt/esp-idf
git pull
git submodule update --init --recursive
```

mikee47, 2024-03-15:
Note: Currently CI testing is failing for v5.2 release due to deprecation warnings. These will be addressed shortly.

slaff, 2024-03-15:
> when to remove IDF 4.3, 4.4 support from the codebase.

We can deprecate them in 5.2 and remove them officially in 6.0. 6.0 can be the next version after 5.2. 
Are there any known issues with IDF > 4.4 and Sming?

mikee47, 2024-03-17:
> Are there any known issues with IDF > 4.4 and Sming?

Not to my knowledge. Main issue will likely be existing projects using IDF code directly so dependent on a specific IDF version.

closed 2024-12-11 by slaff