SmingHub/Sming #2772: esp32c3 redirection of uart serial to usb serial ?

profjmer, 2024-05-07:
On esp32c3 super mini board, I can flash from USB, but serial.println are using uart on other pins.
If in menuconfig->component config->esp system settings->channel for console output(usb serial/JTAG controller) is chosen, then output is written to USB-CDC but chip is always resetting rapidly.
What would be the procedure to enable redirection of serial.println() to USB-CDC ?

mikee47, 2024-05-07:
Sming doesn't use the IDF console code so alternative output mechanisms won't work.

The C3 doesn't have real USB hardware but instead provides ROM functions (`esp_rom_uart_xxx()`) to access USB/CDC port.

The issue with `Serial` specifically is that it refers to an instance of `HardwareSerial` which is a wrapper for the low-level uart driver: code is in `Sming/Arch/Esp32/Components/driver/uart.cpp`. The Esp8266 implementation supports 'virtual' ports to allow redirection inside GDB, so a similar approach could be taken here. However, serial input would need to be polled as there's no interrupt provision for the C3 that I can see.

An initial, simpler approach would be to implement a `Stream` class to provide access specifically for the C3.

If you only require the port for `debug_x` functions then provide an `m_nputs` implementation and pass it to `m_setPuts()`.

The boot loop (rapid resetting) cause could be caused by a number of issues but can usually be diagnosed via bootloader messages.

profjmer, 2024-05-07:
thanks for info

closed 2024-05-07 by profjmer

mikee47, 2024-07-03:
Just looking at this issue again and I was wrong about the use of `esp_rom_uart_xxx`. In fact, it's got a well-defined set of low-level routines **with** interrupts available which can be used. I'm working up an extension to the esp32 uart driver so this can be used easily.

mikee47, 2024-07-03:
The reference code is in the IDF under `components/driver/usb_serial_jtag`.