SmingHub/Sming #2780: Websocket server has big delay when sending messages

xeonqq, 2024-05-30:
Issue can reprodued in PR https://github.com/SmingHub/Sming/pull/2779

https://github.com/SmingHub/Sming/assets/4160429/5959ff74-4de7-4e7e-a0d1-c383e5c57d0b

Problem can be seen in the video, in the serial terminal the messages are correctly sent in 4Hz by the timer started in the webserver here https://github.com/SmingHub/Sming/pull/2779/files#diff-5d28eaa0f4c730196fcd5810fce411e7d3ae309cd0815b15c6cd85bb9c8444d5R36

However on the receiving side the messages are not received in 4hz, shown in the web browser, 4 messages are recieved at the same time in 1Hz interval. Websocket client code here https://github.com/SmingHub/Sming/pull/2779/files#diff-4afbd2690aa9973b367ecf1c7b8645eacc30423c4ee0062b0740552a869196c1R41

mikee47, 2024-05-31:
Try this patch:

```
diff --git a/Sming/Components/Network/src/Network/Http/Websocket/WebsocketConnection.cpp b/Sming/Components/Network/src/Network/Http/Websocket/WebsocketConnection.cpp
index e38363e4c..8e50ecd6d 100644
--- a/Sming/Components/Network/src/Network/Http/Websocket/WebsocketConnection.cpp
+++ b/Sming/Components/Network/src/Network/Http/Websocket/WebsocketConnection.cpp
@@ -294,7 +294,11 @@ bool WebsocketConnection::send(IDataSourceStream* source, ws_frame_type_t type,
 	}
 
 	// Pass stream to connection
-	return connection->send(sourceRef.release());
+	if(!connection->send(sourceRef.release())) {
+		return false;
+	}
+	connection->commit();
+	return true;
 }
 
 void WebsocketConnection::broadcast(const char* message, size_t length, ws_frame_type_t type)
```

xeonqq, 2024-05-31:
Hi, Mikee, 
Thanks a lot, I think the problem is solved with your patch.
Are you planning to merge it? 

big delay solved with this patch shown below:
https://github.com/SmingHub/Sming/assets/4160429/e3c2bb62-fe45-4307-b955-0077ff6230a0

mikee47, 2024-05-31:
> Thanks a lot, I think the problem is solved with your patch.
> Are you planning to merge it?

Great! PR en-route.

PR #2782 (2024-06-01): Fix delayed sending of websocket messages
Fixes #2780 by kicking TCP when messages are queued. Without this messages are sent lazily by lwip stack.
commit a9021960 (2024-06-01): Fix delayed sending of websocket messages (#2782)

Fixes #2780 by kicking TCP when messages are queued. Without this messages are sent lazily by lwip stack.
closed 2024-06-01 by slaff