SmingHub/Sming #2781: ESP32C3 WiFi problems

profjmer, 2024-05-30:
ESP32 vs ESP32C3 : same samples, different results.
Do we have to do a specific configuratiion for soc esp32c3 ?
[ESP32 vs ESP32C3 SMING-IDF-v5.2-beta1.pdf](https://github.com/SmingHub/Sming/files/15502381/ESP32.vs.ESP32C3.SMING-IDF-v5.2-beta1.pdf)

mikee47, 2024-05-31:
For reference, you should see `boot: ESP-IDF v5.2.1-sming 2nd stage bootloader`  and later on `phy_init: phy_version 1150,7c3c08f,Jan 24 2024,17:32:21` for the current IDF sming fork release.

profjmer, 2024-05-31:
ok

profjmer, 2024-06-06:
Updated to latest, test on c3 for Basic_WiFi compiles and same problem persist. AP activated but not visible in AccessPoint list on PC

mikee47, 2024-06-06:
Re-checked and works fine for me. This is the boot log, see how it compares:

```
mike@fedora:/stripe/sandboxes/sming-dev/samples/Basic_WiFi$ make terminal

Basic_WiFi: Invoking 'terminal' for Esp32/esp32c3 (debug) architecture
Killing Terminal to free /dev/ttyUSB0
make: [/stripe/sandboxes/sming-dev/Sming/Components/terminal/component.mk:32: kill_term] Killed (ignored)
/usr/bin/python3 -m serial.tools.miniterm --raw --encoding ascii --rts 0 --dtr 0 /dev/ttyUSB0 115200
--- forcing DTR inactive
--- forcing RTS inactive
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd5820,len:0x1738
load:0x403cc710,len:0xb9c
load:0x403ce710,len:0x2e40
entry 0x403cc71a
I (32) boot: ESP-IDF v5.2.1-sming 2nd stage bootloader
I (32) boot: compile time Jun  6 2024 18:56:38
I (32) boot: chip revision: v0.3
I (35) boot.esp32c3: SPI Speed      : 40MHz
I (40) boot.esp32c3: SPI Mode       : DIO
I (45) boot.esp32c3: SPI Flash Size : 4MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 spiFlash         unknown          02 01 00000000 00400000
I (73) boot:  1 nvs              WiFi data        01 02 00009000 00006000
I (81) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (88) boot:  3 factory          factory app      00 00 00010000 000f0000
I (96) boot: End of partition table
I (100) esp_image: segment 0: paddr=00010020 vaddr=3c080020 size=1efa0h (126880) map
I (136) esp_image: segment 1: paddr=0002efc8 vaddr=3fc8fa00 size=01050h (  4176) load
I (137) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=78bc8h (494536) map
I (248) esp_image: segment 3: paddr=000a8bf0 vaddr=3fc90a50 size=01890h (  6288) load
I (250) esp_image: segment 4: paddr=000aa488 vaddr=40380000 size=0f9a0h ( 63904) load
I (274) boot: Loaded app from partition at offset 0x10000
I (275) boot: Disabling RNG early entropy source...
I (286) cpu_start: Unicore app
I (295) cpu_start: Pro cpu start user code
I (295) cpu_start: cpu freq: 160000000 Hz
I (295) cpu_start: Application information:
I (298) cpu_start: Project name:     Sming
I (303) cpu_start: App version:      4.0.0-rc4-640-gd4977598-dirty
I (310) cpu_start: Compile time:     Jun  6 2024 18:56:34
I (316) cpu_start: ELF file SHA256:  d31517087...
I (321) cpu_start: ESP-IDF:          v5.2.1-sming
I (327) cpu_start: Min chip rev:     v0.3
I (331) cpu_start: Max chip rev:     v1.99 
I (336) cpu_start: Chip rev:         v0.3
I (341) heap_init: Initializing. RAM available for dynamic allocation:
I (348) heap_init: At 3FC966B0 len 00029950 (166 KiB): RAM
I (354) heap_init: At 3FCC0000 len 0001C710 (113 KiB): Retention RAM
I (361) heap_init: At 3FCDC710 len 00002950 (10 KiB): Retention RAM
I (368) heap_init: At 50000010 len 00001FD8 (7 KiB): RTCRAM
I (375) spi_flash: detected chip: generic
I (379) spi_flash: flash io: dio
I (384) sleep: Configure to isolate all GPIO pins in sleep state
I (390) sleep: Enable automatic switching of GPIO sleep configuration
Sming. Let's do smart things!
I (437) phy_init: phy_version 1150,7c3c08f,Jan 24 2024,17:32:21
I (467) wifi:mode : sta (7c:df:a1:c1:0b:b4) + softAP (7c:df:a1:c1:0b:b5)
I (477) wifi:enable tsf
I (477) wifi:Total power save buffer number: 16
I (477) wifi:Init max length of beacon: 752/752
I (477) wifi:Init max length of beacon: 752/752
198878 Station IP successfully updated
READY!
AP. ip: 192.168.4.1 mac: 7c:df:a1:c1:0b:b5
201776 event WIFI_EVENT|40

202900 event WIFI_EVENT|2

E (497) wifi:sta is connecting, return error
210262 event WIFI_EVENT|12

I (507) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
3034609 event WIFI_EVENT|5

3034893 disconnect from ssid PleaseEnterSSID, reason 201

Disconnected from "PleaseEnterSSID", reason: No AP found
5872813 event WIFI_EVENT|1

5873640 scan completed: 0 found
I (14927) wifi:new:<7,1>, old:<7,1>, ap:<7,1>, sta:<0,0>, prof:7
I (14927) wifi:station: e2:5f:7e:ec:b3:1f join, AID=1, bgn, 40U
14638008 event WIFI_EVENT|14

14639317 station: e2:5f:7e:ec:b3:1f join, AID = 1

I (14967) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2
I (15977) wifi:<ba-add>idx:2 (ifx:1, e2:5f:7e:ec:b3:1f), tid:0, ssn:23, winSize:64
I (21077) wifi:station: e2:5f:7e:ec:b3:1f leave, AID = 1, bss_flags is 33688675, bss:3fcaab54
I (21077) wifi:new:<7,0>, old:<7,1>, ap:<7,1>, sta:<0,0>, prof:7
I (21077) wifi:<ba-del>idx:2, tid:0
20789453 event WIFI_EVENT|15

20791989 station: e2:5f:7e:ec:b3:1fleave, AID = 1

20796483 event WIFI_EVENT|40

--- exit ---
mike@fedora:/stripe/sandboxes/sming-dev/samples/Basic_WiFi$ 
```

profjmer, 2024-06-06:
Here's mine.

ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd5820,len:0x1738
load:0x403cc710,len:0xb9c
load:0x403ce710,len:0x2e40
entry 0x403cc71a
[0;32mI (32) boot: ESP-IDF v5.2.1-sming 2nd stage bootloader[0m
[0;32mI (32) boot: compile time Jun  6 2024 13:57:57[0m
[0;32mI (32) boot: chip revision: v0.4[0m
[0;32mI (35) boot.esp32c3: SPI Speed      : 40MHz[0m
[0;32mI (40) boot.esp32c3: SPI Mode       : DIO[0m
[0;32mI (45) boot.esp32c3: SPI Flash Size : 4MB[0m
[0;32mI (50) boot: Enabling RNG early entropy source...[0m
[0;32mI (55) boot: Partition Table:[0m
[0;32mI (59) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (66) boot:  0 spiFlash         unknown          02 01 00000000 00400000[0m
[0;32mI (73) boot:  1 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (81) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (88) boot:  3 factory          factory app      00 00 00010000 000f0000[0m
[0;32mI (96) boot: End of partition table[0m
[0;32mI (100) esp_image: segment 0: paddr=00010020 vaddr=3c080020 size=1ef18h (126744) map[0m
[0;32mI (136) esp_image: segment 1: paddr=0002ef40 vaddr=3fc8fa00 size=010d8h (  4312) load[0m
[0;32mI (137) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=78be0h (494560) map[0m
[0;32mI (248) esp_image: segment 3: paddr=000a8c08 vaddr=3fc90ad8 size=01808h (  6152) load[0m
[0;32mI (256) esp_image: segment 4: paddr=000aa418 vaddr=40380000 size=0f9a0h ( 63904) load[0m
[0;32mI (277) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (278) boot: Disabling RNG early entropy source...[0m
[0;32mI (289) cpu_start: Unicore app[0m
[0;32mI (298) cpu_start: Pro cpu start user code[0m
[0;32mI (298) cpu_start: cpu freq: 160000000 Hz[0m
[0;32mI (298) cpu_start: Application information:[0m
[0;32mI (301) cpu_start: Project name:     Sming[0m
[0;32mI (306) cpu_start: App version:      4.0.0-rc4-605-g0e468916[0m
[0;32mI (312) cpu_start: Compile time:     Jun  6 2024 13:57:31[0m
[0;32mI (318) cpu_start: ELF file SHA256:  8097ad001...[0m
[0;32mI (324) cpu_start: ESP-IDF:          v5.2.1-sming[0m
[0;32mI (329) cpu_start: Min chip rev:     v0.3[0m
[0;32mI (334) cpu_start: Max chip rev:     v1.99 [0m
[0;32mI (338) cpu_start: Chip rev:         v0.4[0m
[0;32mI (343) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (351) heap_init: At 3FC966B0 len 00029950 (166 KiB): RAM[0m
[0;32mI (357) heap_init: At 3FCC0000 len 0001C710 (113 KiB): Retention RAM[0m
[0;32mI (364) heap_init: At 3FCDC710 len 00002950 (10 KiB): Retention RAM[0m
[0;32mI (371) heap_init: At 50000010 len 00001FD8 (7 KiB): RTCRAM[0m
[0;32mI (378) spi_flash: detected chip: generic[0m
[0;32mI (381) spi_flash: flash io: dio[0m
[0;32mI (386) sleep: Configure to isolate all GPIO pins in sleep state[0m
[0;32mI (392) sleep: Enable automatic switching of GPIO sleep configuration[0m
Sming. Let's do smart things!
[0;32mI (440) phy_init: phy_version 1150,7c3c08f,Jan 24 2024,17:32:21[0m
I (480) wifi:mode : sta (34:b7:da:fb:46:a4) + softAP (34:b7:da:fb:46:a5)
I (480) wifi:enable tsf
I (480) wifi:Total power save buffer number: 16
I (480) wifi:Init max length of beacon: 752/752
I (480) wifi:Init max length of beacon: 752/752
200126 Station IP successfully updated
READY!
AP. ip: 192.168.4.1 mac: 34:b7:da:fb:46:a5
203186 event WIFI_EVENT|40

204042 event WIFI_EVENT|2

E (500) wifi:sta is connecting, return error
211318 event WIFI_EVENT|12

[0;32mI (510) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1[0m
3035919 event WIFI_EVENT|5

3036198 disconnect from ssid PleaseEnterSSID, reason 201

Disconnected from "PleaseEnterSSID", reason: No AP found
5872423 event WIFI_EVENT|1

5873134 scan completed: 0 found

mikee47, 2024-06-06:
Hmm. Looks pretty much identical. Have you tried a `make flashinit` then `make flash` ? (In case there's some internal IDF config stored somewhere messing things up.)

profjmer, 2024-06-06:
I changed for another C3, and AP is showing OK.
Problem was a bad C3 super mini.
Thanks for your help

closed 2024-06-06 by profjmer